MarlinFirmware/Marlin #8277: 2.0 error in G33.cpp, sprintf_P not for ARM

Vynt-1, 2017-11-05:
------------------------------------------------------------
### Feature Request
"sprintf_P(...)" and similar functions don't work with ARM core
------------------------------------------------------------
### Compile Error
When I compile with `#define DELTA_AUTO_CALIBRATION` I get an error:
```
...AppData\Local\Arduino15\packages\arduino\hardware\sam\1.6.11\cores\arduino/avr/pgmspace.h:99:59: error: expected primary-expression before ')' token

 #define sprintf_P(s, f, ...) sprintf((s), (f), __VA_ARGS__)

...sketch\src\gcode\calibrate\G33.cpp:652:9: note: in expansion of macro 'sprintf_P'

         sprintf_P(mess, PSTR("Calibration sd:"));

etc.
```
------------------------------------------------------------
### Bug Report
- Description: ---
- Expected behaviour: ---
- Actual behaviour: ---
- Steps to reproduce:
  - Do this
  - Do that

Attach a ZIP of `Configuration.h` and `Configuration_adv.h` by dropping here.

------------------------------------------------------------
### Bug Report Tips
- When troubleshooting, use `M502` followed by `M500` to reset EEPROM to defaults.
- Use `DEBUG_LEVELING_FEATURE` with `M111 S247` for detailed logging of homing/leveling.
- Format text with: **bold**, _italic_, `code`.
- Format C++ with three backticks, plus "cpp":
```cpp
void my_function(bool do_it) {
  // Hold this spot
}
```

thinkyhead, 2017-11-07:
Looks like that particular instance is wasteful. It should just be `strcpy_P`.

I don't think it's a problem of ARM. It's that the definition of `strcpy_P` requires at least 3 arguments, and if there are only two then it chokes.

thinkyhead, 2017-11-07:
Please check against the latest code, patched to remove the problem instances of `sprintf_P`. If the problem isn't resolved, feel free to re-open the issue.

closed 2017-11-07 by thinkyhead