MarlinFirmware/Marlin #8279: Bed size error MARLIN 1.1.6

Stempek5, 2017-11-05:
Hi there,

Hardware:
-Arduino Mega 2560
-RAMPS 1.4
-A4988 stepsticks
-self build drawing plotter made of DVD/floppy drive motors 

I've got an error while trying to compile program:
[image]

I don't use any of extruders, temperature sensors, nor endstops. It's just a drawing machine to prepare myself building larger 3D printer. 

BED area is like 45mm x 50 mm but i set parameters like this:
[image]

I have also set rest of parameters like this:

```
#define DEFAULT_AXIS_STEPS_PER_UNIT   { 30, 30, 40, 500 }
#define DEFAULT_MAX_FEEDRATE          { 10, 10, 2, 25 }
#define DEFAULT_MAX_ACCELERATION      { 100, 100, 10, 10000 }
#define DEFAULT_ACCELERATION          50    
#define DEFAULT_RETRACT_ACCELERATION  50    
#define DEFAULT_TRAVEL_ACCELERATION   10 
```
but I have no idea I did it properly.

Motors used in CNC:
Y axis (ground/working area):
Resistance per coil 9,5ohms
so I calculated Vref=0,420V
X axis (horizontally left-right movement):
Resistance per coil 11,8ohms
so I calculated Vref=0,340V
Z axis (motor used from floppy drive):
Resistance per coil 30,3ohms
so I calculated Vref=0,133V

Do you have any advices what is set wrong (I'm almost sure I set something wrong - but have no idea what)?

RowanMeara, 2017-11-06:
If you attach your config file we can try compiling it to help find your error.

thinkyhead, 2017-11-07:
You will need to upgrade your Arduino IDE. Version 1.5 doesn't do C++11.

thinkyhead, 2017-11-07:
You will need to upgrade your Arduino IDE. Version 1.0.5 doesn't do C++11.

closed 2017-11-07 by thinkyhead