MarlinFirmware/Marlin #8285: Question about pid autotune

GMagician, 2017-11-06:
Is there a reason why not to have pid bed autotune in lcd menu?

tcm0116, 2017-11-06:
I would guess it's because it's not as commonly used as the hotend and no one has taken the time to implement it.

GMagician, 2017-11-06:
I think is not matter of time, it's just 1 line in code and everything is ready to do that.
I wish to add it but before doing that I would like to know if there is something else...

thinkyhead, 2017-11-07:
Indeed, often it's just as quick to make a PR as it is to post a question.

Should it always be in the menu for all machines and all vendors? I think not. It should be an advanced configuration option to enable it.

GMagician, 2017-11-07:
Not enough to enabled when also bedpid active? If you have bed pid is almost sure you need to calibrate it...
But if you repute that an advanced option is better this is what I will do

thinkyhead, 2017-11-07:
Vendors might want to do PID tuning in development, but disable it in production to reduce menu clutter.

GMagician, 2017-11-08:
ok, I'll make a Pr with this.
as advanced option do you mean to put in configuration_adv? or simply a new option in configuration like BEDPID_AUTOTUNE?

GMagician, 2017-11-08:
other question "duplicate" label means that there is another issue with this request?
If affirmative I'll make PR relative to that issue

AnHardt, 2017-11-08:
> BEDPID_AUTOTUNE
Wrong name - confusing.
As far i understood it's only about having a menu option for that.

GMagician, 2017-11-08:
Idea is to act like extruder PID and so add PID values and autotune option.
Code is ready fir bed autotune and almost ready for pid values modification

GMagician, 2017-11-08:
Theoretically I have to add 2 defines. First to add PID values and second to add autotune, since @thinkyhead idea is to hide this new option from menu and not recycle current defines.
I could also use 1 define and recycle autotune from extruders  but I don't know if it's a good idea.
I need some instruction from your skilled dev so I'll prepare an almost ready to merge PR

GMagician, 2017-11-10:
@thinkyhead I added a PR. If you say it's ok I'll do missing things

GMagician, 2017-11-11:
#8365 PR added

closed 2017-11-12 by GMagician