MarlinFirmware/Marlin #8287: ENDSTOPS_ALWAYS_ON_DEFAULT misbehaves when using TMC2130 stallguard for both _MAX and _MIN

marcio-ao, 2017-11-06:
- Description: If you tie both _MAX and _MIN for an axis to the TMC2130 diag pin and attempt to enable ENDSTOPS_ALWAYS_ON_DEFAULT, the axis will lockup as soon as a limit is reached.
- Expected behaviour: Marlin should be able to use stallguard to respect the axis limits.
- Actual behaviour: Marlin gets confused since reaching an axis limit causes both _MAX and _MIN to be activated at the same time. This prevents motion in the opposite direction, which is needed to clear the condition.
- Steps to reproduce:
  - Modify "pins_xxxx.h" for your board so that X_MAX_PIN and X_MIN_PIN both point to the diag pin on the TMC2130 X driver.
  - Configure the axis for stall detection using TMC2130_ADV
  - Enable ENDSTOPS_ALWAYS_ON_DEFAULT
  - Move the X axis to a limit. Motion will stop, but further motion will be ignored. M119 will show both X_MAX and X_MIN as triggered.

It turns out the fix is relatively simple. In "endstops.cpp", _MAX should *not* be updated if _MIN is already triggered; conversely, _MIN should *not* be updated if _MAX is already triggered. This will ensure that once an endstop is hit on one side of an axis, motion in the opposite direction will be allowed, even if X_MAX and X_MIN are tied together (as they must be when stallguard is used for endstops).

This is what the fix looks like for the X axis looks like in "endstops.cpp":

```
  if (X_MOVE_TEST) {
    if (stepper.motor_direction(X_AXIS_HEAD)) {
      if (X_MIN_TEST) { // -direction
        #if HAS_X_MIN
          if((current_endstop_bits & X_MAX) == 0) {
          UPDATE_ENDSTOP(X, MIN);
          }
        #endif
      }
    }
    else if (X_MAX_TEST) { // +direction
      #if HAS_X_MAX
        if((current_endstop_bits & X_MIN) == 0) {
        UPDATE_ENDSTOP(X, MAX);
        }
      #endif
    }
  }
```

Possibly this fix could only be implemented when X_MIN_PIN is equal to X_MAX_PIN.

GMagician, 2017-11-06:
And if for some electrical reason min comes first and you are on a max limit?

marcio-ao, 2017-11-06:
@GMagician: The way "endstops.cpp" is implemented, it already takes into account the direction of motion before setting the endstop bits. If the axis is moving in a positive direction, then _MAX is checked, otherwise _MIN is checked.

The reason that tying _MAX and _MIN together fails is because of the following sequence of events:

1) When X is moving to MAX, the _MAX endstop will be checked and will be set once a stall is detected (_MIN remains unset, because the motion is towards MAX)
2) Suppose a new command is issued telling the stepper to move towards MIN. At this point, the direction is towards MIN, so immediately _MAX endstop will be checked and will be set based on the stall from the previous move.
3) Both _MAX and _MIN are now set and no motion can be changed.

With my fix, this is what happens:

1) When X is moving to MAX, the _MAX endstop will be checked and will be set once a stall is detected (_MIN remains unset, because the motion is towards MAX)
2) Suppose a new command is issued telling the stepper to move towards MIN. At this point, the direction is towards MIN, but since MAX is already set the update to _MIN will not happen due to my extra check. Only _MAX remains set.
3) Since _MIN is cleared, the motion proceeds and the stallguard value is cleared.

teemuatlut, 2017-11-06:
Curious, what are your bump settings?

GMagician, 2017-11-06:
that 's clear..until you turn off ;-)

marcio-ao, 2017-11-06:
@teemuatlut: I am setting my bumps to 0 mm. The issue does not occur during homing, that works fine.

marcio-ao, 2017-11-06:
@GMagician: I made a couple mistakes in my original post. Corrected.

marcio-ao, 2017-11-06:
@GMagician: Oh, you were asking what happens if the printer is turned off with the endstops triggered? Stallguard does not handle that case very well. Let's just pretend it never happens...

I actually modified G28 to move away from the endstops for this very reason, as doing two G28s in a row tends to be one occasion in which stallguard as endstops will be somewhat less effective and more noisy.

teemuatlut, 2017-11-06:
The reason I said about bumping distance is because when the stallGuard gets triggered it tends to stick for a bit. So if your bump distance is say 5mm, you then home and find the hard stop and stallGuard is triggered. Then Marlin proceeds with the 5mm bump move (endstop is not checked). After that the second homing move (post bump) is executed and the diag pin is still triggered and now Marlin thinks it has found the hard stop even though we're 5mm from the stop. That is why there is the notion to set bump to 0.

marcio-ao, 2017-11-06:
@teemuatlut: My experience has been that stallguard doesn't stick for just for a bit, it sticks indefinitely. The only way to clear it is to do another motion which does not stall. This is why I had to add special logic to make the endstops work correctly with ENDSTOPS_ALWAYS_ON_DEFAULT, because of the sticky nature of the stallguard bit.

marcio-ao, 2017-11-06:
It looks like my code had a mistake:

`if((current_endstop_bits & X_MIN) == 0) {`

Should be:

`if(TEST(current_endstop_bits, X_MIN) == 0) {`

I'll update it above.

GMagician, 2017-11-07:
the correct implementation, for all security reasons, should be to have separated inputs...but if you really need to have just one I think your solution is valid.

marcio-ao, 2017-11-07:
@GMagician: I think the correct behavior would be to make the behavior conditional on _MAX_PIN being equal to _MIN_PIN, ie:

if(X_MAX_PIN != X_MIN_PIN || TEST(current_endstop_bits, X_MAX) == 0) {
   UPDATE_ENDSTOP(X, MAX);
}

In the case in which X_MAX_PIN != X_MIN_PIN, I believe the compiler would optimize away the test. If not, this may work too:

#if X_MAX_PIN == X_MIN_PIN
   if(TEST(current_endstop_bits, X_MAX) == 0) {
      UPDATE_ENDSTOP(X, MAX);
   }
#else
   UPDATE_ENDSTOP(X, MAX);
#endif

None of these are particularly readable though. I'm not sure if rolling this into a macro would be any better (macros are already nested pretty deep!):

#define UPDATE_ENDSTOP_MAX(AXIS) if(AXIS##_MAX_PIN != AXIS##_MIN_PIN || TEST(current_endstop_bits, AXIS##_MIN) == 0) UPDATE_ENDSTOP(AXIS, MAX);
#define UPDATE_ENDSTOP_MIN(AXIS) if(AXIS##_MIN_PIN != AXIS##_MAX_PIN || TEST(current_endstop_bits, AXIS##_MAX) == 0) UPDATE_ENDSTOP(AXIS, MIN);

marcio-ao, 2017-11-07:
Admittedly, being able to use a TMC2130 is somewhat of a niche use case for having MAX and MIN on the same pin, but it just occurred to me that someone could also make use of this feature to save on I/O pins when using normally closed _MAX and _MIN end switches wired in series. Anyhow, just trying to justify why this feature may actually be worth the complexity it introduces :)

teemuatlut, 2017-11-07:
I don't even think it's a TMC2130 or a stallGuard specific issue. You'd probably get the same if you wired mechanical endstops to the same pin. If you home to MIN when MAX is already triggered (and thus MIN as well), you'd have the same result.

AnHardt, 2017-11-07:
In my opinion hitting any endstop, when not homing, is a good reason to lock or kill the machine.
Software endstops should have prevented from that. If not - we have a problem.

Manual recovery from the min/max problem is:
* M121 - Disable endstops detection.
* Move axis to the right direction - machine can't know.
* M120 - Enable endstops detection.

marcio-ao, 2017-11-07:
@AnHardt: That's a valid point, my fix is meant to correct a problem that should seldom occur, or can be avoided by other means.

This fix is most useful in the case in which NO_MOTION_BEFORE_HOMING is turned off (this was in fact a feature I had requested, once I realized software endstops are effective only after homing). Our printers have in the past had all six hardware endstops and users could control it interactively from Cura without homing. I fear some users may be disappointed with that change, which is why I chose to investigate whether it was possible to use stallguard to detect both _MIN and _MAX. At least now I know it is possible :)

thinkyhead, 2017-11-07:
Do we like the solution proposed in the original post best, or is something else brewing…?

marcio-ao, 2017-11-07:
@thinkyhead: I'm having second thoughts about using stallguard with ENDSTOPS_ALWAYS_ON_DEFAULT, as I am starting to see it is prone to false positives. During homing, an occasional error can be tolerated; but during a print, it would be catastrophic. So before you merge anything in, let me see whether stallguard is actually usable in this manner.

thinkyhead, 2017-11-07:
Sure! In my opinion the only good reason for `ENDSTOPS_ALWAYS_ON_DEFAULT` to be enabled is if there might be some chance of loose pulleys or a machine getting very much out of alignment, and then only in combination with `ABORT_ON_ENDSTOP_HIT_FEATURE_ENABLED`. Both are essentially pointless on a DELTA.

AnHardt, 2017-11-07:
** stallguard** is that breakthru to have hardware min endstops on a delta - or at least the counterpart of the current top hardware endstops.
Maybe in some configurations a  stallguard-probe is possible. (Reduced current to minimise power?)

marcio-ao, 2017-11-07:
@AnHardt: Actually, I have already been experimenting on using stallguard on Z. Due to the gearing on the motors, I had to reduce the Z current to 400mA during probing and set the sensitivity rather high (SGT=0). It turns out stallguard used in this way allows our printers to home to Z_MIN (saving the round trip to Z_MAX), but I find it deflects the bed quite a bit and isn't very accurate. So once I find a rough estimate for Z using stallguard, I wipe the nozzle and proceed to do a electrical-conductivity probe as in our current generation printers.

psavva, 2017-11-08:
I received my TMC2130 drivers yesterday.  I'm also very interested in getting them setup on the Re-Arm.
If you find any good configuration using stallguard for your Zv stall detection as a Z probe, let me know.

marcio-ao, 2017-11-08:
@psavva: Look at https://code.alephobjects.com/diffusion/MARLIN/ and switch to the "devel" branch. All changes done to Marlin are made as macros prefixed with LULBOTZ_, which defined in "Conditionals_LulzBot.h", so you can easily find the changes in the Marlin source by grepping for LULZBOT_ in the source files. The stuff related to stallguard is under the "SENSORLESS HOMING" heading in "Conditionals_LulzBot.h", and in particular LULZBOT_SENSORLESS_HOMING_Z for Z.

The tweaks in particular:
   - LULZBOT_ADJUST_Z_HOMING_CURRENT adjust the current and TMC sensitivity prior to homing Z (500mA and 1, respectively) and resets it after homing.
   - LULZBOT_TMC2130_ADV inits the drivers
   - LULZBOT_CLEAR_STALLGUARD_FLAG is a patch to clear the stallguard flag in tmc2130_sensorless_homing
   - LULZBOT_AFTER_Z_HOME_ACTION moves the head away from bumpers after homing

thinkyhead, 2017-11-15:
My basic approach to integrating TMC5130 is going to be to grab the TRAMS fork and use it almost completely as-is. I'll check out your fork also @marcio-ao and see what more it brings to the tableau.

teemuatlut, 2017-11-16:
https://github.com/teemuatlut/Marlin/tree/trams-updated
Needs testing and some investigation into lockups it has.

marcio-ao, 2018-08-20:
Closing this issue since we no longer use ENDSTOPS_ALWAYS_ON_DEFAULT when using stallguard due to the risk of false positives. Also, NO_MOTION_BEFORE_HOMING eliminates the need for ENDSTOPS_ALWAYS_ON_DEFAULT

closed 2018-08-20 by marcio-ao