MarlinFirmware/Marlin #8289: Anet A8 Eeprom CRC Mismatch

agreen, 2017-11-06:
### Bug Report
- Description: Running latest version of 1.1.x-bugfix and keep getting Eeprom CRC mismatch errors
- Expected behaviour: load values from eeprom
- Actual behaviour: default values loaded, due to CRC mismatch
- Steps to reproduce:
  - use latest 1.1.x-bugfix, flash to Anet A8 board.

agreen, 2017-11-06:
Also, Not sure if there are any eeprom tests you'd like me to run, just to make sure the eeprom hasn't gone bad...  I did notice the CNC_COORDINATE_SYSTEMS patch made some changes to EEPROM though, so thought I'd check here first.

halfluck, 2017-11-07:
Hey Guys,

Just flashed 1.1.x-bugfix and i'm getting "Error: EEPROM CRC mismatch - (stored) 8865 != 25765 (calculated)!"

in my logs too tried a M502 then M500 multiple times without any luck.

Cheers Rob.

Roxy-3D, 2017-11-07:
Does #8283 or #8284 fix it for you?   I suspect it will....

agreen, 2017-11-07:
Yes it fixed the issue. Thanks.

Roxy-3D, 2017-11-07:
OK...   It should be fixed and working in the main bugfix-v1.1.x branch.    (You don't have a 'special' version.   You have what is in the main branch.)

psavva, 2017-11-07:
Suggest to please close this issue as it is resolved.

closed 2017-11-07 by Roxy-3D