MarlinFirmware/Marlin #8290: [FR] G93 Inverse Time Mode

jminardi, 2017-11-06:
Does Marlin support Inverse Time Mode? I know that grbl does and IIRC Marlin is based on grbl.

See G93 in grbl here:
https://github.com/grbl/grbl/blob/master/grbl/gcode.c#L227

If it isn't supported, would this be a hard feature to add?

thinkyhead, 2017-11-07:
I have it implemented in one of my branches, and may add it in the future.

jminardi, 2017-11-07:
Have you published that branch on github? Can I try it out?

thinkyhead, 2017-11-07:
It's not quite complete yet. See #8311.

jminardi, 2017-11-07:
Thanks for pushing so fast!

Looks like it works for G1 currently which is all I need for now.

thinkyhead, 2017-11-07:
Let me know how it works for you. I'll look into `G2`/`G3` soon. Length of an arc (or multiple circles) is a pretty simple calculation.

boelle, 2019-02-19:
@jminardi did it work?

jminardi, 2019-03-06:
@boelle It has been a while since I worked with this but IIRC I was able to get it to work for me needs.