MarlinFirmware/Marlin #8293: UBL issue/probing air

cdedwards, 2017-11-07:
I've been attempting to use UBL again and I'm thinking that these numbers are not good.. In fact, when I G29 P1, the probe never actually hits the bed. for some reason, I am about 10mm too high. I'm getting the same thing with G33 P1 though, so not sure where the issue comes in. I can print and actually get some really nice prints but this has me stumped.
[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/1448393/Marlin.zip)
file included is actually a rar file since we can't upload them here.

SENT: G29 T
READ: 
READ: Bed Topography Report:
READ: 
READ: (0,6)                                           (6,6)
READ: (-135,135)                                        (135,135)
READ:      .          .       8.321    8.152    8.735       .          .     
READ:  
READ:      .       10.610    8.201    7.945    8.764    8.221       .     
READ:  
READ:   8.235    8.646    8.507    8.449    8.033    8.241    8.209  
READ:  
READ:   8.345    8.096    8.015  [ 8.008]   8.645    7.896    7.506  
READ:  
READ:   8.260    8.284    8.395    2.419    8.027    8.114    7.633  
READ:  
SENT: M105
READ:      .       8.059    8.310    8.290    8.212    2.116       .     
READ:  
READ:      .          .       7.693    7.844    8.098       .          .     
READ: (-135,-135)                                            (135,-135)
READ: (0,0)                                             (6,0)

Roxy-3D, 2017-11-07:
Please turn on the #define DEBUG_LEVELING_FEATURE

It is going to be a little hard for me to help you because you have a Delta.    But UBL uses the standard probing sub-system.   So things can only be getting so crazy.

cdedwards, 2017-11-07:
READ: <<< do_blocking_move_to
READ:   current_position=(0.00, 0.00, 10.07) : sync_plan_position_kinematic
READ:   current_position=(0.00, 0.00, 10.07) : <<< do_probe_move
READ: 1st Probe Z:10.07
READ: >>> do_blocking_move_to(0.00, 0.00, 15.07)
READ:   destination=(0.00, 0.00, 10.07) : set_destination_from_current
READ:   destination=(0.00, 0.00, 15.07) : prepare_uninterpolated_move_to_destination
READ:   current_position=(0.00, 0.00, 15.07) : z raise move
READ:   current_position=(0.00, 0.00, 15.07) : xy move
READ: <<< do_blocking_move_to
READ:   current_position=(0.00, 0.00, 15.07) : >>> do_probe_move
READ: >>> do_blocking_move_to(0.00, 0.00, -10.00)
READ:   destination=(0.00, 0.00, 15.07) : set_destination_from_current
READ:   current_position=(0.00, 0.00, 15.07) : xy move
READ:   destination=(0.00, 0.00, -10.00) : prepare_uninterpolated_move_to_destination
READ:   current_position=(0.00, 0.00, -10.00) : z lower move
SENT: M105
READ: <<< do_blocking_move_to
READ:   current_position=(0.00, 0.00, 10.18) : sync_plan_position_kinematic
READ:   current_position=(0.00, 0.00, 10.18) : <<< do_probe_move
READ:   current_position=(0.00, 0.00, 10.18) : <<< run_z_probe
READ: 2nd Probe Z:10.18 Discrepancy:-0.11
READ: >>> do_blocking_move_to(0.00, 0.00, 15.18)
READ:   destination=(0.00, 0.00, 10.18) : set_destination_from_current
READ:   destination=(0.00, 0.00, 15.18) : prepare_uninterpolated_move_to_destination
READ:   current_position=(0.00, 0.00, 15.18) : z raise move
READ:   current_position=(0.00, 0.00, 15.18) : xy move
READ: <<< do_blocking_move_to
READ: <<< probe_pt
READ: >>> probe_pt(-45.00, 0.00, no stow)
READ:   current_position=(0.00, 0.00, 15.18) : 
READ: >>> do_blocking_move_to(-45.00, 0.00, 15.18)
READ:   destination=(0.00, 0.00, 15.18) : set_destination_from_current
READ:   current_position=(-45.00, 0.00, 15.18) : xy move
READ: <<< do_blocking_move_to
READ:   current_position=(-45.00, 0.00, 15.18) : set_probe_deployed
READ: deploy: 1
READ:   current_position=(-45.00, 0.00, 15.18) : >>> run_z_probe
READ:   current_position=(-45.00, 0.00, 15.18) : >>> do_probe_move
READ: >>> do_blocking_move_to(-45.00, 0.00, -10.00)
READ:   destination=(-45.00, 0.00, 15.18) : set_destination_from_current
READ:   current_position=(-45.00, 0.00, 15.18) : xy move
READ:   destination=(-45.00, 0.00, -10.00) : prepare_uninterpolated_move_to_destination
READ:   current_position=(-45.00, 0.00, -10.00) : z lower move
READ: <<< do_blocking_move_to
READ:   current_position=(-45.00, 0.00, 9.95) : sync_plan_position_kinematic
READ:   current_position=(-45.00, 0.00, 9.95) : <<< do_probe_move
READ: 1st Probe Z:9.95
READ: >>> do_blocking_move_to(-45.00, 0.00, 14.95)
READ:   destination=(-45.00, 0.00, 9.95) : set_destination_from_current
READ:   destination=(-45.00, 0.00, 14.95) : prepare_uninterpolated_move_to_destination
READ:   current_position=(-45.00, 0.00, 14.95) : z raise move
READ:   current_position=(-45.00, 0.00, 14.95) : xy move
READ: echo:busy: processing
busy: processing
READ: <<< do_blocking_move_to
READ:   current_position=(-45.00, 0.00, 14.95) : >>> do_probe_move
READ: >>> do_blocking_move_to(-45.00, 0.00, -10.00)
READ:   destination=(-45.00, 0.00, 14.95) : set_destination_from_current
READ:   current_position=(-45.00, 0.00, 14.95) : xy move
READ:   destination=(-45.00, 0.00, -10.00) : prepare_uninterpolated_move_to_destination
READ:   current_position=(-45.00, 0.00, -10.00) : z lower move
READ: <<< do_blocking_move_to
READ:   current_position=(-45.00, 0.00, 9.99) : sync_plan_position_kinematic
READ:   current_position=(-45.00, 0.00, 9.99) : <<< do_probe_move
READ:   current_position=(-45.00, 0.00, 9.99) : <<< run_z_probe
READ: 2nd Probe Z:9.99 Discrepancy:-0.04
READ: >>> do_blocking_move_to(-45.00, 0.00, 14.99)
READ:   destination=(-45.00, 0.00, 9.99) : set_destination_from_current
READ:   destination=(-45.00, 0.00, 14.99) : prepare_uninterpolated_move_to_destination
READ:   current_position=(-45.00, 0.00, 14.99) : z raise move
READ:   current_position=(-45.00, 0.00, 14.99) : xy move
READ: <<< do_blocking_move_to
READ: <<< probe_pt
READ: >>> probe_pt(0.00, -45.00, no stow)
READ:   current_position=(-45.00, 0.00, 14.99) : 
READ: >>> do_blocking_move_to(0.00, -45.00, 14.99)
READ:   destination=(-45.00, 0.00, 14.99) : set_destination_from_current
SENT: M112

I only let it run for a couple of points. I have double probe on.

cdedwards, 2017-11-07:
READ:   current_position=(90.00, -90.00, -10.00) : z lower move
READ: <<< do_blocking_move_to
READ:   current_position=(90.00, -90.00, 4.23) : sync_plan_position_kinematic
READ:   current_position=(90.00, -90.00, 4.23) : <<< do_probe_move
READ:   current_position=(90.00, -90.00, 4.23) : <<< run_z_probe
READ: 2nd Probe Z:4.23 Discrepancy:-0.08
READ: >>> do_blocking_move_to(90.00, -90.00, 9.23)
READ:   destination=(90.00, -90.00, 4.23) : set_destination_from_current
READ:   destination=(90.00, -90.00, 9.23) : prepare_uninterpolated_move_to_destination
READ:   current_position=(90.00, -90.00, 9.23) : z raise move
READ:   current_position=(90.00, -90.00, 9.23) : xy move
READ: <<< do_blocking_move_to
READ: <<< probe_pt
READ: >>> probe_pt(135.00, 0.00, no stow)
READ:   current_position=(90.00, -90.00, 9.23) : 
READ: >>> do_blocking_move_to(135.00, 0.00, 9.23)
READ:   destination=(90.00, -90.00, 9.23) : set_destination_from_current
READ:   current_position=(135.00, 0.00, 9.23) : xy move
READ: <<< do_blocking_move_to
READ:   current_position=(135.00, 0.00, 9.23) : set_probe_deployed
READ: deploy: 1
READ:   current_position=(135.00, 0.00, 9.23) : >>> run_z_probe
READ:   current_position=(135.00, 0.00, 9.23) : >>> do_probe_move
READ: >>> do_blocking_move_to(135.00, 0.00, -10.00)
READ:   destination=(135.00, 0.00, 9.23) : set_destination_from_current
READ:   current_position=(135.00, 0.00, 9.23) : xy move
READ:   destination=(135.00, 0.00, -10.00) : prepare_uninterpolated_move_to_destination
READ:   current_position=(135.00, 0.00, -10.00) : z lower move
READ: <<< do_blocking_move_to
READ:   current_position=(135.00, 0.00, 9.19) : sync_plan_position_kinematic
READ:   current_position=(135.00, 0.00, 9.19) : <<< do_probe_move
READ: 1st Probe Z:9.19
READ: >>> do_blocking_move_to(135.00, 0.00, 14.19)
READ:   destination=(135.00, 0.00, 9.19) : set_destination_from_current
READ:   destination=(135.00, 0.00, 14.19) : prepare_uninterpolated_move_to_destination
READ:   current_position=(135.00, 0.00, 14.19) : z raise move
READ:   current_position=(135.00, 0.00, 14.19) : xy move
READ: <<< do_blocking_move_to
READ:   current_position=(135.00, 0.00, 14.19) : >>> do_probe_move
READ: >>> do_blocking_move_to(135.00, 0.00, -10.00)
READ:   destination=(135.00, 0.00, 14.19) : set_destination_from_current
READ:   current_position=(135.00, 0.00, 14.19) : xy move
READ:   destination=(135.00, 0.00, -10.00) : prepare_uninterpolated_move_to_destination
READ:   current_position=(135.00, 0.00, -10.00) : z lower move
READ: echo:busy: processing
busy: processing
READ: <<< do_blocking_move_to
READ:   current_position=(135.00, 0.00, 9.63) : sync_plan_position_kinematic
READ:   current_position=(135.00, 0.00, 9.63) : <<< do_probe_move
READ:   current_position=(135.00, 0.00, 9.63) : <<< run_z_probe
READ: 2nd Probe Z:9.63 Discrepancy:-0.44
READ: >>> do_blocking_move_to(135.00, 0.00, 14.63)
READ:   destination=(135.00, 0.00, 9.63) : set_destination_from_current
READ:   destination=(135.00, 0.00, 14.63) : prepare_uninterpolated_move_to_destination
READ:   current_position=(135.00, 0.00, 14.63) : z raise move
READ:   current_position=(135.00, 0.00, 14.63) : xy move
READ: <<< do_blocking_move_to
READ: <<< probe_pt
READ: >>> probe_pt(135.00, -45.00, no stow)
READ:   current_position=(135.00, 0.00, 14.63) : 
READ: >>> do_blocking_move_to(135.00, -45.00, 14.63)
READ:   destination=(135.00, 0.00, 14.63) : set_destination_from_current
READ:   current_position=(135.00, -45.00, 14.63) : xy move
READ: <<< do_blocking_move_to
READ:   current_position=(135.00, -45.00, 14.63) : set_probe_deployed
READ: deploy: 1
READ:   current_position=(135.00, -45.00, 14.63) : >>> run_z_probe
READ:   current_position=(135.00, -45.00, 14.63) : >>> do_probe_move
READ: >>> do_blocking_move_to(135.00, -45.00, -10.00)
READ:   destination=(135.00, -45.00, 14.63) : set_destination_from_current
[... 331 lines cut ...]
READ:   current_position=(45.00, -135.00, 15.09) : xy move
READ:   destination=(45.00, -135.00, -10.00) : prepare_uninterpolated_move_to_destination
READ:   current_position=(45.00, -135.00, -10.00) : z lower move
READ: <<< do_blocking_move_to
READ:   current_position=(45.00, -135.00, 10.28) : sync_plan_position_kinematic
READ:   current_position=(45.00, -135.00, 10.28) : <<< do_probe_move
READ:   current_position=(45.00, -135.00, 10.28) : <<< run_z_probe
READ: 2nd Probe Z:10.28 Discrepancy:-0.19
READ: >>> do_blocking_move_to(45.00, -135.00, 15.28)
READ:   destination=(45.00, -135.00, 10.28) : set_destination_from_current
READ:   destination=(45.00, -135.00, 15.28) : prepare_uninterpolated_move_to_destination
READ:   current_position=(45.00, -135.00, 15.28) : z raise move
READ:   current_position=(45.00, -135.00, 15.28) : xy move
READ: <<< do_blocking_move_to
READ: <<< probe_pt
READ:   current_position=(45.00, -135.00, 15.28) : set_probe_deployed
READ: deploy: 0
READ: do_probe_raise(5.00)
READ: >>> do_blocking_move_to(45.00, -135.00, 15.28)
READ:   destination=(45.00, -135.00, 15.28) : set_destination_from_current
READ:   current_position=(45.00, -135.00, 15.28) : xy move
READ: <<< do_blocking_move_to
READ: >>> do_blocking_move_to(0.00, 0.00, 15.28)
READ:   destination=(45.00, -135.00, 15.28) : set_destination_from_current
READ:   current_position=(0.00, 0.00, 15.28) : xy move

Roxy-3D, 2017-11-07:
It sure looks like the probing system thinks the Z-Probe is firing at +10 mm from the bed.

- is the nozzle at 15mm above the bed when the probe starts?   That is what the probing system thinks is happening.
- does the probe move down 5 mm before it 'triggers' ?  That is what the probing system thinks is happening.

I wonder if your Z-Probe is always triggered?   Can you please home the machine and do a M119 and see what it says about the Z-Probe.

cdedwards, 2017-11-07:
Z probe is off. it's an IR sensor and will trigger with my finger. I can tell by the LED on it.

Nozzle is homed when I start so yes it's 354.44mm above the bed. 
from 15mm it moves down 5mm and then bounces back up to 15mm. occasionally it drops down further than 5mm. which gives the weird spots on the bed.

Roxy-3D, 2017-11-07:
With the nozzle 15mm above the bed is the Z-Probe triggered?  What does M119 say?

With the nozzle 5mm above the bed is the Z-Probe triggered?  What does M119 say?

It looks from the debug information that the firmware thinks the probe is triggered well before the nozzle gets to 0.0 mm.

cdedwards, 2017-11-07:
READ: X:0.00 Y:0.00 Z:15.00 E:0.00 Count X:52446 Y:52446 Z:52446
READ: Reporting endstop status
READ: x_max: open
READ: y_max: open
READ: z_min: open
READ: z_max: open

SENT: M119
READ: Reporting endstop status
READ: x_max: open
READ: y_max: open
READ: z_min: open
READ: z_max: open
READ: X:0.00 Y:0.00 Z:5.00 E:0.00 Count X:50846 Y:50846 Z:50846

I did a G28 to home. then G1Z15 and then M114/119   
then I did a G1Z5 with finally a M114/119

The probe does not trigger at during the G29/G33. It's like Marlin is making a best guess

cdedwards, 2017-11-07:
It's really cool when you hit a limit I guess you could call it and instead of stopping or giving an error, it reverses direction. I'm giving a G91G1Z1G90  and it should be dropping down to the bed. instead it gets to a certain point and starts rising instead 👎

Roxy-3D, 2017-11-07:
Is the nozzle in the center of the bed when this happens?   There used to be a problem where if the nozzle went outside of the Delta_Probable_Radius, it would go up to the Z-Max.    I don't think that can happen any more.   But the fact you are seeing it reverse direction and go up makes me wonder.

cdedwards, 2017-11-07:
it's at the center of the bed.. it starts at the first point, and continues from there

Tannoo, 2017-11-07:
Did the bed surface change? I was going to use an ir sensor, but couldn't get reliable positions from it.

cdedwards, 2017-11-07:
umm problem is not the sensor.  sensor is not even triggering at all. I've been getting very good results with my DC42 IR sensor. std dev was 0.09 or so over 40 taps on the bed according to M48

Roxy-3D, 2017-11-07:
@thinkyhead   Can you take a look at his DEBUG_LEVELING traces?   I'm sure what ever is wrong is something small.   But it isn't obvious to me what is causing this.

thinkyhead, 2017-11-07:
@cdedwards It's hard to tell from the log output exactly what's going on, but with a little extra detail we might have a better clue.

Please grab the latest `bugfix-1.1.x` and with `DEBUG_LEVELING_FEATURE` enabled do `M111 S247` before starting to enable the most detailed logging possible. Then do a `G28` followed by a `G29` again and post the log output of both of those commands here.

Thanks!

Roxy-3D, 2017-11-07:
This has some similarity to https://github.com/MarlinFirmware/Marlin/issues/8298

tcm0116, 2017-11-07:
@Roxy-3D yes, I think it's the same issue. I submitted a PR a while ago that changed how probing works because I broke some arms on my Delta when I had `DELTA_HEIGHT` set too large. When I was testing that PR, I verified that all sorts of different starting height values worked, so something changed between then and now that's causing the home offset for Z (which adjusts the delta height) to not be used during probing operations.

thinkyhead, 2017-11-07:
> that's causing the home offset for Z (which adjusts the delta height)

One of these days that needs to be separated from `home_offset`, or given another name under `DELTA` anyway. I'll check to see where that got lost — most likely as part of the native workspace patch, in which some hiccups were expected.

thinkyhead, 2017-11-07:
I see the potential issue, but I'm not entirely sure why `home_offset[Z_AXIS]` matters, so I could use a primer on how it works in the DELTA universe.

I'll state my poor understanding of it first:

When you home a Delta the `home_offset[Z_AXIS]` should be added to the current Z so that when you move the nozzle down to touch the bed the current Z is 0.0. This is distinct from `delta_endstop_adj` which is just a tweak for the endstops.

I assume I'm mistaken about how `home_offset` is applied in Delta world, since it seems to matter that it get added to Z within some UBL functions.

The macros such as `LOGICAL_Z_POSITION` were needed before now to deal with workspace offsets, whether on Cartesian or Delta. If `home_offset[Z_AXIS]` on Delta is being used to offset the `current_position` in any way then it would tend to conflict with the native workspace offset idea, in which the machine's physical position belongs in `current_position`, and the `home_offset` should only be added when printing out the current position.

tcm0116, 2017-11-07:
The way it's supposed to work is that `DELTA_HEIGHT` represents the nominal design Z travel limit of the printer. `G33 P1` then finds the actual Z travel limits (i.e. delta height) and uses `home_offset[Z-AXIS]` to represent the difference between `DELTA_HEIGHT` and the measured delta height. As such, the actual delta height is then represented by `DELTA_HEIGHT + home_offset[Z-AXIS]` (although I'm not 100% certain if they're added or subtracted).

I agree that using `home_offset[Z-AXIS]` is not the best way to represent the actual delta height. It would be a lot easier to follow if there was just a `delta_height` variable that gets initialized to `DELTA_HEIGHT` and then updated by `G33`.

Roxy-3D, 2017-11-07:
> I assume I'm mistaken about how home_offset is applied in Delta world, since it seems to matter that it get added to Z within some UBL functions.

I just did a quick search...  And I don't see home_offset[] being referenced anywhere within the UBL code.    It maybe this issue is in the 'support' layers below UBL.

cdedwards, 2017-11-07:
[LOG.TXT](https://github.com/MarlinFirmware/Marlin/files/1451714/LOG.TXT)

When I did the M111 S247 all debugging is turned on. I then M928 log.txt and then G28. M29 to turn off logging. when I looked in the log file, there was pretty much nothing in it. So I cut and pasted from the simplify3d log into that file. you have a G28/G33 P1 sequence. G29 was not working properly at all.. I haven't been using bugfix-1.1.x at all so I had to recreate the config file.
config files are actually a .rar not a .zip.
[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/1451729/Marlin.zip)

cdedwards, 2017-11-07:
oh and I tried M111 S247 M111 S0 to see if it was creating a logfile on it's own, but no such luck

cdedwards, 2017-11-07:
if (strstr_P(command, PSTR("M29"))) {
          // M29 closes the file
          card.closefile();
guess that won't work well.

thinkyhead, 2017-11-07:
> And I don't see `home_offset[]` being referenced anywhere within the UBL code.

The `home_offset[]` used to be referenced implicitly via `RAW_n_POSITION` or `LOGICAL_n_POSITION`.

Now that the internal `current_position` high level coordinate space no longer includes any offset, it may be that the `home_offset[Z_AXIS]` needs to be subtracted from some specified Z coordinate to get the machine native Z position.

See if any of these macro references in https://github.com/MarlinFirmware/Marlin/pull/8229/files stand out as being more important than the others.

thinkyhead, 2017-11-07:
> M928 log.txt

No log files are created by Marlin. The log goes to your host console.

thinkyhead, 2017-11-07:
>  I haven't been using bugfix-1.1.x at all so I had to recreate the config file.

Ah… Well, if you're using Marlin 1.1.6 or earlier then the workspace changes I was concerned about may not matter.

Anyway, please always test with `bugfix-1.1.x` when reporting bugs because there's always a chance we already fixed the issue.

thinkyhead, 2017-11-07:
@tcm0116 When this came up I went back to have a look at your delta safety patch, and the value 10mm does show up there. But is it significant?

https://github.com/MarlinFirmware/Marlin/commit/f54e0fc90f5bc12d665c5d877b73c95bf76e582f#diff-1cb08de130a6ece2d1b5b9c37bcfef48R3770

tcm0116, 2017-11-07:
>Now that the internal current_position high level coordinate space no longer includes any offset, it may be that the home_offset[Z_AXIS] needs to be subtracted from some specified Z coordinate to get the machine native Z position.

That may be the issue. In a Delta machine, the native Z position after homing is defined as `DELTA_HEIGHT + home_offset[Z_AXIS]`, which should cause the nozzle to just touch the bed when you command a movement to Z = 0. When I get home, I can check to see if the delta height adjustment is utilized in non-probing situations.

thinkyhead, 2017-11-07:
If needed, we can replace `home_offset[Z_AXIS]` with a single new variable applicable just to Delta.

To review…

The `home_offset` has always been an interesting feature, simply acting as an additional offset to the workspace. So, if you had `home_offset[X_AXIS]` set to 10, then when you homed, `current_position[X_AXIS]` would actually be set to 10. With the change to native machine space, the `current_position[X_AXIS]` is set to the `X_MIN_POS` (or manual endstop position) with no additional offset. The `home_offset[X_AXIS]` is added onto the output value when you do `M114`, and it is subtracted from the `Xnnn` value immediately with commands like `G1`.

thinkyhead, 2017-11-07:
> That may be the issue.

Considering that @cdedwards is not using the newest code, it's doubtful.

----

If needed, we can replace `home_offset[Z_AXIS]` with a single new variable applicable just to Delta.

To review…

The `home_offset` has always been an interesting feature, simply acting as an additional offset to the workspace. So, if you had `home_offset[X_AXIS]` set to 10, then when you homed, `current_position[X_AXIS]` would actually be set to `X_MIN_POS + 10`.

With the change to native machine space, the `current_position[X_AXIS]` is set to the `X_MIN_POS` with no additional offset. The `home_offset[X_AXIS]` is added onto the output value when you do `M114`, and it is subtracted from the `Xnnn` value immediately with commands like `G1`.

tcm0116, 2017-11-07:
>I went back to have a look at your delta safety patch, and the value 10mm does show up there. But is it significant?

The 10mm value is somewhat arbitrary in that it's a small enough number that if the probe doesn't actually trigger before the nozzle hits the bed (or the home switches), it won't push the nozzle too far into the bed, which could damage the arms. Prior to that change, it would move down to `-(Z_MAX_LENGTH) - 10`, which will most certainly cause damage to a delta printer (ask me how I know).

cdedwards, 2017-11-07:
I was using the latest code. I downgraded to bugfix-1.1.x as per your request.

tcm0116, 2017-11-07:
@thinkyhead you should pickup one of these lipstick Delta printers. It has
a 32-bit controller board, and it's fun to play around with.

Pre-sell BIQU 3D PRINTER BIQU Magician Newest assembled Mini Delta Kossel
Desktop Printer for School/Office/Kids For 4 Colors
 http://s.aliexpress.com/aUfq2U7z
(from AliExpress Android)

thinkyhead, 2017-11-07:
> I downgraded to bugfix-1.1.x as per your request.

Downgraded? Note: `bugfix-1.1.x` is the latest code.

thinkyhead, 2017-11-07:
> you should pickup one of these lipstick Delta printers.

Hmm, I might. If I wait for 11/11 I can save 8 bucks!

thinkyhead, 2017-11-07:
> I downgraded to bugfix-1.1.x as per your request.

Downgraded? Note: `bugfix-1.1.x` is the latest code, with additional patches applied today.

Looking over your LOG.TXT now…

thinkyhead, 2017-11-08:
I see a couple of `G28` invocations in the LOG.TXT, plus a `G33`, but I don't see a `G29`. So far the `G33` log output is inconclusive. I'll look above at the previous `G29` output and see if it adds any clues.

thinkyhead, 2017-11-08:
It seems you have an inductive probe. At what Z height does the inductive probe trigger? From your configuration it appears that it's supposed to trigger when the nozzle is 2.18mm above the bed. Is that correct?

tcm0116, 2017-11-08:
@thinkyhead I just did some testing with my Delta machine on the bugfix-2.0.x as of 10 minutes ago, and I have a few things to report. First, running the Set Delta Height command (`G33 P1`) does appear to offset the reported Z position as expected. For example, my hard-coded `DELTA_HEIGHT` is 165, and prior to running Set Delta Height, the post-home position is around 142. After running Set Delta Height, the reported height is 167.5, and the post-home position is around 147.

However, there is an issue. After running Set Delta Height, I can't get the nozzle down to the bed. Move Z in the LCD lets me command it down to 0.0, but it will actually only go down to about 2.5. If I then command the Z position to 10.0, the LCD shows that it's at 12.5. Based on this, it looks like the `home_offset[Z_AXIS]` is being accounted for in the reporting of the current position, but not in the actual positioning.

cdedwards, 2017-11-08:
I've been unable to complete a G29 P1 with 7 points. probes a few points(10mm above bed) and then fails with a probe error while bouncing up and down at some random point. I have a IR sensor from DC42. and yes, it triggers at 2.18mm above the bed. I'ver verified that the probe actually triggers and appears in M119.

As for the log.txt, G33 P1 does exactly what G29 P1 does in that it probes the bed and thinks it's 10mm higher than it should be.

tcm0116, 2017-11-09:
@cdedwards I just submitted #8345, which may fix your issue. If you want to give it a try, the easiest way would be to pull down the following branch from my repository:

https://github.com/tcm0116/Marlin/tree/1.1.x-delta_height

After you get it loaded, I'd recommended running `G33 P1` first to set your delta height, and then run `G33` to perform the full calibration. You can then try running `G29 P1` to see if UBL is able to successfully probe the bed.

Roxy-3D, 2017-11-09:
I have my fingers crossed!!!!

cdedwards, 2017-11-09:
well here's the thing.. this afternoon I played around some more. No matter what I did, 346mm was the height of my printer. M665 H380, G33 P1 would return 346. It got to the point I was suspecting hardware issue of some sort. This was happening with bugfix-1.1.x and 2.0.x. I loaded Repetier, and then returned to Marlin (don't much like Repetier). I decided todo a print, since 354.44 works fine for printing. After printing out a device, I tried another G33 P1 just for fun. and it worked. G29 P1, same thing. This has had me stumped for 2 days and I have NO idea why it cured itself.

On another note, my delta bed is accurate to 0.003 inches on each home switch. My delta radius is accurate to 0.005inches in 115mm of movement. today I printed out a dial indicator holder which replaces my effector so I could play with accuracy a little more. Tomorrow I'll see about setting the delta height and fixing the delta radius.

closed 2017-11-09 by cdedwards