MarlinFirmware/Marlin #8298: G33 does not use delta height after first probe

tcm0116, 2017-11-07:
Using bugfix-2.0.x as of an hour ago. When I run G33 with `DELTA_HEIGHT` set to a value that is more than 10mm less than the actual height, the first probe is successful, but subsequent probes don't take into account the adjusted delta height and then fail since the bed is more than 10mm away from the starting position of the probe event.

thinkyhead, 2017-11-07:
@LVD-AC — Can you look into this issue? `G33` has changed so often that I don't even look at it any more when merging.

Roxy-3D, 2017-11-07:
This is similar to https://github.com/MarlinFirmware/Marlin/issues/8293

LVD-AC, 2017-11-07:
`    if (do_probe_move(-**10** + (short_move ? 0 : -(Z_MAX_LENGTH)), Z_PROBE_SPEED_SLOW)) return NAN;
`
This is the code that does the probing, so if the probe has to move down more then 10mm below zero something is wrong (like the delta height is set to much off). This is a failsafe in Marlin and to my knowledge always has been this way.

G33 is a tool to calibrate the printer, not a setup tool. One should be able to measure the delta height accurate enough even with a ruler to be precise within 10mm ;)

tcm0116, 2017-11-07:
@LVD-AC Understood. However, I don't understand why the delta height is not used on probe events after the height is ascertained with the first probe. The first probe will actually go more than -10mm to find the bed and succeeds, but the subsequent probes fail.

LVD-AC, 2017-11-07:
Delta height (and the other parameters) can only be calculated and set after all points are probed. But you can always run G33 P1 (1 point calibration - sets height only) on a printer with the height way off, before running a full calibration with more points.

cdedwards, 2017-11-09:
on that note, with G33 should we use G33 P# to probe points and then set our radius, height, towers and endstops and/or use G33 A. Not sure if these two methods are exclusive or not.

LVD-AC, 2017-11-09:
First set radius, height to the specs given by the manufacturer; end-stops and towers should be close enough to zero if your build is half decent. Then run G33 Pn to calibrate these to maximum precision.

G33 A (needs a probe) gives you the optimum parameters for the calibration routine so it iterates fast - a bit similar to PID auto-tune. Without it G33 Pn will do it's job as well, but might just require some more iterations.

The referenced PR's make the first center probe stop at -10 as expected; for one reason or the other at very specific conditions the probe went down forever; so if the switch is broken and doesn't trigger it went on digging into the bed till something breaks. Fixed now.

closed 2017-11-14 by thinkyhead