MarlinFirmware/Marlin #8300: [FR] Different thermal protection

GMagician, 2017-11-07:
Hello,

@thinkyhead, @Roxy-3D, @Bob-the-Kuhn, @AnHardt 

Now thermal protection triggers when, after an amount of time, temperature doesn't increase by a predetermined C° value.
I've done some test with my bed 12V and I see that bigger is the temperature and longer is time to increase...but it will increase.
I'm wondering why not to test if temperature is just increased, without an amount of c°...if temperature is increasing (even slowly) everything is ok...or not?
This new approach can lower timeout (safer) but still give protection.
There is something wrong with this idea? I can proceed if big Marlin devs approve it

AnHardt, 2017-11-07:
We can start an extended discussion here but i'd prefer to make sure you have read #2025 and its follow ups before.

AnHardt, 2017-11-07:
Summary:
One problem is sensor noise,
the other the situation where your heater is very hot, you stop heating, the temperature is falling fast, you switch on the heater again, the temperature keeps failing and needs a good amount of time to reach the temperature where you started heating again.
The tried algorithm worked, but teaching users about the needed parameters and how to tune them seems clothe to impossible. Most users don't get what the currents protections parameter do.

AnHardt, 2017-11-07:
An other promising concept:
Temperature systems are inert. Any measured sudden drop/rise of a temperature is unreal.

GMagician, 2017-11-07:
I read #2025 but I lost myself on all that thread opened on the same argument... I read that that case was related to a particular thermistor were was not possible to detect "detach".
For this thread I was talking about the "reach target" stage not mentioned following one (keep temperature stable).
My discussion is about WATCH_BED_TEMP_PERIOD and WATCH_BED_TEMP_INCREASE (or extruder counterparts)

thinkyhead, 2017-11-07:
> My discussion is about WATCH_BED_TEMP_PERIOD and WATCH_BED_TEMP_INCREASE (or extruder counterparts)

Are you saying you can find no `WATCH_BED_TEMP_PERIOD` and `WATCH_BED_TEMP_INCREASE` values that work? The default values may be overly-conservative, but they can be configured for machines with very slow heaters.

AnHardt, 2017-11-07:
I admit reading #2025 and following all liked threads and PRs and their links is more a task for a complete day - understanding more for a week. Therefore the Summary. :-)
The basic idea for a new algorythm is in https://github.com/MarlinFirmware/Marlin/issues/2025#issuecomment-99823546.
The realisation in #2164.
But most interesting is - in these threads around these complex is said about everything you can know about heaters and their behavior.

Or with other words:
> There is something wrong with this idea?

It is not that simple.

GMagician, 2017-11-07:
>  you saying you can find no WATCH_BED_TEMP_PERIOD and WATCH_BED_TEMP_INCREASE values that work? The default values may be overly-conservative, but they can be configured for machines with very slow heaters."

No no, the only thing I noticed is that, due to 12V power, time to raise WATCH_BED_TEMP_INCREASE is different when target is higher. So to be sure that everything works correctly you need to use larger WATCH_BED_TEMP_PERIOD values (or lower "increase") and this is less protective when working with lower temperatures (eg ABS vs PLA). 
I understand that this is a very long discussion so I don't want to bother all of you more than necessary.
Thanks for answers you gave me...

closed 2017-11-07 by GMagician