MarlinFirmware/Marlin #8301: Sbase: spi_pins.h after sbase_pins.h interference.BUG

karabas, 2017-11-07:
### Bug Report
#define REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER

1 Current spi pins are matched into display pins and cause picture artefacts.
2 Some displays use SPI so these pins must be redefined after  spi_pins.h
3 SDcard on display's cardreader still does not init with any pin compinations I tried.

My spi_pins.h
```
#ifndef SPI_PINS_LPC1768_H
#define SPI_PINS_LPC1768_H

#define LPC_SOFTWARE_SPI

/** onboard SD card */
//#define SCK_PIN           P0_7
//#define MISO_PIN          P0_8
//#define MOSI_PIN          P0_9
//#define SS_PIN            P0_6
/** external */
//#define SCK_PIN           P0_15
//#define MISO_PIN          P0_17
//#define MOSI_PIN          P0_18
//#define SS_PIN            P1_23
//#define SDSS              SS_PIN

#define SCK_PIN           P0_7
#define MISO_PIN          P0_8
#define MOSI_PIN          P0_9
#define SS_PIN            P1_28
#define SDSS              SS_PIN

#if (defined(TARGET_LPC1768) && !(defined(LPC_SOFTWARE_SPI)))   // signal LCDs that they need to use the hardware SPI
  #define SHARED_SPI
#endif
```

thinkyhead, 2017-11-07:
This is a well-known issue with all LPC176x-based controllers, and is the reason that the external SD card included with LCD displays is unsupported in Smoothieware.

closed 2017-11-07 by thinkyhead

karabas, 2017-11-08:
This "well-known issue" does not exist actually. ( Yes I know about some other opinions).

BUT, I have seen personally about 100 delta printers (sold and in production) ) that works on external sdcard perfectly. More over, we does not recommend our users to use USB connection at all and external sdcard is the only user accessible.  
 
MSK Sbase. 
DIsplays are GLDC and RRD Smart controller ( identical card reader).