MarlinFirmware/Marlin #8302: Compilation issues on Delta Kossel with auto calibration and auto bed leveling when MANUAL_PROBE. Head crashes in bed.

MatteoRagni, 2017-11-07:
Marlin is an incredible software. You are really pushing what an arduino can do. Really amazing. Thank you for this great work.

I have an Anycubic Kossel Delta printer, and I'm adapting the `1.1.x` branch for my machine, since the stock firmware (Marlin) has a lot of things to be fixed, and it is an extremely old version (2015).

I don't have a probe, thus I'm configuring manual probing for both:

 * delta calibration
 * automatic bed leveling

I have configured the `AUTO_DELTA_CALIBRATION` to work with the LCD and it works really well. But, I have not found a way to configure the automatic bed leveling, and I'm pretty confident that I'm making some configuration error (no bug). When I start a bed leveling from the LCD, the head homes and waits for my input from the LCD encoder. When I give the click it goes for the first point and crashes without stopping in the bed. I have this problem with **3 points**, **linear** and **bilinear**. I cannot test **mesh** (not supported by delta).

With a linear I got this output from the debugging `M111`:

```
echo:enqueueing "G29 V1"
echo:G29 V1
  current_position=(0.00, 0.00, 265.09) : >>> gcode_G29
Machine Type: Delta
Probe: PROBE_MANUALLY
Auto Bed Leveling: LINEAR (disabled)
G29 Auto Bed Leveling
  current_position=(0.00, 0.00, 265.09) : sync_plan_position_kinematic
  current_position=(0.00, 0.00, 265.09) : setup_for_endstop_or_probe_move
Manual G29 point 1 of 9
echo:busy: processing
echo:busy: processing
echo:busy: processing
Error:KILL caused by KILL button/pin
[ERROR] Error:KILL caused by KILL button/pin
```

What it is strange (and I suspect this is a bug) is what happens when I try to include the `UBL`. I get this error:

```
In file included from /tmp/arduino_build_961156/sketch/MarlinConfig.h:39:0,
                 from /home/nirvana1289/Arduino/Marlin/Marlin.ino:31:
SanityCheck.h:540: error: #error "DELTA_AUTO_CALIBRATION requires a probe: PROBE_MANUALLY, FIX_MOUNTED_PROBE, BLTOUCH, SOLENOID_PROBE, Z_PROBE_ALLEN_KEY, Z_PROBE_SLED, Z Servo."
     #error "DELTA_AUTO_CALIBRATION requires a probe: PROBE_MANUALLY, FIX_MOUNTED_PROBE, BLTOUCH, SOLENOID_PROBE, Z_PROBE_ALLEN_KEY, Z_PROBE_SLED, Z Servo."
      ^
SanityCheck.h:779: error: #error "LCD_BED_LEVELING requires MESH_BED_LEVELING or PROBE_MANUALLY with auto bed leveling enabled."
     #error "LCD_BED_LEVELING requires MESH_BED_LEVELING or PROBE_MANUALLY with auto bed leveling enabled."
```
that disappears **if I disable the `AUTO_DELTA_CALIBRATION`**.

Those are my [basic](https://github.com/MatteoRagni/Marlin.Anycubic.Kossel/blob/master/Configuration.h) and [advanced](https://github.com/MatteoRagni/Marlin.Anycubic.Kossel/blob/master/Configuration_adv.h) configurations.

I'm really sorry if this will be identified only as a mis-configuration.

thinkyhead, 2017-11-07:
`DELTA_AUTO_CALIBRATION` (`G33`) requires a real probe.

For `DELTA` either `AUTO_BED_LEVELING_UBL` or `AUTO_BED_LEVELING_BILINEAR` is recommended. The `LINEAR` variant is not supported.

Please test with the `bugfix-1.1.x` branch for your followup report.

thinkyhead, 2017-11-07:
`DELTA_AUTO_CALIBRATION` (`G33`) doesn't require a real probe, but it is very much in flux and has had limited testing. Coordinate with @LVD-AC to find and fix any bugs.

For `DELTA` either `AUTO_BED_LEVELING_UBL` or `AUTO_BED_LEVELING_BILINEAR` is recommended. The `LINEAR` variant is not supported.

Please test with the `bugfix-1.1.x` branch for your followup report.

LVD-AC, 2017-11-07:
G33 and G29 have been developed separately and have their own routines for manual probing. Good to know the G33 approach works, but not very familiar with G29. :(

In Conditionals_LCD.h : 
```
  /**
   * UBL has its own manual probing, so this just causes trouble.
   */
  #if ENABLED(AUTO_BED_LEVELING_UBL)
    #undef PROBE_MANUALLY
  #endif

I do not know what kind of trouble; but this is the culprit why G33 and G29-UBL do not work together with manual probing.

```

thinkyhead, 2017-11-07:
Sorry about that. Instead of disabling `PROBE_MANUALLY`, there should have been a UBL-is-on condition added for those instances where `PROBE_MANUALLY` shouldn't generate different code. The UBL code itself is agnostic to the existence of `PROBE_MANUALLY`.

thinkyhead, 2017-11-08:
@LVD-AC For now, treat `PROBE_MANUALLY` as a sub-option of `AUTO_BED_LEVELING_3POINT`, `AUTO_BED_LEVELING_LINEAR`, `AUTO_BED_LEVELING_BILINEAR`, which has no meaning outside of those options.

For your `G33` code you should use `HAS_BED_PROBE` instead of `DISABLED(PROBE_MANUALLY)`, and `!HAS_BED_PROBE` instead of `ENABLED(PROBE_MANUALLY)`.

Roxy-3D, 2017-11-08:
Right now...  It is the Delta's that are having problems with probing.   For example: https://github.com/MarlinFirmware/Marlin/issues/8293#issuecomment-342628984   

The big problem is we don't have a lot of developers that have Delta's.   We could use some help getting this problem corralled.

rszasz, 2017-11-08:
Is there a discord or similar for folks needing a bit of help getting things compiled and working?

rszasz, 2017-11-08:
I've created a configuration and configuration_adv that uses values from the release by anycubic, Can I either push this to a branch, or just post the files under main issues?

MatteoRagni, 2017-11-08:
@rszasz I had to do some minor modifications in other part of the firmware, for example in `pins_RAMPS.h` and in lcd, to have it working. Are you able to handle that only with configuration and configuration_adv?

rszasz, 2017-11-08:
Things seemed to work, but the configuration variables weren't doing what I expected (changing rod length didn't flatten the curvature, height could be adjusted so Z0 worked at the center, but the  value was way off the true height). I was tweaking values, compiling, loading and repeating so there is some chance the small testprint (2x2x2 cm) that worked was using a different version. I'll try and test later today.

MatteoRagni, 2017-11-08:
I have merged the `bugfix-1.1.x` branch. I still have some compilation errors.

**CASE 1 - `AUTO_BED_LEVELING_BILINEAR`**

With the [configuration as here](https://github.com/MatteoRagni/Anycubic.Kossel.Marlin/blob/b92b2c08eeb865f56e107106b44631ebce078d00/Marlin/Configuration.h) I got the following compilation error:
```
/tmp/arduino_build_53093/sketch/Marlin_main.cpp: In function 'void update_software_endstops(AxisEnum)':
Marlin_main.cpp:1467: error: 'soft_endstop_radius' was not declared in this scope
           soft_endstop_radius = MIN3(FABS(max(soft_endstop_min[X_AXIS], soft_endstop_min[Y_AXIS])), soft_endstop_max[X_AXIS], soft_endstop_max[Y_AXIS]);
           ^
Marlin_main.cpp:1468: error: 'soft_endstop_radius_2' was not declared in this scope
           soft_endstop_radius_2 = sq(soft_endstop_radius);
```
the problematic variable is [declared at line 485](https://github.com/MarlinFirmware/Marlin/blob/bugfix-1.1.x/Marlin/Marlin_main.cpp#L485):
```
// Software Endstops are based on the configured limits.
float soft_endstop_min[XYZ] = { X_MIN_BED, Y_MIN_BED, Z_MIN_POS },
      soft_endstop_max[XYZ] = { X_MAX_BED, Y_MAX_BED, Z_MAX_POS };
#if HAS_SOFTWARE_ENDSTOPS
  bool soft_endstops_enabled = true;
  #if IS_KINEMATIC
    float soft_endstop_radius, soft_endstop_radius_2;
  #endif
#endif
```

**CASE 2 - `AUTO_BED_LEVELING_UBL`**

with only `AUTO_BED_LEVELING_UBL` the error I got is:
```
In file included from /tmp/arduino_build_53093/sketch/MarlinConfig.h:39:0,
                 from /home/nirvana1289/Arduino/Marlin/Marlin.ino:31:
SanityCheck.h:814: error: #error "LCD_BED_LEVELING requires MESH_BED_LEVELING or ABL with PROBE_MANUALLY."
     #error "LCD_BED_LEVELING requires MESH_BED_LEVELING or ABL with PROBE_MANUALLY."
      ^
```

MatteoRagni, 2017-11-08:
Mh: maybe related also to this one: #7172

rszasz, 2017-11-09:
I haven't been touching the auto leveling since it crashed the head repeatedly. Things do seem to compile just fine with the configuration I'm using (I have what I'm working on in a public fork). Since I figured out how the M665 and M500 commands work I'm attempting to set the parameters fully manually.

MatteoRagni, 2017-11-14:
@rszasz I'm happy if you are getting it working without auto leveling, but this IS an issue for bed leveling. It is known that works if bed leveling is disabled and delta calibration enabled.

But if both Delta Calibration and Bed Leveling (both with manual probing) are requested in configuration the firmware does not compile anymore.

@thinkyhead I given the feedback with the modification you requested. Unfortunately for some time I will not have time to play around with my delta. Sorry for that.

thinkyhead, 2017-11-16:
> with only AUTO_BED_LEVELING_UBL the error I got 
> "LCD_BED_LEVELING requires MESH_BED_LEVELING or ABL with PROBE_MANUALLY."

Disable `LCD_BED_LEVELING` to get rid of that error.

boelle, 2019-02-19:
@MatteoRagni have you tried latest bugfix 2.0?

MatteoRagni, 2019-02-21:
@boelle I'm sorry I've not. I'm really busy at work right now and it is a couple of month since the last time I turned on my kossel... 😞

Phr3d13, 2019-06-10:
@thinkyhead Another one that needs to be closed? Sorry, I got bored.

MatteoRagni, 2019-06-14:
@Phr3d13 in the next two week I will test the current situation...

boelle, 2019-06-22:
@thinkyhead Another one that needs to be closed? Sorry, I got bored.

boelle, 2019-07-21:
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

boelle, 2019-10-24:
@MatteoRagni what did the test say?

MatteoRagni, 2019-10-26:
I forgot to write the results. Sorry for that. I think it is working as should, even if my head crashes the bed AFTER the calibration. But after some more inspection, I think the results of the calibration are good, but I'm not 100% sure. I think I have a problem with the drivers that overheat. For me the issue can be closed.

I did not try the 2.0 though.

closed 2019-10-26 by MatteoRagni