MarlinFirmware/Marlin #8316: Connect self-made printer with bluetooth?

Hypersparking, 2017-11-08:
Now I'm trying to connect my self-made printer just with bluetooth on my mobile phone. However, my printer can't get rid of the USB wire connected with my printer while controlling it by my phone. Could you tell me how to set serial ports so that I can pull it off?

Hypersparking, 2017-11-08:
Marlin 1.1.5 though

thinkyhead, 2017-11-08:
Presumably you just need to enable `BLUETOOTH`. I've never personally used it, so I'm not sure what else may be required.

Hypersparking, 2017-11-08:
Emmm...However,I've enabled bluetooth in configuration.h. But I still need to plug the wire on my computer otherwise my phone can't control it. Quite puzzled XD.

Tannoo, 2017-11-08:
A bluetooth module of some type is needed for most boards.

Tannoo, 2017-11-08:
I have done it on the Mega2560 / Ramps. The phone only gives basic controls, but it did work for me.

That was a long time ago.

Tannoo, 2017-11-08:
I like the Octopi setup better and can access / control / watch print from any computer / tablet / phone that's on the same wifi network.

Hypersparking, 2017-11-08:
@Tannoo  I do use a special kind of bluetooth module. How did you get it work without the USB cable? I use mega2560/Ramps too. I knew it might can only control the printer basically but I really want to do it.

Tannoo, 2017-11-08:
Well, aux2 on the Ramps has pins for the main serial connection.

Tannoo, 2017-11-08:
I've used that for the bluetooth and an serial based lcd touch screen.

Hypersparking, 2017-11-10:
@Tannoo Not need to modify some parameters in Marlin Firmware?

Tannoo, 2017-11-10:
I don't think anything other than enabling BLUETOOTH.

Hypersparking, 2017-11-13:
Anyway thank you very much!@Tannoo

Tannoo, 2017-11-13:
I haven't found the bluetooth module I used, so I can't test it.

closed 2018-02-09 by thinkyhead