MarlinFirmware/Marlin #8320: concurrent manual movement of axis and extruder not possible

ruggb, 2017-11-08:
Maybe this is a problem in 1b09733
If I try to manually running the extruder:
Before I home all, I must wait till the extruder stops b4 it will execute a home.
If I home first, it is the same. I can't manually move any axis until the extrude movement is finished.

I haven't tried to print, But during a print, extruder and axis move concurrently so why is manual movement different.

Is there a reason for this hangup or should it not operate that way?

AnHardt, 2017-11-08:
I have similar experience.
When i have sent a long move in x, i can't do a a move in y, while x is running.
I i can't do B while A is running.
 
That's no an error. It's a consequence of doing things in a sequence.
I'm quite sure you don't want to move x while it's homing.

MagoKimbra, 2017-11-08:
Th eproble is on prepare_move_to_destination_cartesian in last commit:

```
inline bool prepare_move_to_destination_cartesian() {
    const float fr_scaled = MMS_SCALED(feedrate_mm_s);
      #if HAS_MESH
        if (!planner.leveling_active) {
          line_to_destination(fr_scaled);
          return false;
        }
        #if ENABLED(AUTO_BED_LEVELING_UBL)
          ubl.line_to_destination_cartesian(fr_scaled, active_extruder);  // UBL's motion routine needs to know about all moves,
          return true;                                                    // even purely Z-Axis moves
        #else
          if (current_position[X_AXIS] != destination[X_AXIS] || current_position[Y_AXIS] != destination[Y_AXIS]) {
            #if ENABLED(MESH_BED_LEVELING)
              mesh_line_to_destination(fr_scaled);
            #elif ENABLED(AUTO_BED_LEVELING_BILINEAR)
              bilinear_line_to_destination(fr_scaled);
            #endif
            return true;
          }
          else
            line_to_destination();
        #endif
      #endif // HAS_MESH

    return false;
  }
```

If Hasn't MESH no mevement is applicated!!!

AnHardt, 2017-11-08:
The difference is the same as in:
```
G0 E100
G0 X100 ; a sequence
```
vs.
```
G0 X100 E100 ; a move in X and E at the same time
```

tcm0116, 2017-11-08:
I would say that this is by design. Each individual movement command must finish before the next can start. It's technically feasible to change this behavior for manually initiated movements, but it would require significant re-work of the planner. I don't imagine that's something on anyone's TODO list currently.

ruggb, 2017-11-08:
I am not following you.
@AnHardt 
"_The difference is the same as in:  ............._"
Yes, it does it either way. Is that what you are saying?

@tcm0116 
"_Each individual movement command must finish before the next can start_"
I have a corexy. it is always moving x&y at the same time. It must do that on any printer to print.
I have not tried to print yet.

"_It's technically feasible to change this behavior for manually initiated movements,_"
The situation occurs when I **manually** tell it to extrude and while that is running tell it to home.
It also occurs when I tell it to home then tell it to extrude while it is homing. Hopefully, it doesn't operate that way when it is printing, so why does it when it is manual movement?

If I clk home X, then home Y, then home Z. each one must finish b4 the other starts. Good Idea for Z, but with Quickhome enabled and I clk home all, X & Y home concurrently.

Sounds more like inconsistency then design intent.

It isn't like the buffer is full.

AnHardt, 2017-11-08:
Quickhome may appear more intelligent to you, but is also strictly sequential.
( Simplified - for the real thing look up the code)
Make a diagonal move towards the endstops
Wait for the first triggered endstop.
Mark that axis as homed.
Move the other axis towards the endstop
Wait for the endstop
Mark the remaining axis as homed.

Let's assume the manual sequence
G0 X0 ; go to origin
G0 Y0
G0 X100 ; now paint a square
G0 Y100
G0 X0
G0 Y0

What, in your opinion should be done if the commands are running all at the same time?
Go to origin and stay there?
Or, because you can't send all commands at the same time - run a small circle - then end at origin?
Or will it look like a spiral around your start point, ending at 0,0?
What do you want it to do?

In my opinion, doing it in the given sequence is the only way to get what we intended.

AnHardt, 2017-11-09:
Or is a full featured g0-editor, where you can compose a g0 command with all possible parameters, and then execute it, what you want?
No,  you will not be able to home during that move.

tcm0116, 2017-11-09:
@ruggb the printer can certainly move multiple axes simultaneously as part of a single movement command. However, once a movement command has been submitted, that movement must finish before another movement can begin. In the case of the manually initialed movements via the LCD, each adjustment of an axis is considered a single movement (there is a slight delay in processing the movement to allow for large movements to be dialed in before actually queuing the movement command, but it's not long enough for you to switch the axis being commanded). 

Given that explanation, the quick home option just queues a single movement command to move the X and Y axes to the limit switches simultaneously. 

I hope that clarifies your question.

thinkyhead, 2017-11-09:
Some patches were made to movement today. Hopefully clears up the more glaring issues!

closed 2017-11-09 by thinkyhead

ruggb, 2017-11-09:
so G0 x0 y0 e100 would be a single command executing simultaneous movement - assuming the heat was up.
got it - not sure why I didn't see that. makes perfect sense now.
works in Repetier - obviously, it can't be done on the LCD, but I wasn't using the LCD.
therefore a "normal" home executes 
G0 X0
G0 Y0
G0 Z0
and a quickhome executes
G0 X0 Y0
G0 Z0

thanks guys