MarlinFirmware/Marlin #8323: Latest 2.x branch: fatal error: lib/u8g.h: No such file or directory

aplocher, 2017-11-08:
When I compile the 2.x branch using the default Anet A8 Configuration.h and Configuration_adv.h files I recieve the following error:

```
sketch\src\lcd\dogm\u8g_dev_ssd1306_sh1106_128x64_I2C.cpp:72:21: fatal error: lib/u8g.h: No such file or directory

 #include <lib/u8g.h>
                     ^
compilation terminated.

exit status 1
Error compiling for board Anet V1.0 (Optiboot).

```

benlye, 2017-11-08:
> Build would likely fail with any config file I think.

Yep, fails for me in the Arduino IDE with a RAMPS config as well.  Same error.  Build works in Visual Studio Code with PlatformIO though.

benlye, 2017-11-08:
Seems most likely related to #8262.

Roxy-3D, 2017-11-08:
@Bob-the-Kuhn   I just ran into also...     I think it blows up when a Graphical LCD Panel is selected.

benlye, 2017-11-08:
I tweaked the `#include` directives and got past that failure but now the Arduino IDE tells me that the 'filename or extension is too long'. A separate issue, I guess.

The changes I had to make can be seen here:
https://github.com/MarlinFirmware/Marlin/compare/bugfix-2.0.x...benlye:fix-u8g-includes

I think it's to do with how the IDE loads the U8g library, and the syntax the Arduino IDE expects for an include on the include path (`#include <foo.h>`) vs. the syntax for a file path (`#include "bar.h"`).  There were also a couple of paths which weren't relative, but needed to be.

I'm probably wrong though, I'm really just guessing ;-)

fiveangle, 2017-11-08:
Dupe of #7931

Follow trail for debug fix available from arduino/arduino-builder project (and please close once confirmed - thx)

benlye, 2017-11-08:
@fiveangle, my latest issue was that, and I resolved it by changing the temp path but, as described by the OP, there are other things wrong here (including the `#includes` directives).

Now that I've fixed the path-too-long thing, I get these errors from the compiler:
```
C:\Temp\libraries\U8glib\clib\u8g_dev_ssd1306_128x64.c.o (symbol from plugin): In function `u8g_dev_ssd1306_128x64_fn':

(.text+0x0): multiple definition of `u8g_dev_sh1106_128x64_2x_i2c'

C:\Temp\sketch\src\lcd\dogm\u8g_dev_ssd1306_sh1106_128x64_I2C.cpp.o (symbol from plugin):(.text+0x0): first defined here

c:/program files (x86)/arduino/hardware/tools/avr/bin/../lib/gcc/avr/4.9.2/../../../../avr/bin/ld.exe: Disabling relaxation: it will not work with multiple definitions

C:\Temp\libraries\U8glib\clib\u8g_dev_ssd1306_128x64.c.o (symbol from plugin): In function `u8g_dev_ssd1306_128x64_fn':

(.text+0x0): multiple definition of `u8g_dev_sh1106_128x64_2x_pb'

C:\Temp\sketch\src\lcd\dogm\u8g_dev_ssd1306_sh1106_128x64_I2C.cpp.o (symbol from plugin):(.text+0x0): first defined here

C:\Temp\libraries\U8glib\clib\u8g_dev_ssd1306_128x64.c.o (symbol from plugin): In function `u8g_dev_ssd1306_128x64_fn':

(.text+0x0): multiple definition of `u8g_dev_sh1106_128x64_2x_buf'

C:\Temp\sketch\src\lcd\dogm\u8g_dev_ssd1306_sh1106_128x64_I2C.cpp.o (symbol from plugin):(.text+0x0): first defined here

C:\Temp\libraries\U8glib\clib\u8g_dev_ssd1306_128x64.c.o (symbol from plugin): In function `u8g_dev_ssd1306_128x64_fn':

(.text+0x0): multiple definition of `u8g_dev_ssd1306_128x64_2x_i2c'

C:\Temp\sketch\src\lcd\dogm\u8g_dev_ssd1306_sh1106_128x64_I2C.cpp.o (symbol from plugin):(.text+0x0): first defined here

C:\Temp\libraries\U8glib\clib\u8g_dev_ssd1306_128x64.c.o (symbol from plugin): In function `u8g_dev_ssd1306_128x64_fn':

(.text+0x0): multiple definition of `u8g_dev_ssd1306_128x64_2x_pb'

C:\Temp\sketch\src\lcd\dogm\u8g_dev_ssd1306_sh1106_128x64_I2C.cpp.o (symbol from plugin):(.text+0x0): first defined here

C:\Temp\libraries\U8glib\clib\u8g_dev_ssd1306_128x64.c.o (symbol from plugin): In function `u8g_dev_ssd1306_128x64_fn':

(.text+0x0): multiple definition of `u8g_dev_ssd1306_128x64_2x_buf'

C:\Temp\sketch\src\lcd\dogm\u8g_dev_ssd1306_sh1106_128x64_I2C.cpp.o (symbol from plugin):(.text+0x0): first defined here

collect2.exe: error: ld returned 1 exit status
```

benlye, 2017-11-08:
And the penny dropped - @Bob-the-Kuhn's changes require the U8Glib-HAL library, not the standard U8g library.

Replacing the original U8g library with U8glib-HAL make all those last errors go away and everything compiles cleanly (using my version with the modified `#include` directives; the current bugfix-2.0.x still doesn't work even with U8glib-HAL).

benlye, 2017-11-08:
> And the penny dropped - @Bob-the-Kuhn's changes require the U8Glib-HAL library, not the standard U8g library.
>
> Replacing the original U8g library with U8glib-HAL make all those last errors go away [...]

Nope, I was being an idiot and hadn't enabled an LCD - still a lot of errors.  I'm going to leave this to the experts, sorry if my meddling added any confusion.

Bob-the-Kuhn, 2017-11-08:
> fatal error: lib/u8g.h: No such file or directory

This is a platformio.ini issue.   

The **U8glib-HAL** in the [common] section is supposed to automatically install the needed library.  

Try the following change for your board:
**FROM**
`lib_deps    = ${common.lib_deps}`
**TO**
```
lib_deps    = ${common.lib_deps}
  U8glib-HAL
```

There needs to be **EXACTLY** 2 spaces between the start of the line and U8glib-HAL_ID1932.

This works on my system but then so does the current bugfix.

I'll push this to the code base after I do a few sanity checks.

benlye, 2017-11-08:
@Bob-the-Kuhn it works in PlatformIO for me (in VS Code).  It doesn't work in the **Arduino IDE**.  Is there a way to make hte U8glib-HAL library play nice with the Arduino IDE?

Bob-the-Kuhn, 2017-11-08:
Shoot - I hadn't thought about the Arduino IDE.

Looks like the idea of having just one library doesn't work in all situations.

Back to u8g for Arduino/AVR boards and U8glib-HAL for the others.

fiveangle, 2017-11-08:
> I resolved it by changing the temp path...

Are we confident that w/a resolves ?  There's no guarantee that shortening the temp path from `C:/User/<username>/AppLocal/Arduino/Temp` to `C:/Temp` is enough to resolve all issues as bobc pointed out, their is a sensitivity very close to 32k command length as well.

I too found a seemingly different failure mode after the first debug build of arduino-builder which theoretically shoud have resolved.  However it wasn't until the newer debug build that moves knee point closer to 31k vs 32k that I got a successful compile (albeit with different binary sizes after every build, so still needs resolution, but at least compiled).

Can you confirm installation of the latest debug arduino-builder I linked above does not get you further along ?

benlye, 2017-11-08:
@fiveangle, this is a different problem. The error related to the length of the command line was a red herring. Bob's work on the U8glib doesn't seem to be compatible with the Arduino IDE.

fiveangle, 2017-11-08:
good good

Bob-the-Kuhn, 2017-11-08:
There's several changes that need to be done to resolve the Arduino IDE problem:
- replace `<lib/u8g.h>` with `<clib/u8g.h>` in multiple files
- replace `<src/lcd/dogm/HAL_LCD_com_defines.h>` with` "HAL_LCD_com_defines.h"` in multiple files

Unfortunately that breaks the LPC1768 code.  That fix involves a change to the library which can take up to 24 hours to happen.

Looks like the solution will take multiple steps:
1. Make the LPC1768 platformio.ini change right away
2. Modify U8glib-HAL
3. Wait for the changes to U8glib-HAL to become active
4. Make the Arduino IDE changes.

Bob-the-Kuhn, 2017-11-08:
I've just merged PR #8338 to take care of the Platformio folks.

On to the U8glib-HAL changes.

Bob-the-Kuhn, 2017-11-08:
@thinkyhead 

Scott - I don't have the rights to fork U8glib-HAL so I can't do a PR against it.

Please replace U8glib-HAL with [this.](https://www.dropbox.com/s/fvup7v2b6uemysv/U8glib-HAL%20version%200.5%20%28Arduino%20update%20part%201%20of%202%29.zip?dl=0)

----

Here's some proposed text for the PR.

Arduino IDE compatibility (part 1 of 2)

1) change LIB directory name to CLIB
2) in U8glib point include to new directory name
3) in JSON change version

NOTE - the LIB directory needs to stay around until Bugfix2.0.x is updated.  LIB will be deleted on a follow up PR.

Bob-the-Kuhn, 2017-11-09:
I just ran into a "filename or extension is too long" error when compiling with Arduino.  Is this the 32K limit issue being discussed earlier?

I think I have the issues fixed but ...  Could a couple of you try the following with Arduino & with PlatformIO?  https://www.dropbox.com/s/euw84aace3yqbbz/2.0.x%20compile%20fixes.zip?dl=0

Bob-the-Kuhn, 2017-11-09:
I was able to get Arduino to compile using the beta version.

I've just merged the updates via PR #8341

benlye, 2017-11-09:
Works for me too - error-free compilation in VS Code and Arduino IDE.  Thanks, Bob.

thinkyhead, 2017-11-09:
Sorry for the late reply, @Bob-the-Kuhn — You now have full powers on https://github.com/MarlinFirmware/U8glib-HAL

Bob-the-Kuhn, 2017-11-09:
Great news - others can compile the new code.

Now to see if it can run displays besides VIKI2 & RepRap Discount Full Graphics Smart Controller.

aplocher, 2017-11-10:
Hello sorry for posting this issue and disappearing.  I grabbed the latest as of just now and I'm now receiving this build error (no doubt) related to the previous one:

sketch\src\lcd\dogm\u8g_dev_ssd1306_sh1106_128x64_I2C.cpp:72:20: fatal error: U8glib.h: No such file or directory

 #include <U8glib.h>

                    ^

compilation terminated.

exit status 1
Error compiling for board Anet V1.0 (Optiboot).

benlye, 2017-11-10:
Have you installed the U8glib library using the library manager in the Arduino IDE?

aplocher, 2017-11-10:
Oops sorry nope I didn't.  Just did now.  Building, it's definitely gotten further than last time so I'm thinking it is good.  Will confirm in just a moment...

aplocher, 2017-11-10:
Ok, so now I installed the U8glib stuff and it goes through almost the entire compile (err, "verify") process and then at the end I get this:

```
avr-gcc: error: CreateProcess: No such file or directory

exit status 1
Error compiling for board Anet V1.0 (Optiboot).
```

Some people mention that Arduino IDE may have gotten corrupted, but I loaded up Skynet and an older version of Marlin v2 and they both compiled without issue in the same version of the IDE (1.8.5).  Any ideas?

aplocher, 2017-11-10:
Interestingly, compiling it from my Raspi3 with Arduino IDE returns a flood of build warnings (as opposed to the virtually silent verbosity of my windows box), but it actually seems to succeed:

Ugh I can't paste this in pastebin because of a 512k limit.  I'll just include the top and bottom warnings (its 2.5mb of text for the whole thing)

```
In file included from /home/pi/src/arduino-1.8.5/hardware/arduino/avr/cores/arduino/Arduino.h:257:0,
                 from sketch/Marlin.ino.cpp:1:
/home/pi/src/arduino-1.8.5/hardware/anet/avr/variants/sanguino/pins_arduino.h:43:0: warning: "TIMER2" redefined
 #define TIMER2  5
 ^
In file included from sketch/Marlin.ino.cpp:1:0:
/home/pi/src/arduino-1.8.5/hardware/arduino/avr/cores/arduino/Arduino.h:210:0: note: this is the location of the previous definition
 #define TIMER2  6
 ^
In file included from /home/pi/src/arduino-1.8.5/hardware/arduino/avr/cores/arduino/Arduino.h:257:0,
                 from sketch/Marlin.ino.cpp:1:
/home/pi/src/arduino-1.8.5/hardware/anet/avr/variants/sanguino/pins_arduino.h:44:0: warning: "TIMER2A" redefined
 #define TIMER2A 6
 ^
In file included from sketch/Marlin.ino.cpp:1:0:
/home/pi/src/arduino-1.8.5/hardware/arduino/avr/cores/arduino/Arduino.h:211:0: note: this is the location of the previous definition
 #define TIMER2A 7
 ^
In file included from /home/pi/src/arduino-1.8.5/hardware/arduino/avr/cores/arduino/Arduino.h:257:0,
                 from sketch/Marlin.ino.cpp:1:
/home/pi/src/arduino-1.8.5/hardware/anet/avr/variants/sanguino/pins_arduino.h:45:0: warning: "TIMER2B" redefined
 #define TIMER2B 7
 ^
In file included from sketch/Marlin.ino.cpp:1:0:
/home/pi/src/arduino-1.8.5/hardware/arduino/avr/cores/arduino/Arduino.h:212:0: note: this is the location of the previous definition
 #define TIMER2B 8
 ^
In file included from /home/pi/src/arduino-1.8.5/hardware/arduino/avr/cores/arduino/Arduino.h:257:0,
                 from sketch/Marlin.ino.cpp:1:
/home/pi/src/arduino-1.8.5/hardware/anet/avr/variants/sanguino/pins_arduino.h:47:0: warning: "TIMER3A" redefined
 #define TIMER3A 8
 ^
In file included from sketch/Marlin.ino.cpp:1:0:
/home/pi/src/arduino-1.8.5/hardware/arduino/avr/cores/arduino/Arduino.h:214:0: note: this is the location of the previous definition
 #define TIMER3A 9
 ^
In file included from /home/pi/src/arduino-1.8.5/hardware/arduino/avr/cores/arduino/Arduino.h:257:0,
                 from sketch/Marlin.ino.cpp:1:
/home/pi/src/arduino-1.8.5/hardware/anet/avr/variants/sanguino/pins_arduino.h:48:0: warning: "TIMER3B" redefined
 #define TIMER3B 9
 ^
In file included from sketch/Marlin.ino.cpp:1:0:
/home/pi/src/arduino-1.8.5/hardware/arduino/avr/cores/arduino/Arduino.h:215:0: note: this is the location of the previous definition
 #define TIMER3B 10
 ^
In file included from /home/pi/src/arduino-1.8.5/hardware/arduino/avr/cores/arduino/Arduino.h:257:0,
                 from sketch/Marlin.ino.cpp:1:
/home/pi/src/arduino-1.8.5/hardware/anet/avr/variants/sanguino/pins_arduino.h:49:0: warning: "TIMER3C" redefined
 #define TIMER3C 10
 ^
In file included from sketch/Marlin.ino.cpp:1:0:
/home/pi/src/arduino-1.8.5/hardware/arduino/avr/cores/arduino/Arduino.h:216:0: note: this is the location of the previous definition
 #define TIMER3C 11
 ^
In file included from /home/pi/src/arduino-1.8.5/hardware/arduino/avr/cores/arduino/Arduino.h:257:0,
                 from sketch/Marlin.ino.cpp:1:
/home/pi/src/arduino-1.8.5/hardware/anet/avr/variants/sanguino/pins_arduino.h:50:0: warning: "TIMER4A" redefined
 #define TIMER4A 11
 ^
In file included from sketch/Marlin.ino.cpp:1:0:
/home/pi/src/arduino-1.8.5/hardware/arduino/avr/cores/arduino/Arduino.h:217:0: note: this is the location of the previous definition
 #define TIMER4A 12
 ^
In file included from /home/pi/src/arduino-1.8.5/hardware/arduino/avr/cores/arduino/Arduino.h:257:0,
                 from sketch/Marlin.ino.cpp:1:
[... 170 lines cut ...]
In file included from /home/pi/src/arduino-1.8.5/hardware/arduino/avr/cores/arduino/main.cpp:20:0:
/home/pi/src/arduino-1.8.5/hardware/arduino/avr/cores/arduino/Arduino.h:221:0: note: this is the location of the previous definition
 #define TIMER5A 16
 ^
In file included from /home/pi/src/arduino-1.8.5/hardware/arduino/avr/cores/arduino/Arduino.h:257:0,
                 from /home/pi/src/arduino-1.8.5/hardware/arduino/avr/cores/arduino/main.cpp:20:
/home/pi/src/arduino-1.8.5/hardware/anet/avr/variants/sanguino/pins_arduino.h:54:0: warning: "TIMER5B" redefined
 #define TIMER5B 15
 ^
In file included from /home/pi/src/arduino-1.8.5/hardware/arduino/avr/cores/arduino/main.cpp:20:0:
/home/pi/src/arduino-1.8.5/hardware/arduino/avr/cores/arduino/Arduino.h:222:0: note: this is the location of the previous definition
 #define TIMER5B 17
 ^
In file included from /home/pi/src/arduino-1.8.5/hardware/arduino/avr/cores/arduino/Arduino.h:257:0,
                 from /home/pi/src/arduino-1.8.5/hardware/arduino/avr/cores/arduino/main.cpp:20:
/home/pi/src/arduino-1.8.5/hardware/anet/avr/variants/sanguino/pins_arduino.h:55:0: warning: "TIMER5C" redefined
 #define TIMER5C 16
 ^
In file included from /home/pi/src/arduino-1.8.5/hardware/arduino/avr/cores/arduino/main.cpp:20:0:
/home/pi/src/arduino-1.8.5/hardware/arduino/avr/cores/arduino/Arduino.h:223:0: note: this is the location of the previous definition
 #define TIMER5C 18
 ^
Sketch uses 111638 bytes (85%) of program storage space. Maximum is 130048 bytes.
Global variables use 4305 bytes of dynamic memory
```

benlye, 2017-11-10:
I think your Pi is probably not using the correct compiler flags (I think the missing one is -fpermissive).  Make sure it has the latest [board definition for the Anet board](https://github.com/SkyNet3D/anet-board).

On Windows you're running into the Arduino bug fiveangle mentioned, where the compiler is trying to execute a command more than 32K characters long.

An easy way to try to work around that is to change the build path by opening up the Arduino IDE preferences file and adding `build.path=C:\Temp`.  (Make sure C:\Temp exists).

You need to edit the preferences file while the Arduino IDE is not running.  The file is here:
%localappdata%\Arduino15\preferences.txt

Otherwise, another workaround is to follow the link above and update your Arduino environment to use the bugfix build tools:
https://github.com/arduino/arduino-builder/pull/250

cdedwards, 2017-11-10:
in windows, the command line for compiling get's much too long. so you need to
set C:\temp 
which will move the compile directory and shorten the command line.
right click on My Computer/ThisPC, Properties, Advanced system settings and then Enviroment Variables
add set TEMP=C:\TEMP and set TMP=C:\TEMP
might need to create that directory aka folder though

Bob-the-Kuhn, 2017-11-10:
Another Arduino work around is to use the Beta version of Arduino.  That worked for me.

aplocher, 2017-11-10:
Thanks guys

@benlye not sure why I'm getting those warnings.  I've loaded up Arduino 1.9.0 beta and verified that I'm using the latest anet machine configs:

```
pi@BC-REPRAP-01:~/src/anet-board $ pwd
/home/pi/src/anet-board
pi@BC-REPRAP-01:~/src/anet-board $ git remote show origin
* remote origin
  Fetch URL: https://github.com/SkyNet3D/anet-board.git
  Push  URL: https://github.com/SkyNet3D/anet-board.git
  HEAD branch: master
  Remote branch:
    master tracked
  Local branch configured for 'git pull':
    master merges with remote master
  Local ref configured for 'git push':
    master pushes to master (up to date)
pi@BC-REPRAP-01:~/src/anet-board $ git pull
Already up-to-date.
pi@BC-REPRAP-01:~/src/anet-board $ ls -l ~/Arduino/hardware/
total 0
lrwxrwxrwx 1 pi pi 37 Nov 10 13:59 anet -> /home/pi/src/anet-board/hardware/anet
pi@BC-REPRAP-01:~/src/anet-board $ git diff
pi@BC-REPRAP-01:~/src/anet-board $ 

```

As for the compile flag, I checked the platform.txt in the "anet-board" stuff and it did have it under cpp flags.  Is that the right place to check for it?

```
pi@BC-REPRAP-01:~/src/anet-board/hardware/anet/avr $ grep permis platform.txt 
compiler.cpp.flags=-c -g -Os {compiler.warning_flags} -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -MMD -flto

```

aplocher, 2017-11-10:
@Bob-the-Kuhn and @benlye I loaded up 1.9.0 beta on my Windows box and changed the build path in the preferences.txt to c:\tmp.   Sorry I should have attempted one at a time to pinpoint the actual fix, but I didn't.  Everything is working great now - from Windows at least, and I guess the Pi is just showing me harmless(???) warnings.

I'll go ahead and close this issue out unless you see a reason to keep it open.

closed 2017-11-14 by thinkyhead

fiveangle, 2017-11-14:
Path length issue is supposedly fixed in 1.9 so changing build dir should not be needed.