MarlinFirmware/Marlin #8327: M928 SD write  how to use

NUU-B2-417, 2017-11-08:
I am a college student from Taiwan!
Just started to contact Marin!
I want to use M928 to record to SD card ，But I can not do it!

**How to use M928?** 
I want to record the string to the data in. Txt,，but there is always nothing in. Txt!
Please!help me!

thinkyhead, 2017-11-09:
It's supposed to write all commands that Marlin processes out to the SD file you specify.

```gcode
M928 LOGFILE.TXT
M111 S32 ; Write this command
M117 And this one...
M106 S0 ; me too
M29 ; closes the file
```

If you find that it doesn't work in Marlin 1.1.6 then there might be a bug.

NUU-B2-417, 2017-11-11:
but why mine LOGFILE.TXT wasn't logged~?

thinkyhead, 2017-11-12:
Try switching the position of the little R/W tab on the SD card and see if it still can't write.

tonypan2000, 2018-10-21:
I would like to log data on Marlin with the M928 command. I want this to happen automatically, and I see that M928 is callling `card.openLogFile(parser.string_arg);` in Marlin_main.cpp. I am wondering where in my code I should call openLogFile() and where I should set card.saving to false in Marlin_main if I need to log the printing data. 
Thanks in advance for your help!

thinkyhead, 2018-10-26:
Just use something like `enqueue_and_echo_commands_P(PSTR("M928 auto.log"));` at the point where you want the logging to start.

closed 2018-10-26 by thinkyhead