MarlinFirmware/Marlin #8339: Manual Delta Tuning?

rszasz, 2017-11-08:
Manual automatic tuning for Deltas doesn't seem to be functional right now, so I'm trying to figure out how to manually tune the parameters. 

I see that M665 is supposed to write the parameters to memory, but is it possible to read the values out for verification ?

rszasz, 2017-11-09:
Rebooting while pronterface is connected gave a m665 readout, now I'm just trying to figure out why changing the values in the configuration.h, compiling and uploading doesn't change the stored values. 

(though changing them with the M665 command seems to work)

AnHardt, 2017-11-09:
EEPROM?

rszasz, 2017-11-09:
AnHardt, I think I've started to figure it out. Storing to EEPROM M500 saves the values, and seems to override the hard coded parameters even when a new program is first uploaded. (Which makes sense, I just wasn't expecting it)

I assume the M502 command resets EEPROM parameters to whatever was coded into the configuration.h file?

Is there a way to get values echoed back over serial other than performing a reset?

AnHardt, 2017-11-09:
https://github.com/MarlinFirmware/Marlin/blob/1.1.x/Marlin/Marlin_main.cpp#L197-L200
http://reprap.org/wiki/G-code#M503:_Print_settings
M503

thinkyhead, 2017-11-09:
> I assume the M502 command resets EEPROM parameters to whatever was coded into the configuration.h file?

Yes, but it doesn't save them to EEPROM. For that, you must follow with M500.

LVD-AC, 2017-11-09:
You can view/change them also on the LCD in the delta calibration menu.

closed 2017-11-25 by thinkyhead