MarlinFirmware/Marlin #8346: Not a issue but a question about Mesh Leveling

budi67, 2017-11-09:
Hello all

Sorry for my bad english.
I would want to know how to have operational the mesh leveling.
There is a command to be written in the gcode ?
I have make the mesh with the LCD menu and when i make a G29 S0 i have my grid
[image]
But when i launch a print, the Z don't move

A big thanks

JimBrown, 2017-11-09:
Try adding M420 S1 to your start gcode. http://reprap.org/wiki/G-code#M420:_Leveling_On.2FOff.2FFade_.28Marlin.29

budi67, 2017-11-09:
Thanks JimBrown ;) i will try this when the print is finished.
In the gcode, i have to add G29 or no ?

JimBrown, 2017-11-10:
My understanding is no, you do not add G29 to the start gcode when using manual mesh leveling.

thinkyhead, 2017-11-12:
You should use `M420 S1` in your startup G-code, as this will make sure your saved leveling mesh is used for the print.

budi67, 2017-11-12:
A big thanks, M420 S1 work great !!!

closed 2017-11-17 by thinkyhead