MarlinFirmware/Marlin #8348: [FR] Junction deviation instead of Jerk settings

Sineos, 2017-11-09:
I have been playing with the Smoothieware Firmware on a friend's printer (and I have to say I like Marlin a lot more).
Nevertheless I stumbled upon an interesting setting that I really did like:
http://smoothieware.org/motion-control#junction-deviation

Playing around with this setting on his printer was a real eye-opener and the logic behind is more easily to grasp compared to the Jerk setting.

It also seems to be a common setting for CNC type of machines when going by this information: 
https://github.com/grbl/grbl/wiki/Configuring-Grbl-v0.8#9---junction-deviation-mm

thinkyhead, 2017-11-10:
Looks like a neat alternative to jerk. We'll add it to the things-to-try list.

lenne0815, 2017-11-11:
First i didnt even understand the implications of "jerk" in marlin and thought of it as the physics term jerk. Later on i realised that jerk was in fact an instantaneous change in speed causing all kinds of problems ( ringing, inconsistent stepper torque output etc etc etc ). After being dumbfounded by that ( Coming from mach 3 and cncing in general where something like "jerk" is unheard of ) I started looking for alternatives and luckily enough quickly found one; kevinoconnors klipper firmware does proper lookahead / toolpath prediction on 8 bit boards while being able to pulse up to 150khz.

Sineos, 2017-11-13:
I tried to integrate https://github.com/grbl/grbl/blob/658eb6a8b3cff0da59a61cbb5aa04d2c693c36c3/grbl/planner.c#L356-L402 into Marlin but failed miserable. As a "script kiddie" (not age wise :-P) this is way above my head.

thinkyhead, 2018-05-10:
Junction Deviation is now integrated. See #10650, #10651.

closed 2018-05-10 by thinkyhead

boelle, 2019-01-05:
are there any documentation on how to tune it? i just found the feature today

kAdonis, 2019-01-06:
@boelle No, not that I know
You are not alone
#11599
https://github.com/MarlinFirmware/Marlin/issues/9917#issuecomment-396085333
You could try the Jerk Tower:
https://www.youmagine.com/designs/multi-angle-test-tower-for-calibrating-jerk#documents

boelle, 2019-01-06:
read through the jerk tower explanation and see this:

"Jerk" is called "Junction Deviation" in other firmware and is calculated differently, and so has different setting/values.

spec the part: is calculated differently, and so has different setting/values.

so far a dummie like me that does not help much as he does not explain how to find junction deviation 

if it's as simple as just turning on junction deviation and using the same test method then i get it, but some more detailed explanation could be very usefull

does it even matter in regards to print quality if i use jerk or deviation?

boelle, 2019-01-06:
i think the best bet is to use the jerk tower guide and use values 0.01 to 0.3 and see what is the fastest before print quality suffers

boelle, 2019-01-06:
oki, then i just have to wait until they fix an issue on the latest version where i cant control a 2nd extruder fan, currently i have to run a 2 month old copy where junction deviation is not introduced yet

i use the 2nd extruder fan to control a fan that cools of the bed after print is complete so the PEI releases faster

https://github.com/MarlinFirmware/Marlin/issues/12831

Squid116, 2019-01-06:
@boelle if it is that important to you, you could put a bounty on the documentation ;)

kAdonis, 2019-01-07:
@Squid116 of course, you are right. Z-jerk isn't used
I was confusing Z-jerk with E-jerk