MarlinFirmware/Marlin #8349: [FR] Proper board & pin definitions for Anycubic TriGorilla board

rszasz, 2017-11-09:
I'm working on creating the correct files for my Anycubic Kossel but would like to know the proper way to assign a new number. The firmware from anycubic just bodged on top of the RAMPS definition which is particularly annoying when trying to create clean files for the current versions of Marlin.

 
A picture with pin mapping is available at
http://www.xcsource-pic.com/TE627-E-10-9.jpg

A picture with port descriptions is at
https://www.thingiverse.com/download:4012609

from the manual the extruder heater is on heater0, part cooling fan on fan0 and extruder fan on fan2, Extruder server is on E0, a probe would go on Z- endstop

MatteoRagni, 2017-11-09:
Check out [pins.h](https://github.com/MatteoRagni/Anycubic.Kossel.Marlin/blob/1.1.x-anycubic/Marlin/pins.h#L119) and [pins_ANYCUBIC.h](https://github.com/MatteoRagni/Anycubic.Kossel.Marlin/blob/1.1.x-anycubic/Marlin/pins_ANYCUBIC.h). I have also to correct my [fans settings](https://github.com/MatteoRagni/Anycubic.Kossel.Marlin/blob/1.1.x-anycubic/Marlin/Configuration_adv.h#L214) for extruder though.

The files are in line with `bugfix-1.1.x` commit [`87b655e59`](https://github.com/MarlinFirmware/Marlin/commit/87b655e59e67189faeea395ef3aa51134f22fa39)

MatteoRagni, 2017-11-09:
And also thank you for doing this. You will simplify the game for A LOT of people...

rszasz, 2017-11-11:
At this point I need to get a number assigned for the Trigorilla board, or should I just go with a random 3 digit number that is otherwise unused?

rszasz, 2017-11-11:
https://github.com/rszasz/Marlin/tree/bugfix-1.1.x-trigorilla is compiling, heating, and the hotend fan is running automatically. Now I need to figure out how to command the part cooling fan (fan0) using gcode and this should be good as a base install.

rszasz, 2017-11-11:
Working with LCD driven autoleveling, and a good board definition. 

https://github.com/rszasz/Marlin/tree/bugfix-1.1.x-trigorilla

thinkyhead, 2018-02-09:
Merged.

closed 2018-02-09 by thinkyhead

timmeygasmus, 2018-03-31:
Hello @rszasz , I stumbled upon the fact that in your definition the FAN_PIN is D44, which is on the board labeled as "FAN2". From my understanding (and in standard RAMPS) the default pin for the part cooling fan is FAN_PIN D9 (which is on the Trigorilla labeled as FAN0). Is there a specific reason to handle the pin assignment this way?

rszasz, 2018-03-31:
Looks like a mistake on my part, heating connections go from high to low (base, 1, 0), fan connections from low to high, (0,1,2) and going from the pin diagram I messed up the order.

Thanks for catching that. Looks like Fan 0 should be pin 9

thinkyhead, 2018-03-31:
@rszasz — What change do you recommend we make to [pins_TRIGORILLA.h](https://github.com/MarlinFirmware/Marlin/blob/bugfix-1.1.x/Marlin/pins_TRIGORILLA.h)?

Does there need to be an alternative configuration to distinguish between the original and the 1.3 version of the Trigorilla board?

rszasz, 2018-03-31:
I have to check when I get back to my printer and it finishes the current print.

MatteoRagni, 2018-04-01:
I cannot be sure, but I think there are at least two version for this board: the 1.3, with configuration for pin D44 (like mine), and the 1.4 that I **suspect** (I don't have one) needs as configuration D9.

For my setup, the D44 is the define that makes everything work, that is why I committed it.
But at the end of the story I think it is not a big issue. The Anycubic has fans with the same connectors. It is more than enough to handle it with a define, or tho swap the pins on the board itself.

Apart from this, I'm also starting to think that revision 1.4 is far more common with respect to 1.3.

thinkyhead, 2018-04-02:
Well, no harm in making two entries in `boards.h` :
- `BOARD_TRIGORILLA_13` to replace `BOARD_TRIGORILLA`, and
- `BOARD_TRIGORILLA_14` that would exchange the pins.

rszasz, 2018-04-02:
So 0 is the extruder fan, 2 is the part fan at least for the suggested setup for the anycubic kossel (I think it might have been 2 for the extruder fan 0 for part fan)

I've been traveling so I didn't get a chance to do a proper check of what is connected to which port.

joba-1, 2018-04-02:
Brand new infos on "my" brand new kossel - very good :) 

I have issues with the delivered fw (which seems to be a slightly changed Marlin 1.1.0-RC8).
Autoleveling does not work: I end up printing ~2cm in the air.
So, I'd like to try out newer fw and hope this fixes things or I learn how to fix it along the way.
 
Would you recommend this marlin bugfix branch or the rszasz one right now?
How can I know which board rev (1.3 or 1.4) I have? I guess the newest one...

Any pointers on where to start reading if I want to low level test things are working ok?
I mean like sending g-code commands via serial (if possible). Reading serial output already works.
What would be your recommended forum/mailing list/... for questions regarding this stuff?
I know c/c++ and atmega/esp8266 programming/flashing pretty well, but I'm blank regarding 3d-printing stuff (yet).

thinkyhead, 2018-04-03:
What is `SERVO0_PIN` supposed to be for the TriGorilla 1.4?

thinkyhead, 2018-04-10:
I have a set of changes that split up TriGorilla 1.3 from 1.4, but before I go forward I have a question: What is `SERVO0_PIN` supposed to be for the TriGorilla 1.4?

thinkyhead, 2018-04-12:
Enable `PINS_DEBUGGING` and `M43` will list all pins, along with which are analog pins.

Foxgrove, 2018-04-12:
Beauty!!  Thanks for the prompt help.  👍