MarlinFirmware/Marlin #8350: [FR] Add microstepping control over serial with M192

rooiejoris, 2017-11-09:
Hi,

I have a feature request for the most up to date marlin firmware. Add microstepping control over serial.
Now you can do 'realtime babystepping' with a control panel. But at the time [2011-2012?!?] I suggested bernhard to implement that he also implemented the delayed version with a gcode coment.
Why this is useful: When controlling the machines with eg Octoprint or printrun [and don't have the control panel nearby, with bigger printers] you can do the offsets via the terminal.
Diadvantage is that the offset only appears after the buffer is empty, but I can oversea the disadvantage, no problem.

I made this happen with a marlin version a few moths ago, but now the structure is very different and as a product designer and not a programmer I cannot manage anymore to hack it myself. I will attach the files where the changes are, i really think it is a pretty simple cut and paste if you are more into marlin.

Please let me know if there are any question or if things are not clear, i am happy to explain...!!

[please not that in the attached files the changes are my hacks and most likely not the way it should be programmed, but it works for me...]

cheers / joris

[M192_changes.zip](https://github.com/MarlinFirmware/Marlin/files/1457639/M192_changes.zip)

thinkyhead, 2017-11-10:
I've started to clean it up to make sense of it and see how it compares to `M290` in the latest firmware. I'll pick this up more later.

For `Marlin_main.cpp`…
```cpp
inline void gcode_M192() {
  if (parser.seen('R')) {                                       // M192 R does babysteps now
    SERIAL_ECHOLN("baby..");
    realTimeStep(
      parser.linearval('X'),
      parser.linearval('Y'),
      parser.linearval('Z'),
      parser.byteval('D')
    );
    SERIAL_ECHOLN("..stepped");
  }
  else {
    bool didXYZ = false, didE = false;
    LOOP_XYZE(i) {
      destination[i] = current_position[i];                     // Set destination from current
      if (parser.seen(axis_codes[i])) {                         // For each axis code...
        destination[i] -= parser.value_linear_units();          // Back up the current position by the given length
        if (i == E_AXIS) didE = true; else didXYZ = true;       // Remember if E or XYZ axes were done
      }
    }

    if (didXYZ || didE) {                                       // If there are any moves...
      if (didXYZ) {                                             // Any XYZ...
        stepper.synchronize();                                  // Synchronize for an XYZ move
        planner.set_position_mm_kinematic(destination);         // Set planner to the new position, with kinematics
      }
      else if (didE)                                            // For E-only, don't sync (G92 recently changed to ALWAYS sync)
        planner.set_e_position(destination[E_AXIS]);            // Set E position is faster, with no kinematics

      // Move "back" to the old current position
      planner.buffer_line_kinematic(current_position, MMS_SCALED(feedrate_mm_s), active_extruder);
    }
  }
}
```

For `stepper.cpp`...
```cpp
void realTimeStep(int16_t x, int16_t y, int16_t z, uint8_t delaytime) {
  const bool dir[3] = { x > 0, y > 0, z > 0 };
  x = abs(x); y = abs(y); z = abs(z);
  const int16_t maxsteps = MAX3(x, y, z);
  if (!maxsteps) return;
  for (int16_t i = 0; i < maxsteps; i++) {
    if (i < x) babystep(X_AXIS, dir[X_AXIS]);
    if (i < y) babystep(Y_AXIS, dir[Y_AXIS]);
    if (i < z) babystep(Z_AXIS, dir[Z_AXIS]);
    safe_delay(delaytime);
  }
}
```

thinkyhead, 2017-11-10:
Compare to…
```cpp
inline void gcode_M290() {
  #if ENABLED(BABYSTEP_XY)
    for (uint8_t a = X_AXIS; a <= Z_AXIS; a++)
      if (parser.seenval(axis_codes[a]) || (a == Z_AXIS && parser.seenval('S'))) {
        float offs = parser.value_axis_units(a);
        constrain(offs, -2, 2);
        #if ENABLED(BABYSTEP_ZPROBE_OFFSET)
          if (a == Z_AXIS) {
            zprobe_zoffset += offs;
            refresh_zprobe_zoffset(true); // 'true' to not babystep
          }
        #endif
        thermalManager.babystep_axis(a, offs * planner.axis_steps_per_mm[a]);
      }
  #else
    if (parser.seenval('Z') || parser.seenval('S')) {
      float offs = parser.value_axis_units(Z_AXIS);
      constrain(offs, -2, 2);
      #if ENABLED(BABYSTEP_ZPROBE_OFFSET)
        zprobe_zoffset += offs;
        refresh_zprobe_zoffset(); // This will babystep the axis
      #else
        thermalManager.babystep_axis(Z_AXIS, parser.value_axis_units(Z_AXIS) * planner.axis_steps_per_mm[Z_AXIS]);
      #endif
    }
  #endif
}
```

thinkyhead, 2017-11-10:
Please give `M290` a test from your favorite host and see if it is sufficient for your needs. The `M192` methodology of telling the planner that the current position is different, then moving back to the previous current position will only work on Cartesian systems. Twerking the planner position in this way will unfortunately screw up Delta or SCARA kinematics.

rooiejoris, 2017-11-11:
I might be able to try [not soon though unfortunately], not very good in github and branches [and machinefar away], but I saw a m290 in branch bugfix-1.1.x
Is that the right branch?
And how should M290 work?
M290 X2 should offset 2mm positive in x direction...?!

cheers / joris

(side note, I am fighting with UBL, is anywhere described what the difference / relation is between the different Z-offsets: M206 Z10.7, M851 Z-3.7, G29 Z... (M206 Z10.7 works perfect without UBL))

thinkyhead, 2017-11-26:
Sorry to take so long getting back to this issue. The `M290` feature is now documented at http://marlinfw.org/docs/gcode/M290.html. `M290 X2` will move the X axis 2mm in the X direction, with a limit of -2mm to +2mm in a single move.

The `M206` command is the `home_offset` which is rarely used. It's a way to set a default coordinate-space offset for your further use, and it can be saved with `M500`, so for example, you could use G-code sliced for a Delta with `M206 X-100 Y-100`.

The `M851 Z` command is used to tell the machine how far the probe trigger-point is from the tip of the nozzle. The default is set with the config option `Z_PROBE_OFFSET_FROM_EXTRUDER` (along with `X_PROBE_OFFSET_FROM_EXTRUDER` and `Y_PROBE_OFFSET_FROM_EXTRUDER` which cannot be changed once set). If the probe trigger-point is 5mm below the nozzle you can tell Marlin using `M851 Z-5`. The Probe Z Offset is used for all subsequent `G29` or other probing operations, and changing it does not affect previous readings.

rooiejoris, 2017-11-26:
Hey hey,

No problem, I am glad with an answer...!!
Will try to test M290, sounds like what I want. Do you know if this command is placed in the buffer, or does it bypass the buffer...?!

I think the M206 is also used with the UM2 which homes at the bottom end stop...
I use[d] M206 because both Z-endstops are conductive and mounted with a Z offset of M206 Z10.7 [So the bed is 10.7 away from the nozzle when homing]
At the moment I can print and tune the nozzle to bed distance with M851, it works, but I don't understand why with the numbers I use... : )

thanks...!

thinkyhead, 2018-02-09:
> Do you know if this command is placed in the buffer

All commands go into the buffer (unless `EMERGENCY_PARSER` is set to process `M108`, `M112`, `M410` early).

closed 2018-02-09 by thinkyhead

rooiejoris, 2018-04-08:
never got back to this issue, but M290 works great...!!!