MarlinFirmware/Marlin #8351: Marlin 1.1.6 unable to compile after removing comment for, "#define NUM_SERVOS 3"

ebraiman, 2017-11-09:
Thank you for submitting your feedback to the Marlin project.
Please use one of the templates below to fill out this box.

------------------------------------------------------------
### Feature Request
Please put [FR] in the issue title: `[FR] Add-on that goes 'ping'`

------------------------------------------------------------
### Compile Error
When I compile with `#define NUM_SERVOS 3` I get an error:
```
sketch\servo.cpp: In member function 'void Servo::move(int)':

servo.cpp:312: error: static assertion failed: SERVO_DELAY must be an array NUM_SERVOS long.

   static_assert(COUNT(servo_delay) == NUM_SERVOS, "SERVO_DELAY must be an array NUM_SERVOS long.");

   ^

exit status 1
static assertion failed: SERVO_DELAY must be an array NUM_SERVOS long.
```

------------------------------------------------------------
### Bug Report
- Description: Marlin 1.1.6 unable to compile after removing comment for, "#define NUM_SERVOS 3"
- Expected behaviour: Compile without issue
- Actual behaviour: Compile error above
- Steps to reproduce:
  - Remove comment for, "#define NUM_SERVOS 3".
     Compile

  - This appears to occur in  1.1.6.  When I tried 1.0.2.2 and repeated steps above no compile error.

Attach a ZIP of `Configuration.h` and `Configuration_adv.h` by dropping here.

------------------------------------------------------------
### Bug Report Tips
- When troubleshooting, use `M502` followed by `M500` to reset EEPROM to defaults.
- Use `DEBUG_LEVELING_FEATURE` with `M111 S247` for detailed logging of homing/leveling.
- Format text with: **bold**, _italic_, `code`.
- Format C++ with three backticks, plus "cpp":
```cpp
void my_function(bool do_it) {
  // Hold this spot
}
```

GMagician, 2017-11-09:
Check your configuration.h file, message is almost clear SERVO_DELAY must be an array i t means you have to specify as meny values as the number of servo you have
eg 
 #define SERVO_DELAY { 300, 300, 300 }

ebraiman, 2017-11-09:
Closing issue, Per comment above.  Appears to and a I-d-"Ten"-T Error,

closed 2017-11-09 by ebraiman