MarlinFirmware/Marlin #8353: Marlin 2.0 and MKS Sbase - no serial connection with printer

FlameFlameFlame, 2017-11-09:
With the settings from attached files I cannot communicate with printer using Pronterface or similar software: there is no responce from it.
[configuration.zip](https://github.com/MarlinFirmware/Marlin/files/1458841/configuration.zip)
This is Configuration.h section with communication settings: 
`#define SERIAL_PORT 0`
`#define BAUDRATE 9600`
However, there is a COM-port in Device Manager and SD-card installed in printer is shown in My Computer.
I tried different BAUDRATE settings and I had similar results.

Spawn32, 2017-11-09:
Hi,
Use #define SERIAL_PORT -1 and it will work, dont know when it changed, but the example should have -1 :)

thinkyhead, 2017-11-10:
Patched in the relevant config.

closed 2017-11-10 by thinkyhead