MarlinFirmware/Marlin #8354: How would marlin handle `#` in gcode? (Cura comment issue pondering)

filipgoc, 2017-11-09:
There's a known problem in Cura with [handling comments. ](https://github.com/Ultimaker/Cura/issues/1237)

Basically, under some conditions, it strips all lines starting with `;`
- which is a problem for nice start and end gcode scripts. 

One of the possible solutions for Cura was to use `#` as a comment prefix. 

I now wonder, what would that do to Marlin processed gcodes? Any harm in that? Would that be treated as a comment? 

## Example: 
```
#               Print Settings Summary
#              sliced for :  {machine_name}
#         nozzle diameter :  {machine_nozzle_size}
#       filament diameter :  {material_diameter}
#            layer height :  {layer_height}
#  M92 E140 ; optionally adjust steps per mm for your filament
M117 Heating Up       ; write heating up
M107                  ; start with the fan off
...
```

filipgoc, 2017-11-09:
Let it go for now. Cura does not handle `#`...

closed 2017-11-09 by filipgoc

rszasz, 2017-11-09:
Implementing another EOB symbol that nobody else uses seems like a less optimal fix than having Cura parse correctly.

filipgoc, 2017-11-09:
I agree @rszasz, but it seems non-trivial. Alas, let me get more info on the Cura side first before getting to modify the Marlin side.