MarlinFirmware/Marlin #8358: Sbase: How to setup SPI for sdcard reading? "sdcard init failed".

karabas, 2017-11-10:
I use different pin combinations, SPI speeds, but still get sd init failed.
I need sdcard gcode storage to work.

Yes I have heard about:
> This is a well-known issue with all LPC176x-based controllers, and is the reason that the external SD card included with LCD displays is unsupported in Smoothieware.

But.
The truth is that it works on Smoothieware perfectly on about 100 different delta printers. 
The only issue is Power-on hanging with sdcard inserted.

So why it cannot work on Marlin2?
Cardreader EXT2 layout 
2.1 miso  P0.8
2.2 Sck    P0.7
2.3 Enc1 
2.4 SDSS     P0.28 
2.5 Enc2
2.6 mosi  P0.9
2.7 SCDetect P0.27
...

I tried spi_pins,h
```
#ifndef SPI_PINS_LPC1768_H
#define SPI_PINS_LPC1768_H

#define LPC_SOFTWARE_SPI

/** onboard SD card */
#define SCK_PIN           P0_7
#define MISO_PIN          P0_8
#define MOSI_PIN          P0_9
#define SS_PIN            P0_6
/** external */ /*
#define SCK_PIN           P0_15
#define MISO_PIN          P0_17
#define MOSI_PIN          P0_18
#define SS_PIN            P1_23
*/
#define SDSS              -1 //SS_PIN

#if (defined(TARGET_LPC1768) && !(defined(LPC_SOFTWARE_SPI)))   // signal LCDs that they need to use the hardware SPI
  #define SHARED
```_SPI
#endif

in sbase_pins.h I defined
`  #define LCD_SDSS         P0_28  // EXP2.4`

If I comment out #define LPC_SOFTWARE_SPI
I get compilation error
```
Marlin\src\HAL\HAL_LPC1768\HAL_spi.cpp: In function 'void spiBegin()':
Marlin\src\HAL\HAL_LPC1768\HAL_spi.cpp:208:38: error: request for member 'pin' in 'pin_map[263]', which is of non-class type 'const pin_t {aka const short int}'
PinCfg.Pinnum = pin_map[SCK_PIN].pin;
^~~
```

karabas, 2017-11-10:
as I wrote in https://github.com/MarlinFirmware/Marlin/issues/8361
viki2 display works with sdcard but with some strange interference.

Bob-the-Kuhn, 2017-11-10:
What board are you using?

Bob-the-Kuhn, 2017-11-10:
The compile errors are caused by a bug in 2.0.x  We missed a file when converting to the new pin numbering system.

I'll get this fixed in the next 30 minutes.

Bob-the-Kuhn, 2017-11-10:
I think you're on the right track to use the LPC1768 hard spi.  

If I'm not mistaken the on board SD card reader uses the same SCK, MOSI & MISO pins via a hardware SPI.  A software SPI can't use the hardware spi's pins.

There is one change I think you need to make.  Change `#define SS_PIN            P0_6` to `#define SS_PIN            P0_28`.  

I also think that `#define SDSS              -1 //SS_PIN` should be deleted.

The SS_PIN is the chip select for the SPI.  I expect that P0_6 is the select for the onboard SD card.

karabas, 2017-11-13:
I looked to the code. It tries to init display sdcard twice. 
```
  if (!card.init(SPI_SPEED, SDSS)
    #if defined(LCD_SDSS) && (LCD_SDSS != SDSS)
      && !card.init(SPI_SPEED, LCD_SDSS)
    #endif 
```
So if **lcd_sdss** is defined you don't need **sdss** anymore.
I tried to change #define SS_PIN P0_6 to #define SS_PIN P0_28. No difference

karabas, 2017-11-13:
I updated to #8364 
edited spi_pins.h to:
```
//#define LPC_SOFTWARE_SPI

/** onboard SD card */
#define SCK_PIN           P0_7
#define MISO_PIN          P0_8
#define MOSI_PIN          P0_9
#define SS_PIN            P0_6

#define SDSS              -1

#if (defined(TARGET_LPC1768) && !(defined(LPC_SOFTWARE_SPI)))   // signal LCDs that they need to use the hardware SPI
  #define SHARED_SPI
#endif

```
Now it compiles. Results:
Nothing changes for RRD GLDC display. No SDcard init
Mks mini12864 display shows welcome screen and freeses, When I mount the board via USB I get garbage on screen. 
It seems that LCD spi select is still  the same(!) as for onboard sdcard. I tried different combinations.

karabas, 2017-11-15:
SHARER_SPI is not used anywhere in the code.
So the code below must ne removed

```
#if (defined(TARGET_LPC1768) && !(defined(LPC_SOFTWARE_SPI)))   // signal LCDs that they need to use the hardware SPI
  #define SHARED_SPI
#endif
```

Bob-the-Kuhn, 2017-11-19:
Did you get your SD card and display issues sorted out?

I did some looking at the EXP1 & EXP2 connectors and at the current LPC1768 code.

The EXP1 & EXP2 shrouds are physically rotated 180 degrees from those on the RepRap Discount Full Graphics Smart Controller.  You'll either need a custom cable or need to pry the shrouds up and rotate them on one end or the other.

The good news is that EXP1 has 5V on it which means you don't need the Re-ARM cable with pin 10 split out.

To get the display & encoder sections working I believe you'll need the following:
```
uncomment the following in the "#elif ENABLED(U8GLIB_ST7920)" section of the file "ultralcd_impl_DOGM.h"  
   U8GLIB_ST7920_128X64_RRD u8g(LCD_PINS_D4, LCD_PINS_ENABLE, LCD_PINS_RS);

use the following defines:  
  #define BEEPER_PIN          P1_31

  #define BTN_EN1             P3_26
  #define BTN_EN2             P3_25
  #define BTN_ENC             P1_30

  #define KILL_PIN            P1_22
  
  #define LCD_PINS_RS         P0_16
  #define LCD_PINS_ENABLE     P0_18
  #define LCD_PINS_D4         P0_15
```

----

Getting the SD card on the display functional will be a lot harder.  EXP2 uses the same hardware SPI as the SBASE's on board SD card reader.  The current LPC1768 software for the on board SD card can not share that SPI.

I'm afraid that you're stuck printing from the host until someone spends a fair amount of time coming up with new SD card software.

----

For the Mks mini12864 I'm surprised you're getting anything at all on the display.  The LCD and the SD card share the same SPI which is the one on EXP2.

Bob-the-Kuhn, 2017-11-20:
I tried unsuccessfully to access the Re-ARM's on board SD card via Marlin's standard SD card software.  I did this by converting the hardware SPI software to use the same SPI controller as used by the on-board SD card.

The changes I did caused the virtual disk to stop working (the on-board SD card didn't show up as a drive on my PC).  It also caused the controller to not talk to the host software.

It'll take a lot of work to get some working external SD card software.

Bob-the-Kuhn, 2017-11-20:
There is one ugly way of getting access to the external SD card.  It requires the use of the Marlin SD card's software SPI and really ugly octopus cabling.

karabas, 2017-11-20:
I wonder why SPI select does not select anything. 
MKS MINI12864 works until board does not R/W onboard sdcard. 
Reading causes garbage on the screen. Writing causes halt.

Bob-the-Kuhn, 2017-11-20:
It's because the on-board SD card's software assumes it's the only one using the SPI 1 (SSP1) controller on the LPC1768.  Basically the standard LCD SD card software and the on-board SD card software are fighting over the same pins. 

There's no easy way to make them play nicely together.

eyal0, 2017-11-27:
I might have this problem?  My SD card works only sometimes.  It seems that about 80% of the time I get "init failed" but the rest of the times I get "ok".

Am I having the same issue as in this report?  Thanks!

AnHardt, 2017-11-27:
@eyal0 
> Am I having the same issue as in this report?

May be.
With the information you provided it's impossible to say yes or no.

Glod76, 2017-12-31:
I'm using marlin v2 on the mks sbase with the RRDFGD I can access the sd card on the mks sbase through windows but sd card won't show up in octoprint and on the display when I go to sd card says no sd card. so I'm stuck printing from the computer and I've been getting random freezes mid prints. any idea what could be done to fix this yet? Thanks.

Bob-the-Kuhn, 2017-12-31:
"marlin v2" - is that Marlin 2.0.x or something else?

If it's Marlin 2.0.x then the following applies.

The SD card built into the MKS board is used only for configuration and firmware storage.  It can't be printed from.

In order to use the SD card on the RepRap Discount Full Graphics Smart Controller to print from you'll need to use a custom cable.  See the README file in the Marlin\src\config\examples\Mks\Sbase directory.

closed 2018-02-11 by thinkyhead