MarlinFirmware/Marlin #8360: Marlin 2.0, servos with Arduino Due

zlebnik, 2017-11-10:
### Compile Error
Branch 2.x

When I compile with servos enabled (NUM_SERVOS > 0) I get an error:
```
In file included from Marlin/src/HAL/HAL_DUE/MarlinSerial_Due.cpp:34:0:
Marlin/src/HAL/HAL_DUE/../../Marlin.h:191:10: error: 'HAL_SERVO_LIB' does not name a type
extern HAL_SERVO_LIB servo[NUM_SERVOS];
^
*** [.pioenvs/DUE/src/src/HAL/HAL_DUE/MarlinSerial_Due.o] Error 1
```

Looks like HAL_SERVO_LIB is not defined in HAL_Due.h

thinkyhead, 2017-11-15:
Does defining `HAL_SERVO_LIB` in `HAL_Due.h` take care of the compile error for you?

```diff
  // --------------------------------------------------------------------------
  // Types
  // --------------------------------------------------------------------------

  typedef int8_t pin_t;
+
+ #define HAL_SERVO_LIB Servo
```

zlebnik, 2017-11-15:
Hi!
Yes, this fixes compilation issue. I may also try to check if servo works after this