MarlinFirmware/Marlin #8362: M428 changes nothing but says offsets applied

FX117, 2017-11-10:
### Bug Report
- Description: Sending M428 changes nothing but says offsets applied 
- Expected behaviour: The current position is set to the native home position. As described [here](http://marlinfw.org/docs/gcode/M428.html)
- Actual behaviour: It changes nothing, but the display says offsets applied
- Steps to reproduce:
  - Send M428 to Marlin

Background: I thought M428 would be a good way to set a persistent offset that I could return to after power loss on my Mostly printed CNC.

[config.zip](https://github.com/MarlinFirmware/Marlin/files/1462354/config.zip)

thinkyhead, 2017-11-11:
Same with `M206`?

FX117, 2017-11-11:
No M206 works fine

RowanMeara, 2017-11-13:
Interesting, I'll take a look at this.

closed 2017-11-21 by thinkyhead