MarlinFirmware/Marlin #8363: UBL and Z-offset for extruder?

alexxy, 2017-11-10:
I have dondolo extruder, and now using UBL.
Does UBL correcly works with Z offset for extruder?

And another Q:

How extruder offsets are calculater? left- right+?

thinkyhead, 2017-11-11:
In what context?

alexxy, 2017-11-11:
Ok. More concreet description:

Extruder type dondolo. To recalibrate it set inital offsets for second head to 0, 0, 0 for XYZ
Print test object like one attached 
[cube2.zip](https://github.com/MarlinFirmware/Marlin/files/1463589/cube2.zip)
Measure difference with caliper (in my case it was ~13.5 mm for X, ~0.4 mm for Y and ~0.2 mm for Z)
Set extruder offsets with 
'''
M218 T1 X13.50 Y-0.40 Z-0.20
M500
M501
'''
Check it with M503
Reprint test object again, And measure offsets they now should be in sub-mm range, but

Offsets are ~ -13.5mm +0.4mm +0.2mm for Z (so Marlin seems applyed offsets twice....)

alexxy, 2017-11-11:
Example with sample cube
```
M503
SENDING:M503
G21    ; Units in mm
M149 C ; Units in Celsius
Filament settings: Disabled
M200 D3.00     
M200 T1 D3.00  
M200 D0        
Steps per unit:
M92 X80.00 Y80.00 Z1600.00 E149.79
Maximum feedrates (units/s):
M203 X300.00 Y300.00 Z2.00 E25.00
Maximum Acceleration (units/s2):
M201 X1000 Y1000 Z50 E5000
Acceleration (units/s2): P<print_accel> R<retract_accel> T<travel_accel>
M204 P1000.00 R1000.00 T1000.00
Advanced: S<min_feedrate> T<min_travel_feedrate> B<min_segment_time_us> X<max_xy_jerk> Z<max_z_jerk> E<max_e_jerk>
M205 S0.00 T0.00 B20000 X20.00 Y20.00 Z0.40 E5.00
Home offset:   
M206 X0.00 Y0.00 Z0.00
Hotend offsets:
M218 T1 X0.00 Y0.00 Z0.00
Unified Bed Leveling:
M420 S1 Z0.00  
Unified Bed Leveling System v1.01 active.
Active Mesh Slot: 1
EEPROM can hold 50 meshes.
Material heatup parameters:
M145 S0 H180 B70 F0
  M145 S1 H240 B110 F0
PID settings:  
M301 P16.53 I1.53 D44.69
M304 P98.26 I9.26 D260.81
Z-Probe Offset (mm):
M851 Z-0.90    
```
Test object as expected has shift for X Y and Z axis.. X shift is 33.5-20=13.5 mm
[image]

Now set extruder offsets to M218 T1 X13.5 Y0.0 Z0.0
like this:

```
SENDING:M503
G21    ; Units in mm
M149 C ; Units in Celsius
Filament settings: Disabled
M200 D3.00     
M200 T1 D3.00  
M200 D0        
Steps per unit:
M92 X80.00 Y80.00 Z1600.00 E149.79
Maximum feedrates (units/s):
M203 X300.00 Y300.00 Z2.00 E25.00
Maximum Acceleration (units/s2):
M201 X1000 Y1000 Z50 E5000
Acceleration (units/s2): P<print_accel> R<retract_accel> T<travel_accel>
M204 P1000.00 R1000.00 T1000.00
Advanced: S<min_feedrate> T<min_travel_feedrate> B<min_segment_time_us> X<max_xy_jerk> Z<max_z_jerk> E<max_e_jerk>
M205 S0.00 T0.00 B20000 X20.00 Y20.00 Z0.40 E5.00
Home offset:   
M206 X0.00 Y0.00 Z0.00
Hotend offsets:
M218 T1 X13.50 Y0.00 Z0.00
Unified Bed Leveling:
M420 S1 Z0.00  
Unified Bed Leveling System v1.01 active.
Active Mesh Slot: 1
EEPROM can hold 50 meshes.
Material heatup parameters:
M145 S0 H180 B70 F0
  M145 S1 H240 B110 F0
PID settings:  
M301 P16.53 I1.53 D44.69
M304 P98.26 I9.26 D260.81
Z-Probe Offset (mm):
M851 Z-0.90    
```

Expected results: No shift for X axis... (at least it should be in sub-mm range) But... 

[image]

And WOW its now on another side with X-shift of -13.5 =)

alexxy, 2017-11-11:
And btw seems like same problems for X and Y axis, while Z-offset seems doesnt move extruder at all.

[config.zip](https://github.com/MarlinFirmware/Marlin/files/1464070/config.zip)

thinkyhead, 2017-11-12:
Thank you for the clarification! I will tackle this tonight.

thinkyhead, 2017-11-12:
I'm assuming you have two nozzles on the same carriage. Also, I assume that your print job started with things in the following order:
```gcode
M218 T1 X13.50 Y-0.40 Z-0.20
T0
G28
T1
```
Apparently if `M218` is done with the second nozzle selected it doesn't do anything to update the current position, and that's the way it has always been. I will see about making `M218` a little smarter.

thinkyhead, 2017-11-12:
Ok, I think this might do it. Position shift shouldn't be applied in tool change after the native workspace update, but it was left in. (I only looked for the macros when global searching to fix it, and apparently I need to look for `position_shift` in more places.

alexxy, 2017-11-12:
No =) I have [dondolo](https://www.thingiverse.com/thing:673816) like extruder. Its two nozzles + servo. Extruder offsets are set in eeprom, so there  no need to set them every print start. Proble is as i described above
* Extruder offsets for X and Y are actualy applyed twice (If i set them to shift/2 then it will produce good print)
* Extruder offset for Z seems doesnt applyed at all....

And this problem arise after f8393a0 commit.... Before that extruder offsets works as expected.
About Z-offset i'm not sure if it applyed with UBL...

alexxy, 2017-11-12:
@thinkyhead  i tested #8381 it seems solve issue with X and Y, but not wtih Z... 
So Z hotend offset doesnt work... (tryed moving it to 10mm and no effect)

thinkyhead, 2017-11-14:
I will follow up later today…

thinkyhead, 2017-11-15:
Sorry, got caught up in fixing `M32 P` all day. I will get in-depth with all (well, the one or two minor) extant coordinate space issues tomorrow as I audit and test as many functions as I can. I can get my machine to "believe" it has two nozzles and experiment with it directly.

alexxy, 2017-11-15:
@thinkyhead another potentialy related problem...

Seems like UBL correction applyed twice... (i specialy made heatbead that is tilted by its diagonal)
```
SENDING:G29 T
Bed Topography Report:
(0,3)                   (3,3)
(20,158)                (180,158)
  0.319    0.567    0.814    0.869
               
  0.152    0.467    0.742    0.860
               
  0.160    0.353    0.608    0.632
               
[ 0.031]   0.162    0.471    0.520
(20,20)                    (180,20)
(0,0)                     (3,0)

```

thinkyhead, 2017-11-17:
Nice mesh you have there. It doesn't really tell us anything though, does it?

alexxy, 2017-11-17:
@thinkyhead this mesh shows that bed is tilted around its diagonal. So in case UBL correction applyed twice you'll get nice symptoms
1. On upper part of heatbead you'll get big (+) gap between nozzle and heatbead
2. on middle part bed leveling will be almoust right
3. on lower part of heatbead you'll habe (-) gap so nozzle and heatbead

thinkyhead, 2017-11-18:
I see. If the offset were applied twice I would expect a big gap where the bed is tilted up, but a nozzle crash where the bed is tilted down.

thinkyhead, 2017-11-25:
Please test with current `bugfix-1.1.x` and see if the updated `G26_MESH_VALIDATION` feature helps to reveal anything if discrepancies appear.

boelle, 2019-02-19:
@alexxy Please test with current `bugfix-2.0` and see if the updated `G26_MESH_VALIDATION` feature helps to reveal anything if discrepancies appear.

boelle, 2019-03-06:
@thinkyhead i think we can close this one

Bidulglloq, 2019-04-04:
Mode details (facing similar dual extruder offset / BLTouch activated) on current bugfix 1.1.x: 

- Dual Extruder: T0 on the right side, T1 is 33mm to the left: HOTEND_OFFSET_X {0.0, -33.00}
activated the BLTouch options in marlin as recommended
- T0 active: G28 works well, but everything is then shifted 33mm to the right
(READ: **X:107.00** Y:93.00 Z:5.00 E:0.00 Count X:10700 Y:9300 Z:2000)
- G29 (3x3) has an issue on the first 2 points (shifted 33 right), 3rd probe point is on the right (at physical endstop) and X is correct. Following 2 lines (3 + 3 probe points are correct)
(READ: **X:140.00** Y:93.00 Z:5.00 E:0.00 Count X:10700 Y:9300 Z:2000)
- From now, everything is correctly located. Following G29 as well.

As a conclusion, G29 V4 gives same correct sampling coordinates on both attempts, but the 1st G29 after G28 is probing the wrong location on the 1st probing line...

More on: [(https://github.com//issues/13543)]

closed 2019-05-16 by thinkyhead