MarlinFirmware/Marlin #8368: Marlin-bugfix-2.0.x-bug

hn0550, 2017-11-11:
Thank you for submitting your feedback to the Marlin project.
Please use one of the templates below to fill out this box.

------------------------------------------------------------
### Feature Request
Please put [FR] in the issue title: `[FR] Add-on that goes 'ping'`

------------------------------------------------------------
### Compile Error
When I compile with `FEATURE_X` I get an error:
```
Paste_the_error_text_here
```
1. user  MOTHERBOARD BOARD_RAMPS_FD_V1
err:  C:\Users\Administrator\AppData\Local\Arduino15\packages\arduino\tools\arm-none-eabi-gcc\4.8.3-2014q1/bin/arm-none-eabi-gcc.exe: The filename or extension is too long.
2.user  MOTHERBOARD BOARD_RAMPS_FD_V2
 fatal error: pins_RAMPS_FD.h: No such file or directory
------------------------------------------------------------
### Bug Report
- Description: ---
- Expected behaviour: ---
- Actual behaviour: ---
- Steps to reproduce:
  - Do this
  - Do that

Attach a ZIP of `Configuration.h` and `Configuration_adv.h` by dropping here.

------------------------------------------------------------
### Bug Report Tips
- When troubleshooting, use `M502` followed by `M500` to reset EEPROM to defaults.
- Use `DEBUG_LEVELING_FEATURE` with `M111 S247` for detailed logging of homing/leveling.
- Format text with: **bold**, _italic_, `code`.
- Format C++ with three backticks, plus "cpp":
```cpp
void my_function(bool do_it) {
  // Hold this spot
}
```

RowanMeara, 2017-11-12:
I think this is an Arduino IDE windows issue.  WIndows has a short path length limit.  There should be a fair bit of discussion on this on the arduino website.

hn0550, 2017-11-13:
But the previous  version( Marlin-bugfix-2.0.x  2017-09-11) not have this problem。
my syetem: windows 10 
IDE:1.8.4

hn0550, 2017-11-13:
this is confirm a  error :
user MOTHERBOARD BOARD_RAMPS_FD_V2
 fatal error: pins_RAMPS_FD.h: No such file or directory

fiveangle, 2017-11-13:
You're using old file structure somewhere that has not picked up the file name change in #7852

Also, once you fix that, you'll need to fix the issue that @RowanMeara mentioned. I suggest to file an issue with Arduino project for the command length limit issue (search their github: there are a few issues open against it).

hn0550, 2017-11-13:
old file structure ? Refers to the?
windows? IDE ? Marlin2.0?  @fiveangle

Tannoo, 2017-11-13:
In `marlin\src\pins\pins_RAMPS_FD_V2.h`, line 32:

```
#include "pins_RAMPS_FD.h"
```

needs to be changed to:

```
#include "pins_RAMPS_FD_V1.h"
```

That only fixes the pins file issue, but the rest of the build is incomplete.

Tannoo, 2017-11-13:
>user MOTHERBOARD BOARD_RAMPS_FD_V1
err: C:\Users\Administrator\AppData\Local\Arduino15\packages\arduino\tools\arm-none-eabi-gcc\4.8.3-2014q1/bin/arm-none-eabi-gcc.exe: The filename or extension is too long.

I didn't have this issue with PlatformIO. Arduino IDE I don't think handles 32-bit code yet.

fiveangle, 2017-11-13:
> In marlin\src\pins\pins_RAMPS_FD_V2.h, line 32:
> 
> #include "pins_RAMPS_FD.h"
> 
> needs to be changed…

You're more tolerant of digging into the code on a whim :) .  It's so much more likely to be a wild goose chase, so I usually reserve that for when they at least upload their config ;) .  But yeah, appears the V1 file name change was referred to in the V2 pins so didn't get picked up when the user submitted #7852.  I always do a global search across the entire project when I do a change like that, just in case there are others lurking.

> It's not a Windoze issue. I run Win10.

???  What is it from then if not the 32k command length limit of Arduino IDE on Windows ?

Tannoo, 2017-11-13:
I was testing by compiling with those motherboards.

I was using my code used for my printer. I found that V1 had fussed about filament runout sensor. So I diabled that. Then it fused about a variable definition (something to do with max eeprom size for ubl). No path errors.

I set the runour sensor on again and then tried v2... I got the pins file error. Hunted that down. It then fussed about the filament runout sensor.. disabled it. Then fussed about the same definition that v1 did.

I stopped there thinking along the same lines as you about unfinished DUE support.

Tannoo, 2017-11-13:
I believe the path issue is with Arduino IDE.

In the post you linked.. "...maximum command length of 32767 characters on Windows."

It didn't look like his path error was THAT long.

Bob-the-Kuhn, 2017-11-14:
The path length issue is being fixed in Arduino 1.9.  The Beta version has it now.

fiveangle, 2017-11-14:
I linked to the beta version above * shrug *

closed 2017-11-15 by thinkyhead