MarlinFirmware/Marlin #8370: WS2812 VS NEOPIXEL

fred2088, 2017-11-11:
Thank you for submitting your feedback to the Marlin project.
Please use one of the templates below to fill out this box.
enable neopixel,
not work with ws2812 ?
WS2812 work with fastled lib, how to modify to use neopixel ?

Regards
------------------------------------------------------------
### Bug Report
- Description: ---
- Expected behaviour: ---
- Actual behaviour: ---
- Steps to reproduce:
  - Do this
  - Do that

Attach a ZIP of `Configuration.h` and `Configuration_adv.h` by dropping here.

------------------------------------------------------------
### Bug Report Tips
- When troubleshooting, use `M502` followed by `M500` to reset EEPROM to defaults.
- Use `DEBUG_LEVELING_FEATURE` with `M111 S247` for detailed logging of homing/leveling.
- Format text with: **bold**, _italic_, `code`.
- Format C++ with three backticks, plus "cpp":
```cpp
void my_function(bool do_it) {
  // Hold this spot
}
```

Tannoo, 2017-11-12:
From Adafruit:
"Not all addressable LEDs are NeoPixels.“NeoPixel” is Adafruit’s brand for individually-addressable RGB color
pixels and strips based on the WS2812, WS2811 and SK6812 LED/drivers, using a single-wire control protocol.
Other LED products we carry — DotStars, WS2801 pixels, LPD8806 and “analog” strips — use different
methodologies (and have their own tutorials). When seeking technical support in the forums, a solution can be
found more quickly if the correct LED type is mentioned, i.e. avoid calling DotStars “NeoPixels”…similar, but
different!"

Tannoo, 2017-11-12:
This is not a Marlin issue. Leds recognized or not by the Neopixel library is out of Marlin's scope.

Tannoo, 2017-11-12:
FastLED supports more types of 5050 leds than Neopixel.

Tannoo, 2017-11-12:
Also from Adafruit about FastLED:
https://learn.adafruit.com/adafruit-neopixel-uberguide/advanced-coding

closed 2017-11-12 by thinkyhead