MarlinFirmware/Marlin #8371: [FR] Numbering and Style Consistency for Fan Pins

rszasz, 2017-11-11:
Fan numbering currently starts at FAN_PIN then goes to FAN1_PIN etc.. 

It seems like it should be changed from FAN to FAN0 to match the numbering for everything else? Perhaps the change include the underscore to rename the pins FAN_0_PIN through FAN_n_PIN  for matching the way other pins are named?

boelle, 2019-02-19:
is this one still relavant? i mean 1+ year have passed

boelle, 2019-06-22:
@thinkyhead Another one that needs to be closed? Sorry, I got bored.

boelle, 2019-07-21:
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2019-07-21 by boelle