MarlinFirmware/Marlin #8372: U8G HAL issues on the ATMEGA2560

dot-bob, 2017-11-11:
@Bob-the-Kuhn 
It seems the pull request " 2.0.x - U8G HAL LCD support #8262 " has broken support for the U8GLIB_SSD1306 I2C display.

After updating to this pull request my display quits working on my ATMEGA2560 platform.
Reverting the lines below brings it back to life for the time being:
`
    //U8GLIB_SSD1306_128X64_2X_I2C_2_WIRE  u8g(U8G_I2C_OPT_NONE | U8G_I2C_OPT_FAST); // 4 stripes
    U8GLIB_SSD1306_128X64_2X u8g(U8G_I2C_OPT_NONE | U8G_I2C_OPT_FAST); // 4 stripes
`

Bob-the-Kuhn, 2017-11-12:
Sounds like the 2_WIRE version isn't compatible with your display.  What display are you using?

Looks like I better restore the default selection.

dot-bob, 2017-11-12:
This is the actual display: [FUG-2864A SWPG01](http://www.futureelectronics.com/en/Technologies/Product.aspx?ProductID=UG2864ASWPG01WISECHIPSEMICONDUCTORINC8043265&IM=0)

It is the display used in the Wanaho Duplicator 6.  It uses the SSD1306 driver over I2C.

dot-bob, 2017-11-15:
So I have updated another printer to the latest bugfix-2.0.x as of tonight and it also is having similar display issues.  
This machine is running a Rumba mainboard with a ATMEGA 2560 using a RepRap Discount Full Graphics Smart Controller.  

Like the printer above to get it working I needed to uncomment the old line and comment out the new line that declares the function u8g in the file ultralcd_impl_DOGM.h.
`U8GLIB_ST7920_128X64_4X u8g(LCD_PINS_D4, LCD_PINS_ENABLE, LCD_PINS_RS);`
`//U8GLIB_ST7920_128X64_RRD u8g(LCD_PINS_D4, LCD_PINS_ENABLE, LCD_PINS_RS);`
The other definition: `U8GLIB_ST7920_128X64_4X u8g(LCD_PINS_RS); // 2 stripes, HW SPI` does not work either.

Bob-the-Kuhn, 2017-11-18:
@dot-bob - please provide your config files so I can try to re-produce the problem.

Are you using Arduino or PlatformIO?

Were you seeing compile errors or functional problems?   If functional problems, what are the symptoms?

dot-bob, 2017-11-19:
I have been building these with Platform IO.  The only problem I am noticing is nothing is being drawn on the displays.  The encoders and SD card function fine, there is just no output on the display.

The two printer configurations are here under the following branches: 
Duplicator 6: [https://github.com/dot-bob/Marlin/tree/MMU-2.0.x](https://github.com/dot-bob/Marlin/tree/MMU-2.0.x)

Rumba Based Machine:  [https://github.com/dot-bob/Marlin/tree/Spectra-2.0.x](https://github.com/dot-bob/Marlin/tree/Spectra-2.0.x)

The branches contains the 2.0.x code at the time of the issue.

Bob-the-Kuhn, 2017-11-25:
@dot-bob - I've finally had a chance to look at the problems you encountered.

**U8GLIB_ST7920_128X64_4X u8g(LCD_PINS_RS); // 2 stripes, HW SPI** doesn't work.  This one is an exact copy from 1.1.x.  The only hardware SPI on a 2560 is also used by the SD card.  In order to use this selection you'd have to use custom cabling (not the standard LCD interface card).

I did find an error in **U8GLIB_ST7920_128X64_RRD**.  It was hardwired for the RAMPS board.

I'll have the updates/corrections in a PR later tonight.

Bob-the-Kuhn, 2017-11-25:
The changes have been merged.

closed 2017-11-25 by Bob-the-Kuhn