MarlinFirmware/Marlin #8374: E0_AUTO_FAN_PIN PROBLEM, MKS SBASE, started after pin num change.

Spawn32, 2017-11-11:
I use my E1 hotend output to run my E0 hotend fan so i can use the nice fan on / off function at whatever temp i want.
This used to work before the change of pin numbering:

#define E0_AUTO_FAN_PIN 59 ;( P2_6)
It compiles without error but dosent work, fan never turns on.

So i tested:
#define E0_AUTO_FAN_PIN P2_6
This gives the following error:

Marlin\src\module\temperature.cpp: In static member function 'static void Temperature::checkExtruderAutoFans()':
Marlin\src\module\temperature.cpp:520:130: error: narrowing conversion of '326' from 'int' to 'int8_t {aka signed char}' inside { } [-Wnarrowing]
static const int8_t fanPin[] PROGMEM = { E0_AUTO_FAN_PIN, E1_AUTO_FAN_PIN, E2_AUTO_FAN_PIN, E3_AUTO_FAN_PIN, E4_AUTO_FAN_PIN };
^
*** [.pioenvs\LPC1768\src\src\module\temperature.o] Error 1

[config.zip](https://github.com/MarlinFirmware/Marlin/files/1464145/config.zip)

tig33r, 2017-11-12:
Yeah, auto fan pin in MKS sbase was broken since forever (always was bug im compilation mentioned post earlier).

Spawn32, 2017-11-12:
Worked before, have a feeling it's the same on all LPC1768 / 69 ...

tig33r, 2017-11-12:
Perhaps with transition from re_arm pinmap to lpc1768/9 pinmap #8063 we broke auto fan feature.

apballard, 2017-11-14:
@tig33r  I just commented on #8063 .
This setting did break on that.  It caused my re-arm to freeze, I just thought that I had set it incorrectly and I'm waiting for a fan extender for my ramps to test.

tcm0116, 2017-11-16:
It looks like this will be fixed in #8446 unless @Bob-the-Kuhn wants to pull out the changes to temperature.cpp for the auto fan stuff and push it through sooner.

Bob-the-Kuhn, 2017-11-17:
I'll do a separate PR for this.  PR #8460 should be merged in the next 30 minutes.

hg42, 2017-12-01:
I think my PR could be related.
It's simply because a macro isn't implemented, which makes SanityCheck fail if the AUTO_FAN pwm value isn't 255.

However this issue starts with another error. So may be it's another problem.
But there may be two or more problems one of them might have been #8460.

At least after my change, the AUTO_FAN feature with PWM works for me (but still not with FAN_SOFT_PWM).
You currently cannot use P2_06 with AUTO_FAN and PWM, because it's no hardware PWM pin.
I had to switch P2_04 and P2_06 for the two fans, which works for me.
P2_06 can be used with FAN_SOFT_PWM for the part fan and P2_04 is a hardware PWM pin.

Spawn32, 2017-12-01:
Dont have the possibility to check it now since i am working on a new board, but nice work @hg42 :)

hg42, 2017-12-01:
Thanks...
both Pulls I requested today, are very simple, kind of one liners.
The main work was to find out, what's going on...also my last contact with Marlin was 4 years ago :-) and I am used to smoothie. Would like to have some of the basic features in Marlin (IO-Pin handling etc.).

Bob-the-Kuhn, 2017-12-01:
I wasn't aware that there was still a problem.  Please provide details.

Spawn32, 2017-12-01:
@Bob-the-Kuhn 
To be honest i haven't tested my MKS SBASE after your patch, bee busy with the other board, but i was more referring to @hg42  2 merged request's today:

https://github.com/MarlinFirmware/Marlin/pull/8616
and:
https://github.com/MarlinFirmware/Marlin/pull/8622

closed 2018-02-09 by thinkyhead