MarlinFirmware/Marlin #8375: MKS SBASE board with eeprom enabled crashes if you do a M500 or M501

Spawn32, 2017-11-11:
Board crashes a few seconds after doing a M500 or M501, needs a hard reset to work again.
so no UBL for me yet :)
[config.zip](https://github.com/MarlinFirmware/Marlin/files/1464149/config.zip)

karabas, 2017-11-12:
What display do you use?

Spawn32, 2017-11-12:
MKS TFT 3.2 :)

karabas, 2017-11-13:
Strange. M500 works for me perfectly if I does not use spi display. It saves eeprom.dat. Sdmounting succesfull?

ihagon, 2017-11-13:
M500 works fine for me too. I'm also using a MKS TFT 3.2 I do have a Full Graphic display connected too but I'm using the wifi connection via the MKS TFT to issue the M50X commands.

Spawn32, 2017-11-13:
strange :/
You have only enabled #define eeprom in config ? this happens when i do a 500/502 from simplify3d, could somone test that ?

Spawn32, 2017-11-13:
@karabas 
I dont have #define SDSUPPORT enabled, dont think you need that for eeprom ?

karabas, 2017-11-14:
```
#define EEPROM_SETTINGS // Enable for M500 and M501 commands
//#define DISABLE_M503    // Saves ~2700 bytes of PROGMEM. Disable for release!
#define EEPROM_CHITCHAT   // Give feedback on EEPROM commands. Disable to save PROGMEM.
```

Pentelfrost, 2017-11-14:
I also have an SBASE and it doesn't crash after M500/M501 commands. I'm using printrun via the MKS wifi to issue the commands.

Glod76, 2017-11-15:
I have an anet a8 with the mks sbace and reprap full graphics display and was wondering if someone could give me a full step by step on configuring and compiling this to work on my board. I have an inductive probe as well.

closed 2018-02-09 by thinkyhead