MarlinFirmware/Marlin #8376: Strange output from M114 on MKS SBASE

Spawn32, 2017-11-11:
While calculating my z probe offset i noticed that the M114 has a strange output, or is it me that have done something ?

M114 after a G28:

SENT: M114
READ: X:74.000000 Y:105.000000 Z:6.200000 E:2452.038818 Count X:14800 Y:21000 Z:4960
[config.zip](https://github.com/MarlinFirmware/Marlin/files/1464157/config.zip)

thinkyhead, 2017-11-11:
What should it say instead?

Spawn32, 2017-11-11:
I might have misunderstood this, but take the Z, when i am at Z -1.2, shoulden't M114 say something like z:-1.2 ?

Spawn32, 2017-11-11:
Ahh... sorry, feel like a jerk, i stared my self blind on the last Z that oblivious is the stepper :/
Z:6.200000 is what i should have looked at ...

Closing issue :)

closed 2017-11-11 by Spawn32

thinkyhead, 2017-11-13:
M114 itself had an issue, now patched. See if that clears it up.

thinkyhead, 2017-11-13:
No problemo! That's why we do the interrogations. 😁