MarlinFirmware/Marlin #8377: ld.exe: region `text' overflowed by 32530 bytes

nmpnmp, 2017-11-11:
Moving from Skynet 1.1 on an Anet A8 to Marlin 1.1.6. Want to enable ABL \ UBL but when compiling I am getting the below error message. 

I'm using the stock Anet configuration_adv.h and only a couple of mods to the configuration.h file which I've enclosed, and I've tried to comment out features I wont need, eg. SD card. 

Should this work? 

------------------------------------------------------------
### Compile Error
When I compile with `#define AUTO_BED_LEVELING_UBL` I get an error:
```
Compiling core...
Using precompiled core
Linking everything together...
"C:\Program Files (x86)\Arduino\hardware\tools\avr/bin/avr-gcc" -w -Os -g -flto -fuse-linker-plugin -Wl,--gc-sections -mmcu=atmega1284p  -o "C:\Users\user001.MPA\AppData\Local\Temp\arduino_build_261754/Marlin.ino.elf" "C:\Users\user001.MPA\AppData\Local\Temp\arduino_build_261754\sketch\G26_Mesh_Validation_Tool.cpp.o" "C:\Users\user001.MPA\AppData\Local\Temp\arduino_build_261754\sketch\I2CPositionEncoder.cpp.o" "C:\Users\user001.MPA\AppData\Local\Temp\arduino_build_261754\sketch\M100_Free_Mem_Chk.cpp.o" "C:\Users\user001.MPA\AppData\Local\Temp\arduino_build_261754\sketch\Marlin.ino.cpp.o" "C:\Users\user001.MPA\AppData\Local\Temp\arduino_build_261754\sketch\MarlinSerial.cpp.o" "C:\Users\user001.MPA\AppData\Local\Temp\arduino_build_261754\sketch\Marlin_main.cpp.o" "C:\Users\user001.MPA\AppData\Local\Temp\arduino_build_261754\sketch\Max7219_Debug_LEDs.cpp.o" "C:\Users\user001.MPA\AppData\Local\Temp\arduino_build_261754\sketch\Sd2Card.cpp.o" "C:\Users\user001.MPA\AppData\Local\Temp\arduino_build_261754\sketch\SdBaseFile.cpp.o" "C:\Users\user001.MPA\AppData\Local\Temp\arduino_build_261754\sketch\SdFatUtil.cpp.o" "C:\Users\user001.MPA\AppData\Local\Temp\arduino_build_261754\sketch\SdFile.cpp.o" "C:\Users\user001.MPA\AppData\Local\Temp\arduino_build_261754\sketch\SdVolume.cpp.o" "C:\Users\user001.MPA\AppData\Local\Temp\arduino_build_261754\sketch\blinkm.cpp.o" "C:\Users\user001.MPA\AppData\Local\Temp\arduino_build_261754\sketch\cardreader.cpp.o" "C:\Users\user001.MPA\AppData\Local\Temp\arduino_build_261754\sketch\configuration_store.cpp.o" "C:\Users\user001.MPA\AppData\Local\Temp\arduino_build_261754\sketch\dac_mcp4728.cpp.o" "C:\Users\user001.MPA\AppData\Local\Temp\arduino_build_261754\sketch\digipot_mcp4018.cpp.o" "C:\Users\user001.MPA\AppData\Local\Temp\arduino_build_261754\sketch\digipot_mcp4451.cpp.o" "C:\Users\user001.MPA\AppData\Local\Temp\arduino_build_261754\sketch\endstops.cpp.o" "C:\Users\user001.MPA\AppData\Local\Temp\arduino_build_261754\sketch\gcode.cpp.o" "C:\Users\user001.MPA\AppData\Local\Temp\arduino_build_261754\sketch\hex_print_routines.cpp.o" "C:\Users\user001.MPA\AppData\Local\Temp\arduino_build_261754\sketch\least_squares_fit.cpp.o" "C:\Users\user001.MPA\AppData\Local\Temp\arduino_build_261754\sketch\mesh_bed_leveling.cpp.o" "C:\Users\user001.MPA\AppData\Local\Temp\arduino_build_261754\sketch\nozzle.cpp.o" "C:\Users\user001.MPA\AppData\Local\Temp\arduino_build_261754\sketch\pca9632.cpp.o" "C:\Users\user001.MPA\AppData\Local\Temp\arduino_build_261754\sketch\planner.cpp.o" "C:\Users\user001.MPA\AppData\Local\Temp\arduino_build_261754\sketch\planner_bezier.cpp.o" "C:\Users\user001.MPA\AppData\Local\Temp\arduino_build_261754\sketch\printcounter.cpp.o" "C:\Users\user001.MPA\AppData\Local\Temp\arduino_build_261754\sketch\serial.cpp.o" "C:\Users\user001.MPA\AppData\Local\Temp\arduino_build_261754\sketch\servo.cpp.o" "C:\Users\user001.MPA\AppData\Local\Temp\arduino_build_261754\sketch\stepper.cpp.o" "C:\Users\user001.MPA\AppData\Local\Temp\arduino_build_261754\sketch\stepper_dac.cpp.o" "C:\Users\user001.MPA\AppData\Local\Temp\arduino_build_261754\sketch\stepper_indirection.cpp.o" "C:\Users\user001.MPA\AppData\Local\Temp\arduino_build_261754\sketch\stopwatch.cpp.o" "C:\Users\user001.MPA\AppData\Local\Temp\arduino_build_261754\sketch\temperature.cpp.o" "C:\Users\user001.MPA\AppData\Local\Temp\arduino_build_261754\sketch\twibus.cpp.o" "C:\Users\user001.MPA\AppData\Local\Temp\arduino_build_261754\sketch\ubl.cpp.o" "C:\Users\user001.MPA\AppData\Local\Temp\arduino_build_261754\sketch\ubl_G29.cpp.o" "C:\Users\user001.MPA\AppData\Local\Temp\arduino_build_261754\sketch\ubl_motion.cpp.o" "C:\Users\user001.MPA\AppData\Local\Temp\arduino_build_261754\sketch\ultralcd.cpp.o" "C:\Users\user001.MPA\AppData\Local\Temp\arduino_build_261754\sketch\utility.cpp.o" "C:\Users\user001.MPA\AppData\Local\Temp\arduino_build_261754\sketch\vector_3.cpp.o" "C:\Users\user001.MPA\AppData\Local\Temp\arduino_build_261754\sketch\watchdog.cpp.o" "C:\Users\user001.MPA\AppData\Local\Temp\arduino_build_261754\libraries\LiquidCrystal\LiquidCrystal.cpp.o" "C:\Users\user001.MPA\AppData\Local\Temp\arduino_build_261754/..\arduino_cache_891997\core\core_anet_avr_anet_0c812875ac70eb4a9b385d8fb077f54c.a" "-LC:\Users\user001.MPA\AppData\Local\Temp\arduino_build_261754" -lm
c:/program files (x86)/arduino/hardware/tools/avr/bin/../lib/gcc/avr/4.9.2/../../../../avr/bin/ld.exe: C:\Users\user001.MPA\AppData\Local\Temp\arduino_build_261754/Marlin.ino.elf section `.text' will not fit in region `text'

c:/program files (x86)/arduino/hardware/tools/avr/bin/../lib/gcc/avr/4.9.2/../../../../avr/bin/ld.exe: region `text' overflowed by 32530 bytes
[Configuration.h.txt](https://github.com/MarlinFirmware/Marlin/files/1464225/Configuration.h.txt)
[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/1464229/Marlin.zip)
[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/1464230/Marlin.zip)

collect2.exe: error: ld returned 1 exit status

Using library LiquidCrystal at version 1.0.7 in folder: C:\Program Files (x86)\Arduino\libraries\LiquidCrystal 
exit status 1
Error compiling for board Anet V1.0.

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

```

------------------------------------------------------------
### Bug Report
- Description: Cannot compile with #define AUTO_BED_LEVELING_UBL enabled 
- Expected behaviour: to be able to compile
- Actual behaviour: Errors as above. 
- Steps to reproduce:
  - enable #define AUTO_BED_LEVELING_UBL
  - compile. 

Attach a ZIP of `Configuration.h` and `Configuration_adv.h` by dropping here.

------------------------------------------------------------
### Bug Report Tips
- When troubleshooting, use `M502` followed by `M500` to reset EEPROM to defaults.
- Use `DEBUG_LEVELING_FEATURE` with `M111 S247` for detailed logging of homing/leveling.
- Format text with: **bold**, _italic_, `code`.
- Format C++ with three backticks, plus "cpp":
```cpp
void my_function(bool do_it) {
  // Hold this spot
}
```

Roxy-3D, 2017-11-11:
If you have a processor with 128KB of program space...   You will have to turn off lots of stuff to get UBL to fit.

fiveangle, 2017-11-11:
Search “128k PROGMEM” across open and closed issues and you’ll find lots of resources about this.

nmpnmp, 2017-11-12:
Thanks both, I did search and found your previous post fiveangle, showing the snapshot of mem usage per feature, which is really useful. I will have a play see if I can get this work.  Failing which, I guess bilinear auto levelling should work fine? Do you know what type of auto levelling Skynet uses? 

Thanks, I’m on a steep learning curve with this as I’m sure you can tell.

closed 2017-11-12 by thinkyhead