MarlinFirmware/Marlin #8378: [FR] platformio setup/configuration guide

Phr3d13, 2017-11-12:
### Feature Request
Please put [FR] in the issue title: `[FR] Add-on that goes 'ping'`

Looking for a generalized "how to set up and use platformio to compile marlin for your target" guide

thinkyhead, 2017-11-12:
That's a big guide to write. PlatformIO is differently set up on every Operating System, and integrates differently with every IDE and toolset that can integrate with it. Sublime Text requires a plugin called DevIoT. Atom and Visual Studio Code each have their own way of dealing with PlatformIO. The best guides to setting these up and building with them is provided by PlatformIO itself on their website. Building Marlin in PlatformIO is not any different from building other stuff, except it might be a little larger than your average sketch.

boelle, 2019-02-19:
i dont think its fair to demand this from Marlin as its different for every OS

its more a user problem to figure this out

thinkyhead, 2019-02-19:
http://marlinfw.org/docs/basics/install_platformio.html

closed 2019-02-19 by thinkyhead