MarlinFirmware/Marlin #8379: M3 S0 does not set PWM freq, only changes SPINDLE_LASER_ENABLE_PIN state.

hunagyp, 2017-11-12:
Whenever I send M3 S0 gcode, the laser is not set to 0% PWM duty cycle. 
- Expected behaviour: 
I think M3 S0 should only influence the laser power
- Actual behaviour: 
With S0 parameter the code only switches SPINDLE_LASER_ENABLE_PIN state but does not change the value of SPINDLE_LASER_PWM_PIN.
In Marlin_main.cpp this is the affected part, I guess:
`        if (spindle_laser_power == 0) {
          WRITE(SPINDLE_LASER_ENABLE_PIN, !SPINDLE_LASER_ENABLE_INVERT);                                    // turn spindle off (active low)`
I guess, there is a reason behind this, but for those, who might not use ENABLE pin at all with a laser, it might be confusing that the power level cannot be set to 0%.

thinkyhead, 2017-11-12:
I think the intention was that the laser should be disabled by `M3` imitating what `M5` does. In the case of no enable pin I agree it should set the PWM to 0. So that sounds like a bug!

thinkyhead, 2017-11-12:
How about if it turns off both with `M3 S0` since any future `M3` will just turn it back on, and I don't know if it's good for lasers to have their power enabled with PWM of 0.

thinkyhead, 2017-11-12:
I've applied that notion to the bugfix branches. Hope it makes sense!

hunagyp, 2017-11-12:
I did the same, I set PWM to 0 in the IF clause in my local copy.
Power at 0% is "no power", so I assume it does not hurt anything. :)
By not using ENABLE pin, I'm just risking a bit because during programming/booting, the laser is turned on at 100% for some sec, so maybe that is the reason, why ENABLE pin should be used. But I can live with it for a while.

thinkyhead, 2017-11-13:
@hunagyp It definitely makes sense to have the PWM start ahead of the ENABLE, keeping power out of the laser until its PWM is settled. The current code is tailored partly to a Spindle, so there is a momentary delay on power-up for that. Do you find that there is a similar delay for laser?

closed 2017-11-13 by thinkyhead

hunagyp, 2017-11-13:
@thinkyhead No, I did not see bigger delay for the laser (only some milisec to switch on/off).

thinkyhead, 2017-11-13:
> only some milisec to switch on/off

Ok, good to know. Some capacitors can take many seconds to drain.