MarlinFirmware/Marlin #8380: [FR] Display Z-offset on status bar

stuartmp, 2017-11-12:
Hi all,

I have a feature request suggestion which I would be happy to help implement.
I currently print from a SD card and I do not use a ABL system.
So if I need to change my Z offset I plug in my laptop run a M503 command and see what my Z-offset is. I then adjust it to the new value. 

This becomes a pain after awhile.

Is it possible to echo or print out (M117) to the status bar the Z-offset shown in the M503 command.

I see under the user macros that there are some which include M503.
 
#define USER_DESC_5 "Home & Info"
#define USER_GCODE_5 "G28\nM503"

I tried running these but I see no settings on my display so I am assuming they just
go back to a host if you had one plugged in.

Any advise on this work be great. 

I could then possibly start write up some code once I get some direction.

Thanks

Stuart

Tannoo, 2017-11-12:
The user gcode is there to modify as the user sees fit. Shortcuts to common tasks.

Tannoo, 2017-11-12:
Change any or all of them to suit your needs.

LVD-AC, 2017-11-12:
Why don't you go to the 'control' - 'motion' menu on the LCD? Z_offset is just 2 clicks away...

stuartmp, 2017-11-12:
Hi Tannoo... thanks for checking in on this issue.

I can see that those macro can be customized but I don't see any way of displaying the value of the z offset to the status bar.

Do you know how I might go about this?

stuartmp, 2017-11-12:
Hi LVD-AC.

I am familiar with the location of the z offset adjustment. But the issue is that I do not know what the current value is so lets say I need to increase my z offset 0.15mm and the current z offset is 0.2 that means that I need to set my z offset to 0.35mm which I know how to do. But unless I plug my laptop in and run a M503 I don't know what the original z offset is. So I am trying to display the z offset in the status bar so I do not have to plug my laptop in to get the value 0.2mm...

LVD-AC, 2017-11-12:
The value displayed on the LCD is the current value

stuartmp, 2017-11-12:
Wait... are you saying that there is an actual z offset menu option... that is different to the move axis -> z axis -> 0.1mm etc
Something in the control motion menu.... just Double checked I only have velocity, acceleration, jerk and steps per mm in the motion menu....  I just searched the firmware I do see something to do with z-offset in the menu but that looks to be related to some mesh leveling system which I am not using....

Tannoo, 2017-11-12:
Ok. I see what you are saying. You want M503 to display the offset on the status line?

stuartmp, 2017-11-12:
Yes... that's what I need because I don't have my computer plugged in. I use a sd card.... Do you have any idea how to do this?

LVD-AC, 2017-11-12:
It's there in the control-motion menu : z-offset, velocity, acceleration,... 
```
    #elif HAS_BED_PROBE
      MENU_ITEM_EDIT_CALLBACK(float32, MSG_ZPROBE_ZOFFSET, &zprobe_zoffset, Z_PROBE_OFFSET_RANGE_MIN, Z_PROBE_OFFSET_RANGE_MAX, lcd_refresh_zprobe_zoffset);
    #endif

```

Tannoo, 2017-11-12:
Yes, the callback is how it shows you the current offset while you adjust it.

stuartmp, 2017-11-13:
Ok, Thanks... I am not using a Z Probe so I guess that I can't just enable 'HAS_BED_PROBE'
I will need to look at this function call and see were it goes and trace it out so I get a understanding of how the method / function works. 

Then I maybe able to create a new menu item in the ultralcd.cpp file with a variable to enable it in the Configuration_adv.h file something like. DISPLAY_CURRENT_ZOFFSET what do you guys think am I on the right track?

Tannoo, 2017-11-13:
Well, maybe the zoffset can have the probe requirement removed.

stuartmp, 2017-11-13:
Good point... I might look into that first... Tweak the code and see what happens. Maybe I should just try changing the #elif to a #else first and see what happens

`#if ENABLED(BABYSTEP_ZPROBE_OFFSET)
      MENU_ITEM(submenu, MSG_ZPROBE_ZOFFSET, lcd_babystep_zoffset);
    #elif HAS_BED_PROBE
      MENU_ITEM_EDIT_CALLBACK(float32, MSG_ZPROBE_ZOFFSET, &zprobe_zoffset, Z_PROBE_OFFSET_RANGE_MIN, Z_PROBE_OFFSET_RANGE_MAX, lcd_refresh_zprobe_zoffset);
    #endif`

LVD-AC, 2017-11-13:
#elif to a #else won't work because zprobe_zoffset will be undefined.

Try to enable BABYSTEP_ZPROBE_OFFSET this should enable zprobe_zoffset , M851 and the LCD-menu

Tannoo, 2017-11-13:
```#elif ENABLED(BABYSTEP_ZPROBE_OFFSET) && !HAS_BED_PROBE
    #error "BABYSTEP_ZPROBE_OFFSET requires a probe."
``
That can't happen.

Tannoo, 2017-11-13:
Enable fixed probe and then it will show up.

Tannoo, 2017-11-13:
Either setup a NO_PROBE option or remove the probe requirement for ZOFFSET.

stuartmp, 2017-11-13:
Ok I think your right, setting up a NO_PROBE option makes more sense and keeps the other functionality the same.
I will look into it further.

cdedwards, 2017-11-13:
one thing I noticed today with current 2.0, is that babystep is not saved. to verify this, start a print and enter babystep mode and change the value. exit babystep mode and then re-enter. right back to zero

Tannoo, 2017-11-13:
Babystepping by itself will not.

Tannoo, 2017-11-13:
That is what BABYSTEP_ZPROBE_OFFSET does.

tcm0116, 2017-11-13:
`BABYSTEP_ZPROBE_OFFSET` only works if ABL is enabled and `G29` has been completed. Z Offset was only really intended to be used for probes. There's already home offset and workspace offsets that can be used, but they're not as convenient to adjust as Z offset is for probes.

Perhaps it would make sense to re-work Z offset to be more generic and apply to any situation, and to modify the home offset instead of a dedicated z offset variable. By doing so, we can generalize the Z offset menu as well as `BABYSTEP_ZPROBE_OFFSET` to work for any printer configuration.

@thinkyhead as you've been re-working all of the offset code lately, what are your thoughts on my proposal?

LVD-AC, 2017-11-13:
Home offsets (by default set to off on delta's because X and Y do very strange things to them) are quite different than probe offsets; using the nozzle to manually probe needs at least a configurable zprobe_zoffset. Just relax the HAS_BED_PROBE testing to HAS_BED_PROBE || DELTA or even to none on all zprobe_zoffset events.

stuartmp, 2017-11-14:
Hi all. Well I spent a few hours on this tonight and managed to get declare variables and fill out the code in the relevant places to enable the option. I have the z offset on the display now and the value changes with turning the dial. But the z axis does not move at the moment. But this is a work in progress.

Here is my first commit.

https://github.com/stuartmp/Marlin within the branch zoffset.

LVD-AC, 2017-11-14:
No need for that! The existing functionality just needed to be enabled if no probe was defined see #8422 and #8423 for that.

stuartmp, 2017-11-14:
Ok see the commits I will have to have a closer look tomorrow. Thank you for keeping me updated.

Would it be possible for you to tell me what needs to be done... so I just need to enable something?  Just a bit more detail would be great.

LVD-AC, 2017-11-14:
all the detail is in the PR's mentioned above

stuartmp, 2017-11-18:
Hi All... Well I had another go at adding this feature.
I have it basically working accept for displaying the current home offsets.
I have it displaying static values to the status bas but can't work out how to get it to display the current home offset values.

Maybe some one can try and help out.

`void lcd_prepare_display_home_offsets(){ 
		#define LINEAR_UNIT(N) ((N) / parser.linear_unit_factor)
		#define CURRENT_HOME_X_OFFSET 0.0 //LINEAR_UNIT(home_offset[X_AXIS])  //	<-- Don't know how to get current home offset
		#define CURRENT_HOME_Y_OFFSET 0.0 //LINEAR_UNIT(home_offset[Y_AXIS]) //	<-- Don't know how to get current home offset
		#define CURRENT_HOME_Z_OFFSET 0.2 //LINEAR_UNIT(home_offset[Z_AXIS]) //	<-- Don't know how to get current home offset		
		#define CURRENT_HOME_OFFSETS "M117" " X:" STRINGIFY(CURRENT_HOME_X_OFFSET) " Y:" STRINGIFY(CURRENT_HOME_Y_OFFSET) " Z:" STRINGIFY(CURRENT_HOME_Z_OFFSET)
				
		_lcd_user_gcode(PSTR(CURRENT_HOME_OFFSETS));
		lcd_return_to_status();		
	}`

Here is my current commit 
[https://github.com/stuartmp/Marlin/tree/bugfix-1.1.x-home_z_offset](url)

Any advise would be much appreciated

stuartmp, 2017-11-19:
@thinkyhead can you offer any advise?

LVD-AC, 2017-11-19:
Give this a try: (in ultralcd.ccp)

 ```
 void lcd_control_motion_menu() {
    START_MENU();
    MENU_BACK(MSG_CONTROL);

    #if ENABLED(BABYSTEP_ZPROBE_OFFSET)
      MENU_ITEM(submenu, MSG_ZPROBE_ZOFFSET, lcd_babystep_zoffset);
    #elif HAS_BED_PROBE
      MENU_ITEM_EDIT_CALLBACK(float32, MSG_ZPROBE_ZOFFSET, &zprobe_zoffset, Z_PROBE_OFFSET_RANGE_MIN, Z_PROBE_OFFSET_RANGE_MAX, lcd_refresh_zprobe_zoffset);
    #elif DISABLED(DELTA)
      MENU_MULTIPLIER_ITEM_EDIT(float32, MSG_ZPROBE_ZOFFSET, &home_offset[Z_AXIS], -20, 20);
    #endif

```

stuartmp, 2017-11-19:
Thanks.... I will see how I go patch that in and the tweaking it to suit my sub menu.

Really appreciate that you tried to help me out.

I made a little sub menu were you can also disable and enable software endstops.... which comes in handy if your build plate is to low

stuartmp, 2017-11-20:
The only thing is that you have to run store settings then load settings before you hit set home offsets otherwise it just clears the z offset because the z offset adjustment doesn't actually move the head. Any idea on how we can save these settings.

I also noticed that this method is not in my code 'lcd_refresh_zprobe_zoffset'
but it still seams to work..

I see it in your pull request 8422

LVD-AC, 2017-11-20:
Do not create pull request for this; this is little hack for your use only (it is not complying with Marlin philosophy). The new menu item does what M206 Zx.xx does; that does not move the head as well.

Save settings: M500

closed 2018-02-09 by thinkyhead