MarlinFirmware/Marlin #8383: Compile Error with VIKI2

WOOKIE111, 2017-11-12:
### Compile Error
------------------------------------------------------------

When I compile with VIKI2 enabled or commented in I get an error:

`Most Current Marlin 1.1.x from marlinfw.org as of Nov 11 4:14am EST`

even when the old pins_RAMPS.h and pins_AZTEEG_X3_PRO.h are pasted into the new directory. as well as the other ramps related files. (as requested to do by another user here)

it seems there is something broken for the VIKI as just uncommenting it seems to break marlin, when compiling anyways

using the most current marlin and not modifying ANYTHING and only enableing/commenting viki2 back in on line 1355 of configurtation.h throws this crazyness

Pastebin used as its rather long.

https://pastebin.com/zQggRL7f

but commenting viki2 back out. and recompiling clears it up. and its repeatable.

**TLDR ;**

## Compile error happens with:

- Sterile and Brand new Marlin as of Nov 11 2017
- Modified Marlin with Pins files from an older version of marlin ( as requested by another user)

## Compile Error Happens When:

- Viki2 is enabled in Configuration.h line 1355
**or**
- Makrpanel is enabled in Configuration.h line 1342
**or**
- miniViki is enabled in Configuration.h line 1356 

## Error goes away when:

- Viki2 is commented Back out to disable Viki2
**or**
- Makrpanel is commented Back out to disable Makrpanel 
**or**
- miniViki  is commented Back out to disable miniViki  

## Compile Error:
`exit status 1
'DIODOGLCD_SCK_RPORT' was not declared in this scope`
https://pastebin.com/zQggRL7f

`I have tested a few of the Screen Defines but not all. 
im not sure what links the offending defines to throw the same error. 
the errors are repeatable on multiple pc's and with multiple different unzips and many different downloads`

thinkyhead, 2017-11-12:
The proximal cause of the issue is that `DOGLCD_SCK` isn't being defined. Check your pins and see if there's an obvious value to use. Use other pins files for reference, if needed.

WOOKIE111, 2017-11-12:
> The proximal cause of the issue is that DOGLCD_SCK isn't being defined. Check your pins and see if there's an obvious value to use. Use other pins files for reference, if needed.

So from the get go. a brand new only thing modified is the enabling of one of those screens and marlin is already broken?? ouch.

when an old pins file is used from a working marlin the same error rears its head. 

is it the basic pins.h i should be checking or the board specific one and going all the way down to ramps

because so far i have changed azteeg all the way down to ramps

WOOKIE111, 2017-11-12:
heck that shouldnt even matter. because in a vanilla marlin it defaults to a ramps board.

thinkyhead, 2017-11-12:
So this is compiling with Viki2 _and_ the Azteeg — or just any RAMPS-based board? I'll try compiling now and see what's up!

WOOKIE111, 2017-11-12:
this is with compiling ANY board even with a pure vanilla marlin configuration

and as noted in the original post it doesnt just affect the viki. it affects many.

also attached you will find an image of a search of all and any pins files for the term `DOGLCD_SCK`

the files labelled test are the BRAND new marlin as of Nov 11th and i notice an extra entry in `pins_ramps.h` asking for an actual pin

i will see what happens when i remove that pinout it is asking for.

[image]

WOOKIE111, 2017-11-12:
# update:
It now compiles when line 294 in `pins_ramps.h` is commented OUT
as such
`        //#define DOGLCD_SCK      23`

i am not sure what that exact define does so i dont know if it SHOULD be enabled or not.

thinkyhead, 2017-11-12:
Using the current Marlin `bugfix-1.1.x` I was able to compile without error, just enabling the VIKI2 and otherwise everything at the default.

Can you post your modifications?

WOOKIE111, 2017-11-12:
I dont use bugfix as its supposedly unstable.

hence why i go for the stable build

i not once said i used bugfix

https://github.com/MarlinFirmware/Marlin/archive/1.1.x.zip is the precise link i am talking about

WOOKIE111, 2017-11-12:
Let me download bugfix to see if this error was caught by you guys and just not updated in the "stable" build

thinkyhead, 2017-11-12:
It's no use reporting issues in release versions without first checking to see if we already fixed it, because we don't patch releases, we patch up bugfix and then later it becomes the next release.

WOOKIE111, 2017-11-12:
Confirm: BugFix compiles with Viki2 and with `pins_ramps.h` untouched.

wonder what changed, between the release and the bug fix to make it work again. or perhaps the old version i had and the current release. as it seems the generation in the middle fudged something.

I do feel like reporting bugs in a release is fruitful. as not everyone uses a bugfix or as some people call BETA firmware.

Maybe a changelog is needed so you can refer people to it and what not.

WOOKIE111, 2017-11-12:
I just checked for every mention of DOGLCD_SCK in the bugfix and its identical to the release i have of Nov 11

so this makes me more curious as to what went wrong?

WOOKIE111, 2017-11-12:
I have got the Release version from Nov 11 to compile with Viki and Viki2 etc.

I had to change `Conditionals_post.h` to include

`  /**
   * VIKI2, miniVIKI, and AZSMZ_12864 require DOGLCD_SCK and DOGLCD_MOSI to be defined.
   */
  #if ENABLED(VIKI2) || ENABLED(miniVIKI) || ENABLED(AZSMZ_12864)
    #ifndef DOGLCD_SCK
      #define DOGLCD_SCK  SCK_PIN
    #endif
    #ifndef DOGLCD_MOSI
      #define DOGLCD_MOSI MOSI_PIN
    #endif
  #endif`
somewhere before the last endif statement

not sure what this new functionality does. as in my older version of marlin 1.1.x ( sometime in may) doesnt have this chunk of code.

WOOKIE111, 2017-11-12:
Could you explain what this functionality is. and if im doing anything harmful ?

thinkyhead, 2017-11-12:
> Maybe a changelog is needed so you can refer people to it and what not.

There's no way that we can maintain a "Changelog" on our daily bug patches, which come fast, furious, haphazardly, and don't always survive until the release. But, if you want to compare the current `bugfix-1.1.x` to any previous release, Github provides excellent tools for the job. You can search on the commits or the Pull Requests using a date range or two version tags, branch names, etc.

For example, here's a link to list all merged [Pull Requests between 1.1.5 and 1.1.6](https://github.com/MarlinFirmware/Marlin/pulls?utf8=%E2%9C%93&q=is%3Apr%20merged%3A2017-08-28..2017-10-10).

thinkyhead, 2017-11-12:
> Could you explain what this functionality is. and if im doing anything harmful ?

Basically, it ensures that for LCDs that don't use some custom SPI pins that they can be assured of using the standard (hardware) SPI pins.

closed 2017-11-12 by thinkyhead

cwlandorf, 2017-12-02:
I ran into this error on whatever version I downloaded on November 22, 2017.  After setting variables for my printer (It will be a reconfigured Airwolf HD2X with two E3D Volcano printheads, Rambo 1.3 board with Viki 2) I would prefer not to have to download another Marlin and try and reconfigure all of my changes.  How can I best do that?

thinkyhead, 2018-01-26:
> I would prefer not to have to download another Marlin and try and reconfigure all of my changes. How can I best do that?

Do `M503 S0`, copy the G-code output, and save it to a file named "`restore.gcode`". After flashing the new firmware, send `restore.gcode` to the machine, then do `M500` to save the settings. This ensures your tuned settings will be preserved. You should also set the same values in your `Configuration.h` so that `M501` always restores to your preferred values.