MarlinFirmware/Marlin #8387: Compile error caused by wrong MIN/MAX_PROBE values (Cartesian printer)

Cycor, 2017-11-12:
- Description: The values for MIN_PROBE_X,MIN_PROBE_Y,MAX_PROBE_X,MAX_PROBE_Y add the offset of the probe to the X_MIN_POS,Y_MIN_POS etc, but it should substract it.

Values:
// The size of the print bed
#define X_BED_SIZE 180
#define Y_BED_SIZE 200

#define X_PROBE_OFFSET_FROM_EXTRUDER -27     // X offset: -left  +right  [of the nozzle]
#define Y_PROBE_OFFSET_FROM_EXTRUDER -32     // Y offset: -front +behind [the nozzle]
#define Z_PROBE_OFFSET_FROM_EXTRUDER -0.65  // Z offset: -below +above  [the nozzle]

which causes:
static assertion failed: UBL_PROBE_PT_3_X can't be reached by the Z probe.
for value: UBL_PROBE_PT_3_X 170

- Expected behaviour: 
MIN_PROBE_X: 27
MIN_PROBE_Y: 32
MAX_PROBE_X: 207
MAX_PROBE_Y: 232

- Actual behaviour: ---
MIN_PROBE_X: -27
MIN_PROBE_Y: -32
MAX_PROBE_X: 153
MAX_PROBE_Y: 168

- Steps to reproduce:
Use values above in configuration.h

[configuration.zip](https://github.com/MarlinFirmware/Marlin/files/1464825/configuration.zip)

GMagician, 2017-11-12:
It seems correct to me....look ad the schematic in configuration.h:
     +-- BACK ---+
      |                    |
   L |    (+) P         | R <-- probe (20,20)
   E |                    | I
   F | (-) N (+)       | G <-- nozzle (10,10)
   T |                     | H
     |    (-)              | T
     |                     |
    O-- FRONT ---+
 (0,0)

since X negative values means that your probe is on le left and your min nozzle position is 0 then probe can move till -27

Cycor, 2017-11-12:
You are right, I was looking at it from the nozzle position, not the probe position, i'll close the issue.

closed 2017-11-12 by Cycor