MarlinFirmware/Marlin #8389: Enabling 8-bit Teensyduino digtialWrite optimizations…

fiveangle, 2017-11-12:
I've found that 8-bit targets that use the Teensyduino board support package (Printrboard, Brainwaive, 5DPrint, etc.) can theoretically gain some small code optimizations by using more expensive (but faster ?) digtialWrite inlining and omitting digitalWrite  valid pin bounds checking.

From core_pins.h in the Teensyduino board support package:

```
// This will speed up digitalWrite slightly, but it
// adds to your code size.  However, if you don't
// have many digitalWrites, or most are compile time
// constant inputs, then this might be worthwhile
//#define DIGITAL_WRITE_EXPENSIVE_INLINE_OPTIMIZATION

// This removes checking in digitalWrite for pin numbers
// beyond the maximum number of pins.  Your program
// WILL CRASH is digitalWrite is called with a pin number
// too large, when this is enabled.  However, if you don't
// make such mistakes, you'll get a minor speedup.
//#define DIGITAL_WRITE_RISKY_OMIT_OVERFLOW_CHECK
```

I expect this could improve performance slightly everywhere since digitalWrite is a core function for Marlin, but on compile I noticed SRAM usage identical (PROGMEM usage does change though).

Not super familiar with the AVR implemenation of the Harvard model, does the fact that SRAM usage doesn't change allude that there really wasn't any optmization possible in the Marlin code ? (or that the compiler/linker was doing said optimization before).   Or is the PROGMEM usage change directly indicative of the optimizations effects ?

**Stock (PIO/Arduino IDE):**
```
Program:  118190 bytes (90.2% Full)
Data:       3901 bytes (47.6% Full)

Sketch uses 118408 bytes (96%) of program storage space. Maximum is 122800 bytes.
Global variables use 3897 bytes (47%) of dynamic memory, leaving 4295 bytes for local variables. Maximum is 8192 bytes.
```

**`DIGITAL_WRITE_EXPENSIVE_INLINE_OPTIMIZATION ` enabled:**

```
Program:  118224 bytes (90.2% Full)
Data:       3901 bytes (47.6% Full)

Sketch uses 118442 bytes (96%) of program storage space. Maximum is 122800 bytes.
Global variables use 3897 bytes (47%) of dynamic memory, leaving 4295 bytes for local variables. Maximum is 8192 bytes.
```

**`DIGITAL_WRITE_RISKY_OMIT_OVERFLOW_CHECK` enabled:** 
```
Program:  118172 bytes (90.2% Full)
Data:       3901 bytes (47.6% Full)

Sketch uses 118390 bytes (96%) of program storage space. Maximum is 122800 bytes.
Global variables use 3897 bytes (47%) of dynamic memory, leaving 4295 bytes for local variables. Maximum is 8192 bytes.
```

[_Note: don't pay attention to the percentages in the PIO usage reports, as it's assuming HalfKay bootloader, while I've hand-jammed the boards.txt in the Arduino IDE to reflect the larger DFU bootloader used on Printrboard_.]

fiveangle, 2017-11-12:
I fully expect these optimizations cause PINS_DEBUGGING to be unusable.

thinkyhead, 2017-11-14:
We don't use `digitalWrite` on the 8-bit targets, except for a very few cases.

fiveangle, 2017-11-14:
A surprise answer but one that makes my results more expected ! :)

closed 2017-11-14 by fiveangle