MarlinFirmware/Marlin #8390: Corexy setup crashes the table at the bottom when homing x or y irrelevant to where the table is

cncvac, 2017-11-12:
Thank you for submitting your feedback to the Marlin project.
Please use one of the templates below to fill out this box.

------------------------------------------------------------
### Feature Request
Please put [FR] in the issue title: `[FR] Add-on that goes 'ping'`

------------------------------------------------------------
### Compile Error
When I compile with `FEATURE_X` I get an error:
```
Paste_the_error_text_here
```

------------------------------------------------------------
### Bug Report
- Description: ---
- Expected behaviour: ---
- Actual behaviour: ---
- Steps to reproduce:
  - Do this
  - Do that

Attach a ZIP of `Configuration.h` and `Configuration_adv.h` by dropping here.

------------------------------------------------------------
### Bug Report Tips
- When troubleshooting, use `M502` followed by `M500` to reset EEPROM to defaults.
- Use `DEBUG_LEVELING_FEATURE` with `M111 S247` for detailed logging of homing/leveling.
- Format text with: **bold**, _italic_, `code`.
- Format C++ with three backticks, plus "cpp":
```cpp
void my_function(bool do_it) {
  // Hold this spot
}
```

cncvac, 2017-11-12:
Why you have close the report of the bug? I still haven't described what the problem is.

Roxy-3D, 2017-11-12:
There was no bug displayed.   It was an empty issue with no content.

cncvac, 2017-11-12:
Hi Roxy-3D,

Sorry I am new to this. I have open a new report where I describe what is happening.

You can close this.

George

closed 2017-11-12 by cncvac