MarlinFirmware/Marlin #8391: Full Graphic Smart Controller+Keypad

Lenivets71, 2017-11-12:
Hi, help me please . to connect in the firmware that is http://reprap.org/wiki/Keypad and it http://reprap.org/wiki/RepRapDiscount_Full_Graphic_Smart_Controller

Tannoo, 2017-11-13:
Help us help you....

Thank you for submitting your feedback to the Marlin project.
Please use one of the templates below to fill out this box.

------------------------------------------------------------
### Feature Request
Please put [FR] in the issue title: `[FR] Add-on that goes 'ping'`

------------------------------------------------------------
### Compile Error
When I compile with `FEATURE_X` I get an error:
```
Paste_the_error_text_here
```

------------------------------------------------------------
### Bug Report
- Description: ---
- Expected behaviour: ---
- Actual behaviour: ---
- Steps to reproduce:
  - Do this
  - Do that

Attach a ZIP of `Configuration.h` and `Configuration_adv.h` by dropping here.

------------------------------------------------------------
### Bug Report Tips
- When troubleshooting, use `M502` followed by `M500` to reset EEPROM to defaults.
- Use `DEBUG_LEVELING_FEATURE` with `M111 S247` for detailed logging of homing/leveling.
- Format text with: **bold**, _italic_, `code`.
- Format C++ with three backticks, plus "cpp":
```cpp
void my_function(bool do_it) {
  // Hold this spot
}
```

thinkyhead, 2017-11-14:
I don't think Marlin can do both at the same time.

boelle, 2019-02-19:
@Lenivets71 still having a problem?

thinkyhead, 2019-02-20:
I now believe Marlin *can* do both at the same time now, but it requires some extra tweaking to get working. With `REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER` enabled, just add this to the end of your `Configuration.h`:
```cpp
  #define ADC_KEYPAD_PIN 12 // Analog numbering!
  #define ADC_KEYPAD
  #define REPRAPWORLD_KEYPAD
  #define REPRAPWORLD_KEYPAD_MOVE_STEP 10.0
  #define ADC_KEY_NUM 8

  // Enable and change, if needed:
  //#define REVERSE_MENU_DIRECTION
  //#define ENCODER_PULSES_PER_STEP 1
  //#define ENCODER_STEPS_PER_MENU_ITEM 1
  //#define ENCODER_FEEDRATE_DEADZONE 2
```
You might need to comment out this line in `SanityCheck.h`:
```cpp
#error "Please select no more than one LCD controller option."
```

closed 2019-02-20 by thinkyhead