MarlinFirmware/Marlin #8392: Corexy setup crashes the table at the bottom when homing x or y irrelevant to where the table is

cncvac, 2017-11-12:
Thank you for submitting your feedback to the Marlin project.
Please use one of the templates below to fill out this box.

------------------------------------------------------------
### Feature Request
Please put [FR] in the issue title: `[FR] Add-on that goes 'ping'`

------------------------------------------------------------
### Compile Error
When I compile with `FEATURE_X` I get an error:
```
Paste_the_error_text_here
```

------------------------------------------------------------
### Bug Report
- Description: --- In a CoreXY printer when homing  X axis or Y axis the Z axis moves down until it reaches 5mm from the point where was the Z axis when you have switch off the machine. That has as a result the table to crash at the bottom of the printer. Something is wrong with the safe homing feature.
- Expected behaviour: --- The Z axis should move 5mm down from where it is and then home X and Y axis.
- Actual behaviour: --- The Z axis moves 5mm below the point where the printer was switched on
- Steps to reproduce:
  - Do this Give a G28 command
  - Do that

Attach a ZIP of `Configuration.h` and `Configuration_adv.h` by dropping here.

------------------------------------------------------------
### Bug Report Tips
- When troubleshooting, use `M502` followed by `M500` to reset EEPROM to defaults.
- Use `DEBUG_LEVELING_FEATURE` with `M111 S247` for detailed logging of homing/leveling.
- Format text with: **bold**, _italic_, `code`.
- Format C++ with three backticks, plus "cpp":
```cpp
void my_function(bool do_it) {
  // Hold this spot
}
```

thinkyhead, 2017-11-14:
I assume you have a moving Z bed, and the bed is moving down.

thinkyhead, 2017-11-14:
What is your setting for `Z_HOMING_HEIGHT`?

cncvac, 2017-11-14:
Hi Thinkyhead,

Yes I have a moving table and when you try to home the table is moving down almost for the entire z axis height down and crashes at the bottom.
Usually what you do is the following: When you switch on the printer the table is all the way down. Then you move the table up for some distance (down command to come closer to the nozzle) and the home.
In corexy what the printer does is that assumes that your 0 Z point is where your Z axis was when you have switch on the printer. Then from that point it moves even further. Even G92 X0 Y0 Z0 doesn't affect this behaviour.

My Z_HOMING_HEIGHT is set at 4mm but it is disabled.

George

RowanMeara, 2017-11-14:
I looked at G28 a couple days ago and tested with a few options and it seems to be working perfectly fine on my CoreXY

cncvac, 2017-11-14:
I have the latest bug fix 1.1.x Marlin and it doesn't work. In the previous Marlin 1.0.2 it was working OK.
I can upload the config file if that helps.

George

RowanMeara, 2017-11-15:
That would be great.  I'll stick the file on my printer
.

cncvac, 2017-11-15:
Config file attached.

George

[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/1473087/Configuration.zip)

RowanMeara, 2017-11-15:
This bug occurs on my setup.

AnHardt, 2017-11-15:
> '#define INVERT_Z_DIR true'

When you make manual moves in z+ direction - is your bed going down?

RowanMeara, 2017-11-15:
Inverting the axis has nothing to do with it.  To test this, you just need to disable software endstops for the z axis and then move the bed up (make the z position negative).

thinkyhead, 2017-11-15:
```cpp
#ifndef Z_HOMING_HEIGHT
  #ifndef Z_CLEARANCE_BETWEEN_PROBES
    #define Z_HOMING_HEIGHT 0
  #else
    #define Z_HOMING_HEIGHT Z_CLEARANCE_BETWEEN_PROBES
  #endif
#endif
```
What would your preferred behavior be? If the bed is mostly guaranteed to be far from the nozzle, then you can simply set `Z_HOMING_HEIGHT` to 0 and there will be no raise before homing. This will preserve your `Z_CLEARANCE_BETWEEN_PROBES` so for other types of probing there will be an appropriate clearance.

andreq, 2017-11-17:
What if the raise was Relative to the current position instead of Absolute? If the Z hasn't been homed yet, we can assume the user can move his Z where he want and move Current_Z + Z_HOMING_HEIGHT instead of trying to reach absolute Z_HOMING_HEIGHT.

cncvac, 2017-11-17:
I think andreg's suggested solution is better than move to a specific absolute value.

AnHardt, 2017-11-21:
```cpp
 //#define Z_HOMING_HEIGHT 4  // (in mm) Minimal z height before homing (G28) for Z clearance above the bed, clamps, ...
                             // Be sure you have this distance over your Z_MAX_POS in case.
```
Because of the 'hidden' dependency 
```cpp
 /**
 * Z_HOMING_HEIGHT / Z_CLEARANCE_BETWEEN_PROBES
 */
#ifndef Z_HOMING_HEIGHT
  #ifndef Z_CLEARANCE_BETWEEN_PROBES
    #define Z_HOMING_HEIGHT 0
  #else
    #define Z_HOMING_HEIGHT Z_CLEARANCE_BETWEEN_PROBES
  #endif
#endif
#ifndef Z_CLEARANCE_BETWEEN_PROBES
  #define Z_CLEARANCE_BETWEEN_PROBES Z_HOMING_HEIGHT
#endif
#if Z_CLEARANCE_BETWEEN_PROBES > Z_HOMING_HEIGHT
  #define MANUAL_PROBE_HEIGHT Z_CLEARANCE_BETWEEN_PROBES
#else
  #define MANUAL_PROBE_HEIGHT Z_HOMING_HEIGHT
#endif
```
to
```cpp
 #define Z_CLEARANCE_BETWEEN_PROBES  5 // Z Clearance between probe points
```
what is defined by default, the better default value for `Z_HOMING_HEIGHT` could look like
```cpp
 #define Z_HOMING_HEIGHT 0  // (in mm) Minimal z height before homing (G28) for Z clearance above the bed, clamps, ...
                            // Be sure you have this distance over your Z_MAX_POS in case.
```
--------------------------------------------------------------------------------
If `// Be sure you have this distance over your Z_MAX_POS in case.` is violated this will result in grinding, when the nozzle tries to rise above its limit (bed - drop - below).
The procedure to recover from this situation is a bit inconvenient:
* Move the nozzle down (bed up) 
* Reset Marlin

-------------------------------------------------------------------------------
What do we know when under what circumstances?

Right after a reset:
Our internal counters are at 0.
We do not know anything about our current position, except we are already touching an hardware endstop (homing switch).
The only thing we know is - our current position is somewhere in the allowed area.
Any automated move is in danger to grind, except it's going towards a hardware endstop.

After a (not grinding) manual move:
Our internal counters are at dx, dy, dz,
We know the area between [0,0,0] and [dx,dy,dz] is not in danger to grind.
Any automated move in that area is save.

For the z-move **by** `Z_HOMING_HEIGHT` this means:
After a manual move into the right direction a automated move is save, as long as it's done in the previously moved span.
Even a move bigger than  the previously manual move is saver than before because it's not going that far into the unknown area.
We give the user a chance to, get back the responsibility, from Marlin, for a maybe grinding automated move.

-------------------------------------------------------------------------------------------

Conclusion:
Make that damn  `Z_HOMING_HEIGHT` relative to the current position.
Recovery from booting with a too high z-axis would be a manual move into the right direction with no reset needed.

Sorry for the long posting, but i have to explain it to @thinkyhead, who reverted my changes from ~2 years ago because he did not understand. (I promise to not make a new suggestion for the constants name )

Hopeless Andreas

cncvac, 2018-02-07:
Any update on this one?
As it is now (1.1.8) it doesn't work with a moving table on the X axis.

George

thinkyhead, 2018-02-17:
@AnHardt — What do you think? Check for the Z endstop triggered at startup and use that to set the Z position as known?

boelle, 2019-02-19:
@cncvac have you tried the latest bugfix 2.0?

boelle, 2019-03-06:
@thinkyhead i think we can close this one

closed 2019-05-16 by thinkyhead