MarlinFirmware/Marlin #8401: Bltouch PIN issue?

EnricoSx, 2017-11-13:
[Configuration.h.txt](https://github.com/MarlinFirmware/Marlin/files/1466503/Configuration.h.txt)
[Configuration_adv.h.txt](https://github.com/MarlinFirmware/Marlin/files/1466502/Configuration_adv.h.txt)

i have this issue when use G29
https://youtu.be/Hx6NyPTb1TM

FW: Marlin 1.1.6
Printer: Anycubic Kossel Linear Plus
Bltouch connect to servo and z-min

M401  nozzle go up and touch all microswitch on top

m280 p0 s10   work
m280 p0 s90  work
m280 p0 s120  work
m280 p0 s160 work

thinkyhead, 2017-11-14:
Please test `bugfix-1.1.x` and see if the problem still exists.

EnricoSx, 2017-11-14:
Ok

EnricoSx, 2017-11-18:
@thinkyhead i load bugfix-1.1.x , 
now g29 
https://youtu.be/nT_SW1JG7aY

thinkyhead, 2017-11-18:
It looks to me like the BLTOUCH is failing because it's triggering too often. Try slowing down the feed-rates among these options.… 
```cpp
// X and Y axis travel speed (mm/m) between probes
#define XY_PROBE_SPEED 4000

// Speed for the first approach when double-probing (with PROBE_DOUBLE_TOUCH)
#define Z_PROBE_SPEED_FAST (20*60)

// Speed for the "accurate" probe of each point
#define Z_PROBE_SPEED_SLOW (Z_PROBE_SPEED_FAST / 2)

// Use double touch for probing
#define PROBE_DOUBLE_TOUCH
```

EnricoSx, 2017-11-18:
#define XY_PROBE_SPEED 2000
#define Z_PROBE_SPEED_FAST HOMING_FEEDRATE_Z
#define HOMING_FEEDRATE_Z  (20*60)
#define Z_PROBE_SPEED_SLOW (Z_PROBE_SPEED_FAST / 2)
#define PROBE_DOUBLE_TOUCH

after some probe
20:28:01.118 : Error:Probing failed
20:28:02.046 : X:97.00 Y:-36.00 Z:17.85 E:0.00 Count X:14394 Y:22688 Z:16221
[image]

thinkyhead, 2018-02-09:
Presuming this is solved.

closed 2018-02-09 by thinkyhead