MarlinFirmware/Marlin #8403: printer halted kill called()

MuhammadAlKadii, 2017-11-13:
I'm new to 3d printing and this is my first time to make one, everytime i try to load a file.gcode to print from pronterface the extruder not does not heat and ot give me error heating failed 
printer halted please reset
------------------------------------------------------------
Error:Heating failed, system stopped! Heater_ID: 0
[ERROR] Error:Heating failed, system stopped! Heater_ID: 0

Error:Printer halted. kill() called!
[ERROR] Error:Printer halted. kill() called!
------------------------------------------------------------

Tannoo, 2017-11-14:
Check all connections. Especially thermistor / thermocouples.

thinkyhead, 2017-11-14:
That's an appropriate response to your heater failing.

AnHardt, 2017-11-14:
'Error:Heating failed, system stopped! Heater_ID: 0' needs some time until it pops up.
Did the measured temperature rise in that time?
What is the displayed temperature? Is it far from room temperature?
Is there a light going on/off - indicating the heater is on?
Is your heater heating? Feel it.

If the nozzle gets warm, but it's not recognized by the thermocoupler and it's temperature shows room temperature. Is this the right thermocoupler?
If the heater gets warm, but not fast enough. Tune values in Configuration_adv.h
If heating is indicated by an led, but heater does not heat. Repair the heater.
If heating is not indicated by a LED, but it should do. Check pin-file.

mikko70, 2017-11-14:
Mayby you put it in wrong connecor (fan). Measure with volt meter.
What board you use?, if ramps, you must put power in both connector (jump wire)

http://www.mauroalfieri.it/wp-content/uploads/2015/10/Delta-3d-printer-K250-elettronica-ramps-1.4-schema-600x488.jpg

MuhammadAlKadii, 2017-11-15:
now when I load file and start printing the extruder begin to work fine and the screen writes E1 heating... 
and the temperature raise up will for about 2 or 3 minutes then that error appear again

AnHardt, 2017-11-15:
> If the heater gets warm, but not fast enough. Tune values in Configuration_adv.h

So your heater is slow.

```
/**
 * Thermal Protection provides additional protection to your printer from damage
 * and fire. Marlin always includes safe min and max temperature ranges which
 * protect against a broken or disconnected thermistor wire.
 *
 * The issue: If a thermistor falls out, it will report the much lower
 * temperature of the air in the room, and the the firmware will keep
 * the heater on.
 *
 * If you get "Thermal Runaway" or "Heating failed" errors the
 * details can be tuned in Configuration_adv.h
 */

#define THERMAL_PROTECTION_HOTENDS // Enable thermal protection for all extruders
#define THERMAL_PROTECTION_BED     // Enable thermal protection for the heated bed
```
```
   /**
   * Whenever an M104, M109, or M303 increases the target temperature, the
   * firmware will wait for the WATCH_TEMP_PERIOD to expire. If the temperature
   * hasn't increased by WATCH_TEMP_INCREASE degrees, the machine is halted and
   * requires a hard reset. This test restarts with any M104/M109/M303, but only
   * if the current temperature is far enough below the target for a reliable
   * test.
   *
   * If you get false positives for "Heating failed", increase WATCH_TEMP_PERIOD
   * and/or decrease WATCH_TEMP_INCREASE. WATCH_TEMP_INCREASE should not be set
   * below 2.
   */
  #define WATCH_TEMP_PERIOD 20                // Seconds
  #define WATCH_TEMP_INCREASE 2               // Degrees Celsius
```

closed 2017-11-15 by thinkyhead

sriram135, 2019-03-28:
Hello I have a DIY 3d printer which has E3d v6 hotend.. when I load file and give print in pronterface the printer starts but the hotend temperature never increase it simply fluctuates ( increases and Decreases) and then Error heating failed system stopped comes please help