MarlinFirmware/Marlin #8404: defined BED_MAXTEMP is off by 15 from what we can actually set

Tcoton, 2017-11-13:
### Bug Report
- Description: Max bed temperature cannot be reached as it is minored by 15 on the LCD screen.
- Expected behaviour: I should be able to set the bed temperature to the max set for BED_MAXTEMP
- Actual behaviour: The max temperature available on LCD screen is minored by 15
- Steps to reproduce:
  - #define BED_MAXTEMP 120 in Configuration.h
  - On LCD screen, try to set the bed temperature to 120, it is impossible as it stops at 105
  - #define BED_MAXTEMP 150 in Configuration.h
  - On LCD screen, try to set the bed temperature to 150, it is impossible as it stops at 135

[configuration.zip](https://github.com/MarlinFirmware/Marlin/files/1468034/configuration.zip)

AnHardt, 2017-11-13:
Right. But that's not an error but by will.
BED_MAXTEMP is the temperature where the protection mechanisms will shot down the heaters and the complete machine. Du to overshot when heating the machine will be haltet every time you try to heat to BED_MAXTEMP. To avoid that there is a safety margin of 15K(elvin). Same as for the other heaters.

Tcoton, 2017-11-13:
Did you meant 15 Celsius because it is 15 Celsius on screen. Maybe the description in the configuration file might explain that better, it was driving me mad. Current wording is: "When temperature exceeds max temp, your heater will be switched off."

AnHardt, 2017-11-13:
For differences K = °C.

Tcoton, 2017-11-13:
OK.

closed 2017-11-13 by Tcoton