MarlinFirmware/Marlin #8405: M206 faulty?

chrisqwertz, 2017-11-13:
- Description: doing small (0.1mm) movements in Z axis with M206 offset != 0 will lead to z-movement in either wrong direction or wrong distance.

- Expected behaviour: 
Using M206 Z-0.2 I would correct my Z nozzle offset. From there I do some test probing on the bed using 
 
**G91**
 **G1 Z**[-/+]**0.1**
 **G90**

to validate my offset 
(i'm the jog buttons on Octoprint to be precise).

- Actual behavior: With M206 Z != 0 the commands will lead to either to a move in the reverse direction or more than the wanted 0.1mm in Z height

- Steps to reproduce:

**M502**
**M500**
**M501**
**M206 Z-0.20**

and then use
**G91**
**Z0.1**
**G90**

or

**G91**
**Z-0.1**
**G90**

to try and change z height.

Using **M206 Z0** the unwanted behavior stops again.

I am on the bugfix-1.1.x branch (commit: d2df00bfca8e3edee4db2414de78ba095f8a96c4)

configuration and configuration_adv attached:

[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/1468165/Marlin.zip)

thinkyhead, 2017-11-14:
> Using M206 Z-0.2 I would correct my Z nozzle offset

You should use `M851` for that. `M206` is an offset to the coordinate space.

chrisqwertz, 2017-11-14:
Thanks for the clarification. Still the behaviour seems wrong. On the 1.1.x branch M206 works the way it should, though.

thinkyhead, 2017-11-15:
By "on the 1.1.x branch" do you mean, in version 1.1.6? We know there are issues with some coordinate space elements in recent days since the "native workspace" overhaul. If M206 is misbehaving, it's because `home_offset` is one of the things that has been harder to morph into proper shape under the new system. I hope to do more debugging on M206 and the workspace in-general over the next few days and shake out the remaining bugs.

chrisqwertz, 2017-11-16:
Yes, I mean version 1.1.6.

Tested M851 Z-0.2 on the bugfix branch, but it doesn't do anything to the z height of my nozzle.

thinkyhead, 2017-11-17:
> Tested `M851 Z-0.2` on the bugfix branch, but it doesn't do anything to the z height of my nozzle.

It won't move the Z axis in response unless:
- You have `BABYSTEP_ZPROBE_OFFSET` enabled
- You have previously homed with `G28`.

Since you're using `MESH_BED_LEVELING` you would need to re-probe the bed after updating the Z probe offset, or you may either:
- Manually re-write the mesh with `G29 S3 Xn Yn Zn`
- Set a new mesh Z-offset with `G29 S4 Zn`
```cpp
   * Parameters With MESH_BED_LEVELING:
   *
   *  S0              Produce a mesh report
   *  S1              Start probing mesh points
   *  S2              Probe the next mesh point
   *  S3 Xn Yn Zn.nn  Manually modify a single point
   *  S4 Zn.nn        Set z offset. Positive away from bed, negative closer to bed.
   *  S5              Reset and disable mesh
```

mrvanes, 2017-11-22:
I'm having problems with M206 since 1.1 as well. M206 used to give me an opportunity to correct for a smaller Z_MAX_POS caused by a slightly longer nozzle than standard, saved to eeprom. This was "set once and forget". The new "correct" behaviour of M206 in 1.1 doesn't add to Z_MAX_POS but shifts the whole coordinate system so that my nozzle crashes into the bed on Z=0 (I admit, this is exactly what M206 should do). The problem is, I miss the old M206 to compensate for the new nozzle length in Z_MAX_POS without recompiling the firmware and store the new machine characteristic in eeprom (instead of changing start g-code).
For now, I do recompile the firmware so my problem has a work-around, but I guess more people exchanging nozzles would benefit from a M206 that behaves like the old one?

This is the only patch I make to make M206 usable for me again:
```
--- a/Marlin/Marlin_main.cpp
+++ b/Marlin/Marlin_main.cpp
@@ -2351,7 +2351,7 @@ static void clean_up_after_endstop_or_probe_move() {
     #endif
 
     return RAW_CURRENT_POSITION(Z) + zprobe_zoffset
-      #if ENABLED(DELTA)
+      #if HAS_HOME_OFFSET || ENABLED(DELTA)
         + home_offset[Z_AXIS] // Account for delta height adjustment
       #endif
     ;
```

thinkyhead, 2017-11-26:
The existing `M206` could work…

Let's say that with the shorter nozzle, your Z home position is normally Z=200. If the second nozzle was longer by 2mm, then you want the Z max (home) position to be Z=198. To achieve this you simply issue `M206 Z-2` and `G28`, and now Z=198. Now when you move to Z=0 the longer nozzle will be just touching the bed.

With the current code, coordinates are "offset" as a consequence of `M206`. So the physical movement limits shifted, and now go from Z-2 up to Z198.

mrvanes, 2017-11-26:
Hmm... it's been a while since I thoroughly tested this, but what you describe is what I expected but didn't see. At Z=2 my nozzle touched the bed and Z=0 would dig into the bed. It was only when I patched like above that the behaviour was what like you describe. This is a couple of months ago so I will give it another try to see if things improved?

mrvanes, 2017-11-26:
Maybe it's because my Lulzbot mini uses washers to probe the bed and need probe offset to compensate for washer thickness?

nanotech, 2017-12-29:
I also ran into this issue with M206 in combination with G91 relative movements. (I'm on 1.1.8.) Relative movements have the M206 offset added to them, making opposite movements (e.g. Z0.3 and Z-0.3 in the log below) not return to the same location.

<details>
<summary>Gcode log</summary>

    >>>M206 Z-0.2
    X:0.00 Y:0.00 Z:0.20 E:8.04 Count X:0 Y:0 Z:907
    ok
    >>>G28 Z0
    echo:busy: processing
    X:0.00 Y:0.00 Z:-0.20 E:8.04 Count X:0 Y:0 Z:0
    ok
    >>>G91
    ok
    >>>G0 Z0.3
    ok
    >>>M114
    X:0.00 Y:0.00 Z:0.30 E:8.04 Count X:0 Y:0 Z:1134
    ok
    >>>G0 Z0.3
    ok
    >>>M114
    X:0.00 Y:0.00 Z:0.80 E:8.04 Count X:0 Y:0 Z:2268
    ok
    >>>G0 Z-0.3
    ok
    >>>M114
    X:0.00 Y:0.00 Z:0.70 E:8.04 Count X:0 Y:0 Z:2041
    ok
    >>>G0 Z-0.3
    ok
    >>>M114
    X:0.00 Y:0.00 Z:0.60 E:8.04 Count X:0 Y:0 Z:1814
    ok

</details>

&nbsp;  
Poking around the code, it looks like `gcode_get_destination` is incorrectly applying the `LOGICAL_TO_NATIVE` coordinate conversion to the (relative offset + `current_position`) value even though `current_position` is already in native coordinates (according to its declaration comment). That is, the `v` variable in the function is in logical coordinates when from an absolute position, but is already in native coordinates when from a relative position.

This patch seems to work for me so far, though there may be other things I'm missing:

```diff
diff --git a/Marlin/Marlin_main.cpp b/Marlin/Marlin_main.cpp
index bc2541d..9c049f6 100644
--- a/Marlin/Marlin_main.cpp
+++ b/Marlin/Marlin_main.cpp
@@ -3260,8 +3260,9 @@ static void homeaxis(const AxisEnum axis) {
 void gcode_get_destination() {
   LOOP_XYZE(i) {
     if (parser.seen(axis_codes[i])) {
-      const float v = parser.value_axis_units((AxisEnum)i) + (axis_relative_modes[i] || relative_mode ? current_position[i] : 0);
-      destination[i] = i == E_AXIS ? v : LOGICAL_TO_NATIVE(v, i);
+      const bool relative = axis_relative_modes[i] || relative_mode;
+      const float v = parser.value_axis_units((AxisEnum)i) + (relative ? current_position[i] : 0);
+      destination[i] = i == E_AXIS || relative ? v : LOGICAL_TO_NATIVE(v, i);
     }
     else
       destination[i] = current_position[i];
```

bz007, 2018-01-30:
Same issue as above on 1.1.8 with M206 Zoffset. The above patch is working for me. Thanks, @nanotech!

VibroAxe, 2018-02-05:
Same issues as above, Patch appears to resolve. @nanotech you should submit that patch to the bugfix tree

thinkyhead, 2018-02-05:
The patch has already applied to the bugfix branch, thanks.

closed 2018-02-05 by thinkyhead