MarlinFirmware/Marlin #8407: memory consumption on 1284p Melzi

mamama1, 2017-11-13:
Hi

is it true/possible that the sd card feature alone takes up about 70% of the 1284p memory (= about 11 KB)?

With the SD feature compiled it takes about 92% of dynamic memory and without it takes up about 20%.

Is that expected?

Thanks!

fiveangle, 2017-11-14:
Yes.

More info on how to attack this problem here: https://github.com/MarlinFirmware/Marlin/issues/7190

-=dave

Roxy-3D, 2017-11-14:
I was thinking a person could run the  avr-objdump program and see how big the SD memory card object files are.

thinkyhead, 2017-11-14:
Indeed! Due to the huge amount of memory in the 1284p the CR-10 config has `SDCARD_ALPHA_SORT` set to maximum caching. Feel free to tweak these options for less memory usage.

mamama1, 2017-11-14:
ook, cause I was wondering how THAT would fit into a ATmega2560, everyone is "upgrading" to.
Yes, it has more IOs and more flash but dynamic memory is the half...

Thanks, I'll look into that, was just wondering if that is expected or if there is some unsupported combination of settings in the my configuration file which leads to that huge amount of used dynamic memory.

closed 2017-11-15 by thinkyhead

mamama1, 2017-11-15:
indeed, when I disabled the SDCARD_ALPHA_SORT feature, memory usage dropped to about 25%. unbelievable what a memory hog that feature is. maybe that fact could be mentioned somewhere in the comments really prominently?

if I'd have more time I'd look into the code if there is any chance of optimizing, however I guess that has been written by geniuses and that I won't have any chance of getting that thing optimized in any way...

thinkyhead, 2017-11-15:
> indeed, when I disabled the SDCARD_ALPHA_SORT feature, memory usage dropped to about 25%. unbelievable what a memory hog that feature is. maybe that fact could be mentioned somewhere in the comments really prominently?

It uses very little memory if configured differently. Play with the sub-options, which are described reasonably well in the configuration.

Deneteus, 2017-11-15:
Its designed to eat memory apparently with some of the rest of the features.

thinkyhead, 2017-11-15:
Yes, indeed! The allocation it uses in the CR-10 configuration is an extreme, but the 1284p has all that extra SRAM which otherwise Marlin would just leave sitting unused. Marlin allocates all the SRAM it needs at compile time, so the only thing that it uses the unallocated SRAM for is the stack, and for that ~1K should be more than enough.

mamama1, 2017-11-15:
i like that software ;-) great piece of work!