MarlinFirmware/Marlin #8413: Erratic extrusion

timmit99, 2017-11-14:
### Bug Report
- Description: 
          Erratic movement on extruder when printing using Marlin bugfix_2.0.x
- Expected behaviour: 
          Standard movement and extrusion
- Actual behavior: 
       Print heats up, bed levels, and moves to first location and then does a giant retraction and then seemingly at random decides which direction to move the extruder during the printing moves.

[Config Files.zip](https://github.com/MarlinFirmware/Marlin/files/1469589/Config.Files.zip)

thinkyhead, 2017-11-14:
Can you try sending each command from your test G-code one at a time? First, enable the option `M114_DETAIL` in `Configuration.h`. In-between each command use `M114` to check the `current_position`. When you start to see weirdness happening, either stop there or continue with a few more commands.

Alternatively, edit your G-code file and stick `M114` after every G-code command. Use `M111 S247` to enable max logging before you start, and that way all commands will end up in the log.

Roxy-3D, 2017-11-14:
I've seen this too.   If I reset the board, connect with PronterFace and tell it to extrude some amount....

Usually, the first extrude is  too fast and it jumps.    And it doesn't always go the right direction.   It is like the extruder's position isn't being initialized to a good value.

thinkyhead, 2017-11-14:
@Roxy-3D To help narrow down this issue please apply the debugging procedure described above.

Roxy-3D, 2017-11-14:
I'll try to remember to do that in the morning...

timmit99, 2017-11-14:
@Roxy-3D Thats is exactly what happenes. I have done a driver swap, full reboot, computer restart and no luck. The g-code looks no different than any other ones i have used. the "E" values increase slowly as expected yet the motor seems to jump around as if it was doing random retracts and extrudes.

@thinkyhead let me do  a COMPLETE marlin 2.0.x reinstall. I have been working on the LED menu options and i might have messed something up between versions. I'll do that tomorrow night after work.

Roxy-3D, 2017-11-14:
@timmit99    I'm not seeing it do it again and again.   It is the very first extrusion after a reset.   

Are you seeing it do it repeatably?

timmit99, 2017-11-14:
yeah, ill take a video of it tomorrow if i can. It just seems to go whichever direction it wants with a really large movements

karabas, 2017-11-14:
I saw the similar on marlin RC0.7. But cannot replicate.

thinkyhead, 2017-11-15:
An `M114` at any point may be revealing.

boelle, 2019-02-19:
@timmit99 have you tried bugfix 2.0?

timmit99, 2019-03-03:
@boelle this was over a year ago, I have since switched to Klipper on that machine and didn't even remember this was an open issue.

closed 2019-03-03 by timmit99